esphome/feature-requests #562: Support for Tuya composite devices

epiniguin, 2020-01-11:
ESPHome currently supports only dimmer devices but does not support devices that combine multiple thing, for example dimmer and switch.

I bought Moeshouse "Fan light switch" - https://www.moeshouse.com/collections/smart-fan-light-switch/products/wifi-smart-ceiling-fan-light-wall-switch-smart-life-tuya-app-remote-various-speed-control-compatible-with-alexa-and-google-home-1
It is a Tuya device based on TYWE3S and MCU. It has a triac dimmer for fan or lamp and relay switch.

[image]

[image]

I uploaded ESPHome to it via UART without any problem.
In the log I see three datapoint:
1 - switch for fan(dimmer)
2 - dimmer level
3 - relay switch

In the EPSHome config I wrote:
```
light:
  - platform: "tuya"
    name: "dim1"
    dimmer_datapoint: 2
    switch_datapoint: 1
```
It works for dimmer.
But I can't manage a relay switch(datapoint 3).

Also I can't set dimmer level 51%-99% from Homeassistant. If I set level above 50% the dimmer sets a maximum level. But I can set it via dimmer button.
From the log I see that a maximum level is 255.

brandond, 2020-01-12:
Can you grab the boot/config logs from the Tuya component? I'm curious what datapoints it reports.

epiniguin, 2020-01-13:
Hi @brandond!
There is a log below.
I changed a dimmer level with the device buttons +/- and I switched on/off a switch relay(datapoint 3).

```
[09:02:24][I][app:100]: ESPHome version 1.14.3 compiled on Jan 11 2020, 18:08:17
[09:02:24][C][wifi:415]: WiFi:
[09:02:24][C][wifi:283]:   SSID: 'pine2'
[09:02:24][C][wifi:284]:   IP Address: 11.0.1.207
[09:02:24][C][wifi:286]:   BSSID: 34:08:04:CD:61:7E
[09:02:24][C][wifi:287]:   Hostname: 'fan_lamp'
[09:02:24][C][wifi:291]:   Signal strength: -75 dB ▂▄▆█
[09:02:24][V][wifi:293]:   Priority: -2.0
[09:02:24][C][wifi:295]:   Channel: 6
[09:02:24][C][wifi:296]:   Subnet: 255.255.255.0
[09:02:24][C][wifi:297]:   Gateway: 11.0.1.1
[09:02:24][C][wifi:298]:   DNS1: 11.0.1.1
[09:02:24][C][wifi:299]:   DNS2: (IP unset)
[09:02:25][C][uart:140]: UART Bus:
[09:02:25][C][uart:142]:   TX Pin: GPIO1
[09:02:25][C][uart:145]:   RX Pin: GPIO3
[09:02:25][C][uart:147]:   Baud Rate: 9600 baud
[09:02:25][C][uart:148]:   Stop bits: 1
[09:02:25][C][uart:150]:   Using hardware serial interface.
[09:02:25][C][logger:175]: Logger:
[09:02:25][C][logger:176]:   Level: VERBOSE
[09:02:25][C][logger:177]:   Log Baud Rate: 0
[09:02:25][C][logger:178]:   Hardware UART: UART0
[09:02:25][C][light:177]: Light 'dim1'
[09:02:25][C][light:179]:   Default Transition Length: 0.0s
[09:02:25][C][light:180]:   Gamma Correct: 1.00
[09:02:25][C][tuya.light:027]: Tuya Dimmer:
[09:02:25][C][tuya.light:029]:    Dimmer has datapoint ID 2
[09:02:25][C][tuya.light:031]:    Switch has datapoint ID 1
[09:02:25][C][ota:029]: Over-The-Air Updates:
[09:02:25][C][ota:030]:   Address: fan_lamp.local:8266
[09:02:25][C][api:095]: API Server:
[09:02:25][C][api:096]:   Address: fan_lamp.local:6053
[09:02:25][C][tuya:023]: Tuya:
[09:02:25][C][tuya:026]:   Configuration will be reported when setup is complete. Current init_state: 3
[09:02:25][C][tuya:027]:   If no further output is received, confirm that this is a supported Tuya device.
[09:02:25][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=3
[09:02:25][V][tuya:098]: Received Tuya: CMD=0x00 VERSION=1 DATA=[01 (1)] INIT_STATE=3
[09:02:25][V][tuya:115]: MCU Heartbeat (0x01)
[09:02:26][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=3
[09:02:26][V][tuya:098]: Received Tuya: CMD=0x00 VERSION=1 DATA=[01 (1)] INIT_STATE=3
[09:02:26][V][tuya:115]: MCU Heartbeat (0x01)
[09:02:27][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=3
[09:02:27][V][tuya:098]: Received Tuya: CMD=0x00 VERSION=1 DATA=[01 (1)] INIT_STATE=3
[09:02:27][V][tuya:115]: MCU Heartbeat (0x01)
[09:02:28][V][tuya:098]: Received Tuya: CMD=0x07 VERSION=1 DATA=[01.01.00.01.01 (5)] INIT_STATE=3
[09:02:28][V][tuya:235]: Datapoint 1 update to 1
[09:02:28][D][light:264]: 'dim1' Setting:
[09:02:28][V][tuya:275]: Datapoint 2 set to 112
[09:02:28][V][tuya:279]: Not sending unchanged value
[09:02:28][V][tuya:275]: Datapoint 1 set to 1
[09:02:28][V][tuya:279]: Not sending unchanged value
[09:02:28][V][tuya:098]: Received Tuya: CMD=0x07 VERSION=1 DATA=[02.02.00.04.00.00.00.6D (8)] INIT_STATE=3
[09:02:28][V][tuya:235]: Datapoint 2 update to 109
[09:02:28][D][light:264]: 'dim1' Setting:
[09:02:28][D][light:277]:   Brightness: 43%
[09:02:28][V][tuya:098]: Received Tuya: CMD=0x07 VERSION=1 DATA=[03.01.00.01.00 (5)] INIT_STATE=3
[09:02:28][V][tuya:235]: Datapoint 3 update to 0
[09:02:28][V][tuya:275]: Datapoint 2 set to 109
[09:02:28][V][tuya:279]: Not sending unchanged value
[09:02:28][V][tuya:275]: Datapoint 1 set to 1
[09:02:28][V][tuya:279]: Not sending unchanged value
[09:02:28][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=3
[09:02:28][V][tuya:098]: Received Tuya: CMD=0x00 VERSION=1 DATA=[01 (1)] INIT_STATE=3
[09:02:28][V][tuya:115]: MCU Heartbeat (0x01)
[... 51 lines cut ...]
[09:02:33][D][light:264]: 'dim1' Setting:
[09:02:33][V][tuya:275]: Datapoint 2 set to 106
[09:02:33][V][tuya:279]: Not sending unchanged value
[09:02:33][V][tuya:275]: Datapoint 1 set to 1
[09:02:33][V][tuya:279]: Not sending unchanged value
[09:02:33][V][tuya:098]: Received Tuya: CMD=0x07 VERSION=1 DATA=[02.02.00.04.00.00.00.6A (8)] INIT_STATE=3
[09:02:33][V][tuya:235]: Datapoint 2 update to 106
[09:02:33][D][light:264]: 'dim1' Setting:
[09:02:33][D][light:277]:   Brightness: 42%
[09:02:33][V][tuya:098]: Received Tuya: CMD=0x07 VERSION=1 DATA=[03.01.00.01.00 (5)] INIT_STATE=3
[09:02:33][V][tuya:235]: Datapoint 3 update to 0
[09:02:33][V][tuya:275]: Datapoint 2 set to 106
[09:02:33][V][tuya:279]: Not sending unchanged value
[09:02:33][V][tuya:275]: Datapoint 1 set to 1
[09:02:33][V][tuya:279]: Not sending unchanged value
[09:02:34][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=3
[09:02:34][V][tuya:098]: Received Tuya: CMD=0x00 VERSION=1 DATA=[01 (1)] INIT_STATE=3
[09:02:34][V][tuya:115]: MCU Heartbeat (0x01)
```

With 1.14.1 ESPHome reported a datapoints in the config log. But now with 1.14.3 it reports:
```
[09:02:25][C][tuya:026]:   Configuration will be reported when setup is complete. Current init_state: 3
[09:02:25][C][tuya:027]:   If no further output is received, confirm that this is a supported Tuya device.
```

epiniguin, 2020-01-13:
There is a log where I changed a dimmer level from Homeassistant to 69%.
And after that the dimmer level changed to 100%.

```
[09:13:00][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=3
[09:13:00][V][tuya:098]: Received Tuya: CMD=0x00 VERSION=1 DATA=[01 (1)] INIT_STATE=3
[09:13:00][V][tuya:115]: MCU Heartbeat (0x01)
[09:13:01][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=3
[09:13:01][V][tuya:098]: Received Tuya: CMD=0x00 VERSION=1 DATA=[01 (1)] INIT_STATE=3
[09:13:01][V][tuya:115]: MCU Heartbeat (0x01)
[09:13:02][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=3
[09:13:02][V][tuya:098]: Received Tuya: CMD=0x00 VERSION=1 DATA=[01 (1)] INIT_STATE=3
[09:13:02][V][tuya:115]: MCU Heartbeat (0x01)
[09:13:03][D][light:264]: 'dim1' Setting:
[09:13:03][D][light:277]:   Brightness: 69%
[09:13:03][V][tuya:275]: Datapoint 2 set to 176
[09:13:03][V][tuya:261]: Sending Tuya: CMD=0x06 VERSION=0 DATA=[02.02.00.04.00.00.00.B0 (8)] INIT_STATE=3
[09:13:03][V][tuya:275]: Datapoint 1 set to 1
[09:13:03][V][tuya:279]: Not sending unchanged value
[09:13:03][V][tuya:098]: Received Tuya: CMD=0x07 VERSION=1 DATA=[02.02.00.04.00.00.00.B0 (8)] INIT_STATE=3
[09:13:03][V][tuya:235]: Datapoint 2 update to 176
[09:13:03][D][light:264]: 'dim1' Setting:
[09:13:03][D][light:277]:   Brightness: 69%
[09:13:03][V][tuya:275]: Datapoint 2 set to 176
[09:13:03][V][tuya:279]: Not sending unchanged value
[09:13:03][V][tuya:275]: Datapoint 1 set to 1
[09:13:03][V][tuya:279]: Not sending unchanged value
[09:13:03][V][tuya:098]: Received Tuya: CMD=0x07 VERSION=1 DATA=[01.01.00.01.01 (5)] INIT_STATE=3
[09:13:03][V][tuya:235]: Datapoint 1 update to 1
[09:13:03][D][light:264]: 'dim1' Setting:
[09:13:03](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[09:13:03](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[09:13:03](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[09:13:03](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[09:13:03][V][tuya:275]: Datapoint 2 set to 176
[09:13:03][V][tuya:279]: Not sending unchanged value
[09:13:03][V][tuya:275]: Datapoint 1 set to 1
[09:13:03][V][tuya:279]: Not sending unchanged value
[09:13:03][V][tuya:098]: Received Tuya: CMD=0x07 VERSION=1 DATA=[03.01.00.01.00 (5)] INIT_STATE=3
[09:13:03][V][tuya:235]: Datapoint 3 update to 0
[09:13:03][V][tuya:098]: Received Tuya: CMD=0x07 VERSION=1 DATA=[01.01.00.01.01 (5)] INIT_STATE=3
[09:13:03][V][tuya:235]: Datapoint 1 update to 1
[09:13:03][D][light:264]: 'dim1' Setting:
[09:13:03][V][tuya:275]: Datapoint 2 set to 176
[09:13:03][V][tuya:279]: Not sending unchanged value
[09:13:03][V][tuya:275]: Datapoint 1 set to 1
[09:13:03][V][tuya:279]: Not sending unchanged value
[09:13:03][V][tuya:098]: Received Tuya: CMD=0x07 VERSION=1 DATA=[02.02.00.04.00.00.00.FF (8)] INIT_STATE=3
[09:13:03][V][tuya:235]: Datapoint 2 update to 255
[09:13:03][D][light:264]: 'dim1' Setting:
[09:13:03][D][light:277]:   Brightness: 100%
[09:13:03][V][tuya:098]: Received Tuya: CMD=0x07 VERSION=1 DATA=[03.01.00.01.00 (5)] INIT_STATE=3
[09:13:03][V][tuya:235]: Datapoint 3 update to 0
[09:13:03][V][tuya:275]: Datapoint 2 set to 255
[09:13:03][V][tuya:279]: Not sending unchanged value
[09:13:03][V][tuya:275]: Datapoint 1 set to 1
[09:13:03][V][tuya:279]: Not sending unchanged value
[09:13:03][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=3
[09:13:03][V][tuya:098]: Received Tuya: CMD=0x00 VERSION=1 DATA=[01 (1)] INIT_STATE=3
[09:13:03][V][tuya:115]: MCU Heartbeat (0x01)
[09:13:04][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=3
[09:13:04][V][tuya:098]: Received Tuya: CMD=0x00 VERSION=1 DATA=[01 (1)] INIT_STATE=3
[09:13:04][V][tuya:115]: MCU Heartbeat (0x01)
```

brandond, 2020-01-13:
Your device is using protocol version 1. The current Tuya component only supports MCUs that speak protocol version 3.
Version 3 is well documented and commonly used on dimmers. 
Version 0 is used on battery-powered devices such as door and motion sensors, and is currently supported by a custom component available [here](https://github.com/brandond/esphome-tuya_pir/). I've never seen anything using version 1.

It looks like it works similar to version 3, but something in the initial handshake isn't working right.

epiniguin, 2020-01-15:
And what about switching the relay with datapoint 3?