esphome/feature-requests #563: Support ST7920

Misiu, 2020-01-11:
**Describe the problem you have/What new integration you would like**

Consider adding support for ST7920 display. It is a good alternative for SSD1325. 
**Please describe your use case for this integration and alternatives you've tried:**

Use case is fairly simple, I'd like to be able to use ST7920 with ESP Home
**Additional context**

Recent version of ESP8266 library supports hardware SPI, display code already exists, so it shouldn't be hard for someone with good knowledge of this awesome project

Misiu, 2020-01-14:
I'm trying to add support for this. I have 128x64 ST7920 display so I'll be able to test just this one (there are other sizes).
I've created the first version by copying code for SSD1325 - https://github.com/Misiu/esphome/commit/a770bb10184c7f2a46e401eeaa4cd7be4716be44
I need some guidance with the setup code.
I'm sure ST7920 is different from SSD1325, so that code must be changed.
https://github.com/olikraus/u8g2 has support for that display - https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ist7920.c

What commands I should send to correctly set up the display? Where I should search for them?
Manual (https://www.lcd-module.de/eng/pdf/zubehoer/st7920_chinese.pdf)?

@OttoWinter any hints?

Misiu, 2020-12-08:
Ref: https://thesolaruniverse.wordpress.com/2020/06/09/esp8266-and-st7920-powered-12864-liquid-crystal-display-in-a-test-bench/
@kbx81 maybe you could help with this? 🙏

kbx81, 2020-12-08:
Yes, I can probably help. I'll take a look at your branch later today or tomorrow.

moskovskiy82, 2021-01-24:
Any update on this one? Also interested in this.

kbx81, 2021-01-25:
Sorry -- been distracted with some other project. I'll circle back before too much longer. 😇

nicolaij, 2021-01-25:
Try it https://github.com/nicolaij/esphome_heater

Misiu, 2021-01-26:
there is a PR: https://github.com/esphome/esphome/pull/1440
could you give this a try? I'm waiting for a new display so I can help with testing.

Misiu, 2021-02-16:
@moskovskiy82 did you try the code from the PR I mentioned?