esphome/feature-requests #567: D1 mini motor shield

accelle17, 2020-01-16:
**Describe the problem you have/What new integration you would like**

D1 mini motor shield
[https://www.aliexpress.com/item/32891891983.html?spm=a2g0o.productlist.0.0.e78840feRKJzHN&algo_pvid=9c3c6d7e-7e32-4766-ae1e-a07e5d2b3416&algo_expid=9c3c6d7e-7e32-4766-ae1e-a07e5d2b3416-2&btsid=0f014cc3-6980-4cc5-9c8a-a5112e58ed95&ws_ab_test=searchweb0_0,searchweb201602_3,searchweb201603_53](url)

**Please describe your use case for this integration and alternatives you've tried:**

I’m trying to modify glade air freshener by attaching d1 mini on it.
However, it looks like I need to replace the original board (motor driver) with d1 mini
motor shield.

With esphome, how do you control this kind of motor driver? 

**Additional context**

I tried the following config but it doesn’t work:

```
output:
 - platform: gpio
   pin: D2
   id: direction_pin
 - platform: esp8266_pwm  # or ledc if on ESP32
   pin: D3
   id: speed_pin
switch:
  - platform: template
    name: "Kitchen Glade"
    turn_on_action:
     - output.turn_on: direction_pin
     - output.set_level:
        id: speed_pin
        level: 100%
     - delay: 1000ms
     - output.turn_off: speed_pin
     - output.turn_off: direction_pin
     - output.set_level:
        id: speed_pin
        level: 100%
     - delay: 1000ms
     - output.turn_off: speed_pin

```

glmnet, 2020-01-16:
doesn't it have a "dispense now" button? I'd try to hack it (connect to a ESPHome switch)

accelle17, 2020-01-16:
Yeah it has and I was able to make it work by connecting the dispense to gnd and d1. I was also able to remove the batteries and rely on the d1 mini with battery shield's power supply. However as soon as I introduce deep sleep, the motor automatically activates after sleep.
I was also able to make it work with relay shield. By doing there is already four boards inside (d1+battery shield+ relay+ original board).
 I really want to remove the original board so that there's only three inside.
The aim is to let the air freshener function only if there's someone at home.

glmnet, 2020-01-16:
Well the shield seems to be i2c controlled, so a component is needed, it can be a custom component or ESPHome can provide one if somebody creates it, [here](https://github.com/wemos/WEMOS_Motor_Shield_Arduino_Library) is the sample code

MrRodz, 2022-10-04:
Hi how do we get this added to esp home? Thanks