esphome/feature-requests #568: Add support for Sense-U BLE Baby Monitor

inforsgit, 2020-01-17:
**Describe the problem you have/What new integration you would like**
Add an integration to read values from the Sense-U Baby Monitor (https://www.products.sense-u.com/) - the BLE button, not the camera system. This is a small button that clips to a baby's clothes to monitor breathing rate, position and temperature and transmits this information via BLE. It is designed to work with a smartphone app but that limits range to being close to the smartphone.

**Please describe your use case for this integration and alternatives you've tried:**
 It would be great if an ESP32 running ESPHome could listen to the messages from the device and provide the information as three sensors to Home Assistant for monitoring a baby remotely or to use as an input to automations either within the ESP32 device or in HA.

**Additional context**
The plan I would suggest adopting for this sensor would be similar to the Xiaomi sensors that are available already and very close indeed to the Ruuvi Tag sensor that is currently in development. I have already scanned the BLE messages that are produced by the device and have found that a given UUID notifies on each update of a sensor result. I am currently decoding how the information in this is stored and that can be used as input to the integration. I'm happy to start writing the integration based on the Ruuvi Tag implementation in the dev channel.

caesar1987sk, 2020-05-18:
I plan to buy this baby monitor, so it will be nice to have this support :)

caesar1987sk, 2020-05-29:
@inforsgit did you anything with this baby monitor?

inforsgit, 2020-05-31:
Sadly not, I had too much time looking after the baby :)

caesar1987sk, 2020-05-31:
i am trying to get some data from sensor. when app on mobile is not started it send allways same data, when app on mobile is started it dont send anything. I am getting data using

` logger:
    level: VERY_VERBOSE`

maybe i need to call some of the services but i dont know how.

JamieP83, 2021-06-11:
Hey all. Any progress made here

andyboeh, 2022-09-23:
@inforsgit I'm thinking of buying a Sense-U button, could you share the information you've already gathered?

inforsgit, 2022-09-26:
Sadly I don't have much more information - our little one grew up fast and we've stopped using the sensor. I still think though that in principle it should be possible to integrate this as it seems to send unencrypted bluetooth info.

rbflurry, 2022-09-29:
I have the sense-u, and esp32 and baby on the way. Hope to see what I can figure out.

andyboeh, 2022-10-02:
I bought a Sense-U with the base station (which is ESP-based, btw). Unfortunately, I didn't succeed in even setting up the base station, so I'm sending it back. The button, however, pairs fine with my phone and I started taking BLE logs.

There are some characteristics which need to be written to for the initial setup and some notifications need to be enabled. Afterwards, one characteristic changes with real time data. Now the challenge is figuring out what the data means :)

andyboeh, 2022-10-05:
@rbflurry How are you getting along, any progress?
As I have quite some understanding of the protocol now, I'll create a Python implementation before I move on to an ESPHome implementation.

caesar1987sk, 2022-10-05:
When you want to decode those data, you can download apk and decompile it. In decompiled source code you can find how to "read" data. I was able to decode captured data stored in log file in phone, but I was not able to force sensor to do communication with those initial data sensor need to receive.

andyboeh, 2022-10-05:
@caesar1987sk Thanks, I already did that. I think I have a good understanding how the init sequence works and how the data has to be interpreted. A very quick Python script could connect to the sensor, initialize it and retrieve data. For now, I just replayed my init sequence, but most if not all of the values are known.

I will publish my findings soon, it's just that the baby is also keeping us busy ;)

andyboeh, 2022-10-06:
I've uploaded a README containing a BLE log with comments. This should be sufficient to come up with an implementation: https://github.com/andyboeh/esphome-sense-u

rbflurry, 2022-10-06:
@andyboeh I am just getting started on this. I see you have made plenty of progress, if there is anything I can help with let me know!

andyboeh, 2022-10-06:
There is actually plenty that can be done / investigated:

  * How can this be integrated in ESPHome, i.e. what's the best way to implement such a component?
  * Are there example components that do similar BLE transactions, i.e. read/write characteristics and receive notifications?
  * How is the pairing performed with ESPHome? What should trigger it?
  * Which sensors could represent the device?
    * Breathing [float]
    * Temperature [float]
    * Humidity [float]
    * Position [Text]
    * Breathing Alarm [Binary]
    * Temperature Alarm [Binary]
    * Position Alarm [Binary]
    * Connection State [Binary]
    * Automations?
  * How could a configuration file look like?

And of course: code :) I know that my protocol assumptions are more or less correct as I have a simple Python script at hand that can pair to the device and can receive data. It's just too messy to publish it ;) But I didn't start with any ESPhome/C++ code.

caesar1987sk, 2022-10-06:
Esphome support work with characteristics:
https://esphome.io/components/ble_client.html
https://esphome.io/components/sensor/ble_client.html

andyboeh, 2022-10-07:
I was referring to the C++ side of ESPHome, not the YAML side - I intend to create a component in C++ that does all the handling and parsing.

caesar1987sk, 2022-10-07:
i know, just wanted to let you know that esphome is already supporting that so you can use it. :)

andyboeh, 2022-10-10:
Good news: I just pushed the very first experimental code to my repo. After discovering and resolving some quirks, pairing and configuring works, notifications are received. However, setting the options via YAML and parsing the responses to sensor values is not implemented yet - I'll need a few more days for this.

Quirks: 

1. For whatever reason, they decided to use the BLE address in the UUID field of the characteristics. I discovered this only by chance because I have logs from two different devices at hand. This is taken into account already.
2. There is no real pairing procedure going on. It took me a while to figure out that "pairing" works by sending some kind of user ID to the device and it responds with a baby code. This code has to be sent whenever the device configuration needs to be changed. Pairing status and baby code are not yet stored on the ESPHome device, so every time the ESP is rebooted, the Sense-U needs to be paired again. This will be fixed later, I'll have to take a look at how preferences are stored.
3. Powering on/off the device is actually just the command to enable/disable alerts. This will probably be implemented as a switch.

My suggestion for a configuration file looks like this (not yet supported) - what do you think?

```
esp32_ble_tracker:

ble_client:
  - mac_address: AA:BB:CC:DD:EE:FF
    id: my_senseu_ble

senseu:
  - ble_client_id: my_senseu_ble
    id: my_senseu

sensor:
  - platform: senseu
    id: my_senseu
    breath_rate: 
      name: Breath Rate
    temperature: 
      name: Temperature
    humidity: 
      name: Humidity

text_sensor:
  - platform: senseu
    id: my_senseu
    posture:
      name: Posture
    state:
      name: State

binary_sensor:
  - platform: senseu
    id: my_senseu
    breath:
      name: Breath Alarm
    temperature:
      name: Temperature Alarm

switch:
  - platform: senseu
    id: my_senseu
    power:
      name: Power Switch
```

andyboeh, 2022-10-11:
First working version pushed to my github - has a few caveats and bugs, but the basics are working fine for me. Not tested extensively.

andyboeh, 2022-10-17:
So I've been tracking our baby's sleep for two nights with my component and it's working nicely! All data visible within home assistant and, surprisingly, no false alert. Breath and posture data make sense, it's also clearly visible when my wife left the room to change diapers, automatic reconnection is working fine.

The good thing about the "pairing" process is that you can track different rooms if you have several ESPHome devices. I did not test, however, what happens if more than one device is connected at a given time.

Next step: Port a cheap fitness tracker to ESPHome so that it can send an alert directly to the tracker / watch. There is now no limit on the alarm mechanism, anything supported by ESPHome and/or Home Assistant is going to work. Just keep in mind that the added complexity does not make it safer and that it won't turn the device into a medical device.

rbflurry, 2022-10-24:
@andyboeh incredible work on this, are you still using the base station or app with the sense-u. Just brought our baby girl home yesterday, I have both the sense u + base station and just the sense u. not sure what one to keep or send back,

andyboeh, 2022-10-24:
Congratulations and all the best for your baby girl!

No, I only use the ESPHome integration. You can switch between app and ESPHome integration, but then you need to extract the baby code from the BLE log during pairing with the app. 

I never managed to get the base station to connect so I sent it back (it always paired with the Sense-U but never got a connection to the server).

lordratner, 2022-11-16:
I just wanted to chime in and say that I'm following this work eagerly. Due date in March, and planning to use this integration. If you have a link set up for donations, please post it.

andyboeh, 2022-11-16:
@lordratner thanks for the suggestion - finally, I finished setting up Github Sponsors: https://github.com/sponsors/andyboeh

This is now also reachable via the repository: https://github.com/andyboeh/esphome-sense-u

andyboeh, 2022-11-26:
My primary efforts so far were focused on getting the button supported via BLE - for me, this works stable enough for daily use.
I acquired a Base Station and guess what's inside? An ESP32. The Base Station 2 that I have uses an AiThinker ESP32-S module, Rx/Tx/IO0 are available via an unpopulated header. I will now focus on running ESPHome instead of the proprietary firmware.
The Base Station 3 has an Espressif module inside (that's its DHCP Client ID), but I don't have the BS3 anymore.

andyboeh, 2022-11-26:
Good news: The Base Station 2 is now nearly fully supported. The only thing left is the temperature/humidity sensor because I have yet to identify the pins and the chip in use.

With the Sense-U Base Station 2 powered by ESPHome you get:

- fully configurable behavior
- two independently configurable RGB lights
- one touch button
- one recessed button
- piezo buzzer
- WiFi
- Bluetooth

With the Sense-U component, this makes a fully-featured replacement firmware for the base station with support for Home Assistant. Details will be published in my github repository soon.

lordratner, 2022-11-26:
Well now that's something. So this is the Sense-U 2, not 3, correct? Was it easy to flash ESPHome to it?

andyboeh, 2022-11-27:
Exactly, Sense-U 2 (bought it used). I had a BS3 for a short time and, before sending it back, I noticed that its DHCP Client ID is "espressif" - it's most probably also ESP32-powered. Judging from the location of the power connector on BS2 and BS3, the PCBs differ, but I don't know to which extent.

The flashing procedure is the easy part, there are unpopulated headers for all relevant pins. Figuring out the GPIOs for buttons and LEDs is also relatively easy (I used trial and error, but you can simply measure it, I just forgot my multimeter at home).