esphome/feature-requests #569: Periodical Average within 'filters' in a sensor

sermayoral, 2020-01-17:
**Describe the problem you have/What new integration you would like**

I would like to have a way of obtaining the average of a sensor periodically. For example, a thermometer that reports data every 120 seconds, and that the reported data is the average of all the values read in that period.

**Please describe your use case for this integration and alternatives you've tried:**

Esphome currently have two ways of doing the average values:
  - sliding_window_moving_average
  - exponential_moving_average

But both are based on the number of readings they get. The number of readings is variable, and it depends on many factors such as ble coverage, etc.

It would be ideal to have a way to specify a period, and to know that the sensor will always report data in that period, regardless of the number of readings it takes.

**Additional context**

There is a custom component for HA that works great. It can be used for reference: https://github.com/custom-components/sensor.mitemp_bt

period

(positive integer)(Optional) The period in seconds during which the sensor readings are collected and transmitted to Home Assistant after averaging. Default value: 60.

To clarify the difference between the sensor broadcast interval and the component measurement period: The LYWSDCGQ transmits 20-25 valuable BT LE messages (RSSI -75..-70 dBm). During the period = 60 (seconds), the component accumulates all these 20-25 messages, and after the 60 seconds expires, averages them and updates the sensor status in Home Assistant. The period does not affect the consumption of the sensor. It only affects the Home Assistant sensor update rate and the number of averaged values. We cannot change the frequency with which sensor sends data.

marklister, 2020-12-09:
Use of the heartbeat filter before the sma filter may work where the underlying sensor is converging on the true value.

sermayoral, 2021-10-09:
> I'm also highly interested in such a filter and would code it myself, but got no experience with C++.
> 
> **Problem:** I'm using the fantastic Pulse Meter component by @stevebaxter to track my current power load with high accuracy. This works great for all kinds of loads (especially low loads), but sends a lot of sensor updates for high loads. For example for a load of 18kW this sensor sends out 5 updates per 1 second... For high load conditions this results in more used transmission power (wifi) than necessary and it spams my wifi network. **Current Solutions:**
> 
> 1. I could switch over to the Pulse Counter component, solving the problem for high loads. Unfortunately this results in low accuracy for low loads, which is why Pulse Meter was implemented initially
> 2. I could use filters, like `throttle: 5s` to achieve lower sensor update rates, but am missing sensor values as everything <5s is discarded and thus may miss load spikes.
> 
> **Possible solution:** A timed average filter, like as `timed_average: 5s`. This filter delivers new values if more than 5s passed between 2 values and calculates the average if at least 2 values are received within 5s. Pseudo code:
> 
> ```
> val last_value_time = 0, sum = 0, count = 0
> timed_average (value, interval) {
>   val current_time = now
>   sum += value
>   count += 1
>   if (current_time - last_value_time >= interval) {
>     val average = sum / count
>     sum = count = 0
>     last_value_time = now
>     return average
>   }
>   return undefined
> }
> ```
> 
> So for slow sensor rates this filter outputs `value / 1` and for high sensor rates `sum / count`. This effectively allows me to have accurate sensor updates for low load conditions and to not miss values for high load conditions, like with the `throttle: 5s` filter. It won't even need a synced time source, as the uptime of the ESP would be sufficient. Additionally this filter prevents my wifi network from being spammed and reduces the energy usage of the ESP as it lowers the wifi transmission rates for high load conditions.
> 
> EDIT: the maximum delay of this filter is `interval * 2`. For example for `interval = 5s` and new values coming in every 4.9s, averages are returned after 9.8s. Such delays are preventable by using the same technique as implemented for the [heartbeat](https://github.com/esphome/esphome/blob/5461f87ff016cbb31a1ffe6876062dfee3f0ee63/esphome/components/sensor/filter.cpp#L302) filter: - execute the filter every interval at least once with the last value and indicate if it was triggerd by this heartbeat to prevent the filter from reporting if no value was recorded within the interval.

@rmeissn what do you think about doing the documentation? I can try to code it 😉