esphome/feature-requests #570: Use LED strips in a matrix as display

escoand, 2020-01-17:
**Describe the problem you have/What new integration you would like**
Currently it's not possible to use LED strips organized in a matrix with the ESPHome DisplayBuffer API.

**Please describe your use case for this integration and alternatives you've tried:**
This custom component creates an additional DisplayBuffer to circumvent this problem: https://github.com/escoand/matrix_led_display

**Additional context**

brandond, 2020-01-17:
Oh hey, this is the next place I was going to go with my light grouping component: https://github.com/esphome/feature-requests/issues/554#issuecomment-574045082

My first task was to get grouping working so I can have virtual pixels that are made up of multiple LEDs. Adding DisplayBuffer support was next but this is perfect!

escoand, 2020-01-18:
The main problem I see is that ESPHome currently only allows bi-color (on / off) with the DisplayBuffers. My solution is working but upstream we should have a better way.

brandond, 2020-01-18:
https://github.com/esphome/feature-requests/issues/51 also needs color display support.

docteurzoidberg, 2020-09-29:
fasatled component can be enhanced with the fastledmatrix library and could became a display AND light component ?

escoand, 2021-05-04:
Was added to 1.17.0: https://github.com/esphome/esphome/pull/1272

closed 2021-05-04 by escoand