esphome/feature-requests #572: Fingerprint Sensor supports

minsuke, 2020-01-18:
I found SSL Fingerprints in ESPHOME.io page 
      https://esphome.io/components/mqtt.html#ssl-fingerprints 
      Can I use this sensor for esphome ? 

**Describe the problem you have/What new integration you would like**

      I want to use this sensor for security if it supports 

**Please describe your use case for this integration and alternatives you've tried:**

     I tried in arduino code but I don't know how to make it in ESPHOME 

**Additional context**

     I always thanks for esphome developers Otto and all contributers thanks again.

teitelbot, 2020-01-26:
Looking for the same thing...

daned33, 2020-01-27:
Looks like a FZ2904 fingerprint module.

With any arduino experience you could create support yourself. With calls for enrolling and scanning.

I'd check it out if I had the module, but don't. What I am able to tell you is that is more than likely that you could make esphome work with it nicely.

minsuke, 2020-01-29:
> Looks like a FZ2904 fingerprint module.
> 
> With any arduino experience you could create support yourself. With calls for enrolling and scanning.
> 
> I'd check it out if I had the module, but don't. What I am able to tell you is that is more than likely that you could make esphome work with it nicely.

thanks for your reply 

If you have chance to make it, I hope you can share your creatice codes. 

someday this module can be applied for esphome.

wesparish, 2020-06-14:
I'm also looking for a fingerprint sensor and wouldn't mind taking a stab at creating a new component, however, I have a few questions to help prevent going down the wrong path.

1. Are there any examples of components that require some kind of post-installation setup? Everything I've skimmed over looked to be stateless. These fingerprint sensors have an "enroll" process that must be completed. I could see the component having multiple modes (enroll mode vs verification mode) that are activated based on the status of a pin.
2. Any recommendations for getting started on component development - maybe a good component to look at as a reference?

Thanks!

glmnet, 2020-06-15:
Some components like Dallas and touch esp32 have a setup part. We usually use logs for the user to see which values they will need to use in the yaml. 

So maybe you can turn on an enrollment  mode, with a fixed option in the yaml which you have to upload to activate. Then you download another firmware in verify mode. Or you can set a template binary sensor so you can switch from and to enroll mode via hass ui. 

On the logs when the enroll is going on you should print some information like the fingerprint hash code or whatever it needs to later match in the verification time.

wesparish, 2020-06-16:
Thanks for the tips. I'll see what I can come up with.

loongyh, 2020-11-02:
Hi, I have just submitted a PR to implement support for the Adafruit Fingerprint Library, which seems to support this model according to the amazon reviews.

closed 2021-04-28 by jesserockz