esphome/feature-requests #573: Si7051 temperature (+/- 0.1 C accuracy) sensor support

ronnylov, 2020-01-19:
**Describe the problem you have/What new integration you would like**

I would like having support for Si7051 temperature sensor in ESPhome.

**Please describe your use case for this integration and alternatives you've tried:**

I want a high precision calibrated temperature sensor at low cost that I can use to calibrate my cheaper sensors. I want to try this sensor to get better accuracy than current available alternatives.

**Additional context**

There is an arduino library for this I2C sensor. Maybe it can be used when developing for esphome. https://github.com/closedcube/ClosedCube_Si7051_Arduino
Data cheet: https://www.silabs.com/documents/public/data-sheets/Si7050-1-3-4-5-A20.pdf
I plan to buy a couple of ClosedCube Si7051 breakout sensors for testing.

ruffle-b, 2020-01-22:
That's serendipitous :-)  I just bought some of these myself and managed to get them working as a custom component in esphome.

I did do some flailing around to get the "ClosedCube Si7051" platformio library installed but can't remember the exact steps - I'll leave that as an exercise for the reader :-)

The sensor broadly agrees with my DS18B20's so I guess it's producing the right data.

I've attached my si7051_sensor.h and si7051_sensor.yaml files (with a .txt added as Github doesn't like .h or .yaml) 

[si7051_sensor.h.txt](https://github.com/esphome/feature-requests/files/4096819/si7051_sensor.h.txt)
[si7051_sensor.yaml.txt](https://github.com/esphome/feature-requests/files/4096820/si7051_sensor.yaml.txt)

which should get you going.

electronics4life, 2020-02-24:
Couldn't someone start with the STS3X and tweak it to work with the si7051? They seem like very similar chips, but maybe a different i2c address.