esphome/feature-requests #574: Synchronize Hass Dashboard with files from local PC

pax0r, 2020-01-20:
**Describe the problem you have/What new integration you would like**

I prefer writing and compiling my YAML files using full featured IDE on my PC, but I still would like to use a dashboard in HASS for example for logs or to see which devices are not online. 

I would like to have a possibility to synchronize my local files with the ones on HASS. The best would be a two-way (so simple edits on HASS will also be pushed to PC), but keeping the PC as single point of true may be enough.

**Additional context**

I think git repository might be a nice way to do this (as a bonus it will keep a track of changes). First version might even be a simple bash script to pull the given repo every few minutes.

If you like the idea and could point me where to start I may work on the implementation.

rradar, 2020-01-25:
I really not sure that something like this should be handled by esphome (actually **I think it shouldn't**).

The local PC can exist of various OSes (Linux, Windows, Mac for example) and therefor would need a lot's of work and maintenance from the esphome project. It could also causes a lot's of problems with interactions on the client pc and various software running on it.

It shouldn't really be the cope of esphome to implement something like this but rather the users responsibility.

You can for example use **ssh, ftp** or **smb** on your hassio machine and easily sync this on your client pc with one of the thousand (or more?) available sync tools.

pax0r, 2020-01-26:
While I get your point in fact my idea was to add a support to some sync method just on the side of the ESPHome hass.io dashboard and let users handle local pc by themselfs.

For example synchronization using git could look like that - 
1) user puts a link do their repo in hass ESPHome dashboard (and maybe credentials)
2) files and pull'd from git ever x minutes 
3) it's completly up to user how they will push the files from their local PC 

Also I think nice addition (and super easy to do) would be a button with "upload my YAML config" so one would not need to go through the wizzard to simply upload existing configs to ESPHome dashboard in hass.

torwag, 2020-08-02:
I came here for the same reason... Running esphome as a docker container on a server, which is hard (or at least not pleasant)  to physically access.

I usually start with some local config files and testing of a new device on my bench, esp. if I need to access USB to flash first. If all goes smooth, I finally test the OTA update and if this works, I would love to have an "upload yaml" button in addition to the + button. Thus, one could finally upload a yaml-file easily into the main system on the server, without the need for additional services, etc.
I think technically it is a rather small feature.... upload a file, check yaml for being correct, place file in the config folder.

OttoWinter, 2020-08-06:
No that shouldn't be done on the ESPHome side.

There are plenty of options for this use-case. Use Samba/SFTP (or any other file sharing option) to expose the config directory for external edits. The dashboard will pick up those changes after a reload (the dashboard really is just a text editor, which shows YAML files in the config directory).

Or use GIT and one of the HA addons to load the latest git data (for example the browser terminal one).

closed 2020-08-06 by OttoWinter

torwag, 2020-08-06:
It's ok, I just thought it would be an rather easy implementation in the dashboard and round things up.... There is, albeit it never worked for me, a button to download the bin file, thus it would be kind of logical to enable an easy way for yaml upload. But it is not a big issue and I can easily work around it.