esphome/feature-requests #575: Send unique device ID via zeroconf

frenck, 2020-01-20:
**Describe the problem you have/What new integration you would like**

It would be awesome if ESPHome sended out a unique device ID on its zeroconf broadcast (e.g., the MAC address of the device).

This would allow Home Assistant to identify the devices uniquely better, and also allows for a user to ignore the discovery of devices (as introduced in Home Assistant 0.104).

**Please describe your use case for this integration and alternatives you've tried:**

I would love to hear an alternative, but this seems a valid and pretty simple request.

**Additional context**

None.

Hedda, 2021-05-19:
FYI, thegroove is making a Zeroconf custom component for ESPHome -> https://github.com/thegroove/esphome-zeroconf

More discussion in https://github.com/thegroove/esphome-zbbridge/issues/1 as related to using Zeroconf fro ESPHome based Zigbee serial gateways.

The same type of features is wanted there as well as passing config flow parameters to the ZHA integration via txt records.

frenck, 2021-05-19:
Not sure how that is related, this is about ESPHome its own announcement on zeroconf.

frenck, 2021-05-19:
Actually, this request has been resolved already, thus can be closed.
The mac address of the device is part of the zeroconf already.

[image]

closed 2021-05-19 by frenck