esphome/feature-requests #576: random effect in the rgbw/rgbww platform

phixion, 2020-01-20:
**Describe the problem you have/What new integration you would like**
Hardware: Teckin SB50 Bulbs

Randomization of white values in the rgbw/rgbww platform should be optional when the corresponding effect is choosen. 

**Please describe your use case for this integration and alternatives you've tried:**
Most of the time the much brighter white LEDs sort of overwrites the RGB ones which means that in most cases you get a white colour with a slight touch of rgb. 
My use case is a use the provided random colorcycle effect as a "party mode".

Alternatives I have tested:
- reduce max_power of white output to like 1%  (removes the feature to have a warm light temperature )
- disable white output entirely (same as above)
- load effects off to a homeassistant script/template
```
      entity_id: light.{{['light1','light2','light3','light4']|random}}
      rgb_color:
        [
          "{{ (range(0, 255)|random) }}",
          "{{ (range(0, 255)|random) }}",
          "{{ (range(0, 255)|random) }}",
        ]
      brightness: "{{ (range(254, 255)|random) }}"
      transition: "{{ (range(55, 70)|random) }}"
      white_value: "{{ (range(1, 5)|random) }}"
```
Untested alternatives:
- lambda call

**Additional context**

Idea:

- Add config options brightness, white_value and maybe colortemperature (which I think is just a mix of those?) to random effect
- Add support  to expose the white LEDs as a seperate second light.

quinnhosler, 2020-01-27:
I have an RGBWW led and have noticed this problem with the white LEDs as well. It seems to a couple aspects of selecting colors, especially with the various voice assistants as they don't adjust the white value when adjusting the color. 

One solution I've found to your problem, though, is the below lambda function. 

```
      - lambda:
          name: random
          update_interval: 4.0s
          lambda: |-
            auto call = id(novostella1).turn_on();
            call.set_transition_length(4000);
            light::ESPColor c = light::ESPColor::random_color();
            call.set_rgb((c.red%128)/255.0, (c.green%128)/255.0, (c.blue%128)/255.0);
            call.perform();
```

You'll notice I added a few "%128" to the RBG values. I've noticed that colors are more vibrant if each one of the colors is kept below 50%. Obviously this limits the color palate, but transitions tend to be a bit more interesting.

phixion, 2020-01-27:
> especially with the various voice assistants as they don't adjust the white value when adjusting the color.

I have noticed that with google-home, but never connected the dots. I'll give your solution a shot and report back.

dashdrum, 2020-08-16:
thanks **quinnhosler** for the lambda.  I added this line:

call.set_white(0.0);

after the set_rgbw call to turn off white

closed 2022-07-06 by nagyrobi