esphome/feature-requests #578: Option to disable publishing assumed_state to HA on time based cover

pszafer, 2020-01-21:
**Describe the problem you have/What new integration you would like**
I'd like to ask if you could add option to disable publishind assumed_state in `Time-Based Cover` to Home Assistant.
It is pretty easy to see if button is locked in frontend and time-based cover sends `assumed_state: true` so both button up/down and stop are always active.

**Please describe your use case for this integration and alternatives you've tried:**
I guess I could try somehow create public template cover with internal template cover but want to ask here first if it wouldn't better to add such option.

**Additional context**
Thank you for Esphome!

ricreis394, 2020-01-30:
I had like to see this implemented too.
I have a few covers, and the only way to do that is by creating a template cover off esphome.

glmnet, 2020-01-31:
I don’t understand this request

pszafer, 2020-01-31:
I'll try to explain.
in HA if device publish in `state_attributes` option `assumed_state: true` like this:
[image]

then both buttons on cover are always ACTIVE, even if esphome sends state `closed`.
[image]

If you avoid `assumed_state` in `state_attributes` then it looks like this (this is fork of tasmota based time cover):
[image]

I see that it is set here: https://github.com/esphome/esphome/blob/2d0d794a9dab9f538d65744fddbe6ecbee6c50e9/esphome/components/time_based/time_based_cover.cpp#L54

I imagine something config like this:
```
cover:
  - platform: time_based
    name: "Time-Based Cover"
    public_assumed_state: true/false
```

glmnet, 2020-01-31:
Thanks, I got it now.

We can add `assumed_state` like in https://esphome.io/components/cover/template.html but instead of defaulting to `false` default it to `true`  otherwise this would be somewhat a breaking change.

IMO this makes more sense, for this cover, however in my particular use case of this integration I don't want this.

pszafer, 2020-01-31:
I think option `assumed_state` would be awesome. It can be made default as `true`, so it wouldn't be breaking change.

pszafer, 2020-02-07:
@glmnet just want to ask if you are gonna implement it?
Sorry for asking, just I'm not sure if those labels represents that feature request is accepted or no.

glmnet, 2020-02-07:
I might implement it but I can't promise anything.
The good first issue means any developer is welcomed to implement this, and it is not a complicated stuff to do in the code base.

closed 2020-03-12 by brandond