esphome/feature-requests #580: Individual acceleration setting for stepper

noxhirsch, 2020-01-22:
**Describe the problem you have/What new integration you would like**
The stepper function only allows to set the acceleration/decceleration once in the initial config. It would be great to set individual acceleration/decceleration within a script:
```
- stepper.set_acceleration :
    id: my_stepper
    acceleration : 10000
```

**Please describe your use case for this integration and alternatives you've tried:**
I have a cover with a stepper. When the cover gets power, it has to home first. That happens in a loop with tiny steps until the endstop is reached. After that it opens/closes the cover in one go - there I use the acceleration. But because of the acceleration the homing gets ridiculously slow. It would be great if I could set the acceleration to infinite just for the homing.
I didn't find any alternative ways to solve this problem.

**Additional context**

daned33, 2020-01-25:
How are you doing the homing for the stepper?

daned33, 2020-01-25:
Has this worked for you?

https://github.com/esphome/feature-requests/issues/127#issuecomment-486769288

noxhirsch, 2020-01-25:
> How are you doing the homing for the stepper? I am personally trying to work out how to do it in a smart way.

This is my homing-script:
```
script:
  - id: homing
    then:
      - logger.log: "Script Homing START"
      - stepper.set_speed:
          id: rollo_stepper
          speed: 2000 steps/s
      - while:
          condition:
            binary_sensor.is_off: endstop
          then:
            - stepper.set_target:
                id: rollo_stepper
                target: 20
            - delay: 1ms
            - stepper.report_position:
                id: rollo_stepper
                position: 0
      - lambda: 'id(calibrated) = 1;'
      - lambda: 'id(position) = 0;'
      - logger.log: "Script Homing END" 
```

> Has this worked for you? #127 (comment)
As you can see, I use it in my script above - and it's working perfectly fine. For normal open/close I have Home Assistant sensor with which I can control the speed from Home Assistant: 
Slow speed (and silent because of the low speed) for automatic open/close with alarm and sunrise/sunset. Fast speed if I'm using the buttons on my dashboard.

daned33, 2020-01-26:
Ahh that's exactly what I was looking for. 

I'm creating a DIY curtain pulley system and this is exactly what I needed. Do you have any images of your physical setup? And if poss, can I see the whole yaml config for this project? Thanks a lot.

noxhirsch, 2020-01-26:
I added my complete config above. 
Just in case: I recently changed some things in this config. While it seems to work fine, it might not be perfect yet. There's also no stop function, so you can only completely open or close with this script.

I don't have any images of the electronics and a picture of the outside of the box wouldn't help you, I think ;) But I can describe it:
To move my roller blinds I use a Nema 17 with a gearbox (mostly that my roller blinds don't sag while the stepper has no power). As stepper driver is use a TMC2130 or similar (had one lying around - not sure about the exact model number). This one hast StealthChop which gets rid of the usual stepper whining. As an endstop I use a binary hall sensor from a cheap Arduino sensor kit which detects a magnet I added in the weight bar at the bottom of my roller blinds. An ordinary endstop switch would also do the chop - I just wanted to have it as invisible as possible :)

daned33, 2020-01-27:
Thanks for that, appreciate it.

Yeah, it's a shame that there is no function to stop the stepper motor before it reaches its target.

I'm unable to test for a while. Is it possible to stop the motor by setting it to its target?

The module could be developed a bit more. I'll see if I'm able to this summer.

noxhirsch, 2020-01-27:
> Is it possible to stop the motor by setting it to its target?

What do you mean exactly? You can of course set the target to another point. So half open would be -36000 in my case. I could create a second Home Assistant sensor with a target value and set the target in a Home Assistant script.
The only other option I know (and used before I realised, that I don't need half open) is to put the open/close script in a loop similar to my homing script. Then you have to put in a delay so you can use `script.stop: open-script` when `cover.stop` is called. I don't know how long the delay has to be to make `script.stop` working, but in my case everything above 1ms let the stepper stutter and with that short delay `script.stop` only worked every 10th try. Also you obviously can't use acceleration with those short moves in a loop.

closed 2021-07-05 by jesserockz