esphome/feature-requests #582: Add BT support for AM43 devices ( blinds controller etc ) /( VAR-SOM-AM43)

hostrup, 2020-01-22:
**Describe the problem you have/What new integration you would like**

I would like to be able to control the AM43 devices. Theese are relatively cheap devices - that goes by different names eg. A-OK Blinds Controller / zenismart / am43 blinds controller. They seem to be developed based on the SOC ( VAR-SOM-AM43 ) -

**Please describe your use case for this integration and alternatives you've tried:**

I would like to be able to control them from within home-assitant - and then sending the data back as mqtt. There is an existing solution - based on nodejs allready: 
https://github.com/binsentsu/am43-ctrl
So it is allready possible - allthough it would be smarter and more powerefficient to be able to use it with the esphome boards. 

**Additional context**

buxtronix, 2020-06-05:
I wrote an Arduino IDE based BLE to MQTT gateway for these devices, running on ESP32 boards. Until native ESPHome support is written, this is an effective and cheap low-power solution using the same hardware.

https://github.com/buxtronix/am43

hostrup, 2020-06-05:
Thanks for the Tip buxtronix - will definitely try that out :)