esphome/feature-requests #585: Support `on_change` for outputs

tjschiffer, 2020-01-31:
**Describe the problem you have/What new integration you would like**

I have ESPHome on a Tuya bulb that has RGBWW channels. However, when the bulb is changed from a color to white in HA the RGB outputs stay on. This could cause the bulb to overheat as these LEDs should not all be on at once (for this model bulb). With an `on_change` trigger for the outputs, I can then turn off the other outputs to ensure they aren't all on at the same time.

**Please describe your use case for this integration and alternatives you've tried:**

I've looked through all the triggers and documentation to find a way to hook into the `esp8266_pwm` changes.

**Additional context**

StijnVdd, 2021-03-28:
I would like to have the 'on_change' on lights too.

my use case:

I have a led strip with 2 segments, let say 1 main segment and the other just a few leds.
In case my garage door (to name something) is open, the second segment is set to 'orange'. Just to indicate I need to pay attention to something.
When my garage door is close, I currently turn of the second segment, but in fact I want this segment to follow the color settings of the main segment.

With the on_change on the main led strip, I would be able to trigger a script that 'copies' the rgbw values from the main segment to the second segment.

oxan, 2021-11-27:
> This could cause the bulb to overheat as these LEDs should not all be on at once (for this model bulb).

Take a look at the [color interlock](https://esphome.io/components/light/rgbww.html#color-interlock) option, which is meant for this usecase.

tjschiffer, 2021-11-28:
@oxan Thank you, I was able to create a Custom Light at the time to address this issue. When `color_interlock` became available I switched to that.

closed 2021-11-28 by tjschiffer