esphome/feature-requests #586: Dependency wish list

onkelbeh, 2020-02-01:
Hello Otto,

first, thanks a lot (again) for your phantastic work.

**Describe the problem you have/What new integration you would like**

For running ESPHome and the Dashboard in the same environment with HomeAssistant ~0.105.0b* the following module versions would be required, these are partly already met in your dev branch:
- protobuf-3.10.1
- paho-mqtt-1.5.0
- pyyaml-5.3
- platformio-4.1.0
- colorlog-4.1.0

Would be nice if you could make a new release soon for the upcoming 0.105.0. It always feels better when a system runs with code approved by the author :-)

**Please describe your use case for this integration and alternatives you've tried:**

I am currently running [esphome-1.14.3-r5](https://github.com/onkelbeh/HomeAssistantRepository/blob/master/dev-embedded/esphome/esphome-1.14.3-r5.ebuild) including the following patch:

```bash
src_prepare() {
	sed -e 's;protobuf==3.10.0;protobuf==3.10.1;' \
		-i esphome.egg-info/requires.txt \
		-i setup.py
	sed -e 's;paho-mqtt==1.4.0;paho-mqtt==1.5.0;' \
		-i esphome.egg-info/requires.txt \
		-i setup.py
	sed -e 's;PyYAML==5.1.2;PyYAML==5.3;' \
		-i esphome.egg-info/requires.txt \
		-i setup.py
	sed -e 's;platformio==4.0.3;platformio==4.1.0;' \
		-i esphome.egg-info/requires.txt \
		-i setup.py
	sed -e 's;colorlog==4.0.2;colorlog==4.1.0;' \
		-i esphome.egg-info/requires.txt \
		-i setup.py
	eapply_user
}
```

and can confirm that everything is working (on my ~20 devices) as it should.

**Additional context**

onkelbeh, 2020-04-12:
[My Gentoo Ebuild](https://github.com/onkelbeh/HomeAssistantRepository/tree/master/dev-embedded/esphome) still can be run with HA 0.108.3 requirements with some harder deviations to the dependency check list.

Currently I use 
```
src_prepare() {
	sed -e 's;protobuf==3.10.0;protobuf==3.11.4;' \
		-i esphome.egg-info/requires.txt \
		-i setup.py
	sed -e 's;paho-mqtt==1.4.0;paho-mqtt==1.5.0;' \
		-i esphome.egg-info/requires.txt \
		-i setup.py
	sed -e 's;PyYAML==5.1.2;PyYAML==5.3.1;' \
		-i esphome.egg-info/requires.txt \
		-i setup.py
	sed -e 's;platformio==4.0.3;platformio==4.3.1;' \
		-i esphome.egg-info/requires.txt \
		-i setup.py
	sed -e 's;colorlog==4.0.2;colorlog==4.1.0;' \
		-i esphome.egg-info/requires.txt \
		-i setup.py
	eapply_user
}
```
Everything *I* use (including the dashboard) works again like a charm.

glmnet, 2020-04-15:
Ahh thanks for testing and reporting!!
In which context that code runs? 
I believe I can try modifying dev with the version changes but I'm not sure which files should be upgraded other than 

* https://github.com/esphome/esphome/blob/dev/requirements.txt 
* https://github.com/esphome/esphome/blob/dev/requirements_test.txt
* https://github.com/esphome/esphome/blob/dev/setup.py

onkelbeh, 2020-04-15:
Hi Guillermo,

Here I have several virtual machines used only for Home Assitant, all running Gentoo Linux. I am maintaining [Gentoo Overlay for Home Assitant](https://github.com/onkelbeh/HomeAssistantRepository), containing Ebuild Files for (nearly) all modules Home Assistant uses _and_ for [ESPHome](https://github.com/onkelbeh/HomeAssistantRepository/tree/master/dev-embedded/esphome) & Platformio as long as these cannot be found in [Gentoo's Main Repository](https://github.com/gentoo-mirror/gentoo).

With the help of Gentoo's package manager I can keep tight control of all installed python modules.

On my productive box Home Assistant (0.108.3, yes I know it's old, sorry, short on time during home office days) and ESPHome 1.14.3-r6 run without any virtual environments (except /etc/homeassistant/deps, which i try to keep empty) on Python 3.7.7-r1. Also I have a Box for development and testing with Python 3.7.7-r1, and another one for Python 3.8.2-r1. I wrote more about my setup in my [README.md](https://github.com/onkelbeh/HomeAssistantRepository/blob/master/README.md).

In my home about **30 ESP based devices are in productive duty**, and about 15 more are only for tests and development. Most of these run ESPHome.
Some days ago I ran into an [api problem with lots of output channels](https://github.com/esphome/issues/issues/877), where we already met, this brought me to Platformio 4.3.1, which also runs good, I intend to keep it.

Originated mostly from Home Assitant, after a big cleanup last week, I currently have the following packages installed on the production machine:
- voluptuous-0.11.7-r1
- PyYAML-5.3.1
- paho-mqtt-1.5.0-r1
- colorlog-4.1.0
- tornado-5.1.1
- protobuf-3.11.4
- tzlocal-2.0.0-r1
- pytz-2019.3
- pyserial-3.4
- ifaddr-0.1.6-r1 

As far as I know, https://github.com/esphome/esphome/blob/dev/requirements.txt is not used for setup anyway, so I took ESPHome's dependencies only from setup.py. My Ebuild for ESPHome simply patches esphome.egg-info/requires.txt after unpacking the source prior calling setup.py.

This works great for me (and a unknown number of cloners) since I started using [esphomeyaml-1.10.0](https://github.com/onkelbeh/HomeAssistantRepository/tree/6e8addea7c6b575d474f03935bdccb99cd1c77c3/dev-embedded/esphomeyaml).

:-) Thanks again a lot to you all, guys.

https://xkcd.com/1988/
[image]

onkelbeh, 2020-05-21:
In addition, made some tests with tornado-6.0.4, did not find any problems. 1.14.3 runs good with it.

onkelbeh, 2020-07-28:
Continuously getting better, currently `(1.14.5)` still open are:

```shell
src_prepare() {
	sed -e 's;tornado==5.1.1;tornado==6.0.4;' \
		-i esphome.egg-info/requires.txt \
		-i setup.py
	sed -e 's;protobuf==3.11.3;protobuf==3.11.4;' \
		-i esphome.egg-info/requires.txt \
		-i setup.py
}
```
Last update to `(1.14.5)`: https://github.com/onkelbeh/HomeAssistantRepository/commit/69b526490d696cc9911a975988223c34381fd7b1
Repo: [dev-embedded/esphome](https://github.com/onkelbeh/HomeAssistantRepository/tree/master/dev-embedded/esphome)

Thanks a lot for work.

OttoWinter, 2020-07-28:
Hi, thanks for trying to get gentoo builds.

I have a question though: Why not use python venvs? (Maybe gentoo's package manager doesn't have venv-like features?)

I understand it's a problem to get ESPHome and HA core to install on the same system without venvs because of the version conflicts.

With some packages, we could maybe relax the version pinning a bit (ifaddr, click, colorlog, protobuf, paho-mqtt) - in many other packages we really do need strict version pinning though:

 - voluptuous: We inject our own code to allow for better validation messages
 - PyYAML: We use our own Loader/Dumper - any change to PyYAML *could* potentially break it
 - tornado: AFAIK, the dashboard is incompatible with tornado v6 due to some of the changes to coroutines.
 - tzlocal/pytz: We heavily abuse internal data structures to efficiently build a tz string the ESP understands.
 - platformio: Injecting code

So for a bunch of packages it will be impossible to relax the version pinning.

onkelbeh, 2020-08-02:
Hello Otto,

you are very right, one of my main targets is to use Gentoo’s package manager, and yep, currently there’s no way to let it manage venvs. I kept HA's /etc/homeassistant/deps, but usually it is empty.

I understand that proper function is much more important than external compatibility, and version pinning makes things a lot easier. I mostly wanted to let you guys know about the versions I am using and that I did not see any problems. But I do not track the changes in the dependant libraries in detail, neither do I extensive testing, I just check if they work for my daily use.

On the other hand I believe that using the same libraries (as HA's core does) should make things easier for you, too.

Since [esphome-1.15.0_beta](https://github.com/onkelbeh/HomeAssistantRepository/blob/master/dev-embedded/esphome/esphome-1.15.0_beta2.ebuild) my list is very short anyway:
 * HA's tensorflow component has a [dep to `protobuf-3.6.1`](https://github.com/home-assistant/core/blob/dev/homeassistant/components/tensorflow/manifest.json), which I ignored for now, I assume no one here is using it.
 * check_config.py [pins `colorlog-4.1.0`](https://github.com/home-assistant/core/blob/dev/homeassistant/scripts/check_config.py) (the only thing I had to patch in 1.15.0)
 * `voluptuous`, `PyYAML`, `tzlocal`/`pytz` and `platformio` are OK now

If you could just unpin colorlog, my list would be empty, except the new protobuf-3.12.4 upgrade :-)

As far as I know, a handful users (including me) are using the dashboard with my Ebuild. So I would say that tornado-6.0.4 works well, I[ have it running](https://github.com/onkelbeh/HomeAssistantRepository/commit/d0a322226fb60d98cada5d65af7e3dab305d4298#diff-03924d0b62fe2676a17151a87bb00405) since [it went stable](https://github.com/gentoo/gentoo/commit/595d3b98e8f4134f09b4069801a7b8643c12f59a#diff-c1ea018e3e531d269e232e107cdf9aed) in Gentoo's main repo in the beginning of May (during 1.14.4).

BTW: `ifaddr` and `tornado` are loaded even if the dashboard is not used. [Without them even 'help' fails with import error](https://github.com/onkelbeh/HomeAssistantRepository/pull/19). This behavior changed somewhere around 1.14.4.

If I can help testing anything, please let me know. And: again thanks a lot for this very cool stuff.

OttoWinter, 2020-08-06:
Oh ifaddr and tornado shouldn't be loaded when dashboard is not in use (helps keep startup times low). I'll check where that import is from.

OttoWinter, 2020-08-06:
I don't see any `tornado` import when running --help:

```bash
$ python3 -X importtime -m esphome --help
import time: self [us] | cumulative | imported package
import time:      1708 |       1708 | _frozen_importlib_external
import time:       373 |        373 |   time
import time:       869 |       1241 | zipimport
import time:       227 |        227 |     _codecs
import time:      1528 |       1755 |   codecs
import time:      1353 |       1353 |   encodings.aliases
import time:      1806 |       4913 | encodings
import time:       755 |        755 | encodings.utf_8
import time:       654 |        654 | _signal
import time:       861 |        861 | encodings.latin_1
import time:       179 |        179 |     _abc
import time:       904 |       1082 |   abc
import time:      1052 |       2134 | io
import time:       209 |        209 |       _stat
import time:       959 |       1168 |     stat
import time:      3133 |       3133 |     _collections_abc
import time:       352 |        352 |       genericpath
import time:       506 |        857 |     posixpath
import time:      1616 |       6773 |   os
import time:       422 |        422 |   _sitebuiltins
import time:       177 |        177 |     _locale
import time:       306 |        482 |   _bootlocale
import time:       438 |        438 |         types
import time:       636 |        636 |         warnings
import time:       287 |       1361 |       importlib
import time:       242 |        242 |       importlib.machinery
import time:       886 |       2488 |     importlib.abc
import time:       143 |        143 |             _operator
import time:       523 |        665 |           operator
import time:       176 |        176 |           keyword
import time:        70 |         70 |             _heapq
import time:       234 |        303 |           heapq
import time:       145 |        145 |           itertools
import time:       271 |        271 |           reprlib
import time:      1092 |       1092 |           _collections
import time:      1363 |       4012 |         collections
import time:        84 |         84 |           _functools
import time:       983 |       1066 |         functools
import time:       806 |       5882 |       contextlib
import time:       414 |       6296 |     importlib.util
import time:       255 |        255 |       _weakrefset
import time:       824 |       1078 |     threading
import time:       835 |      10696 |   _virtualenv
import time:       151 |        151 |     apport_python_hook
import time:       155 |        305 |   sitecustomize
import time:      7513 |      26190 | site
import time:       379 |        379 |     weakref
import time:       529 |        908 |   pkgutil
import time:       217 |       1124 | runpy
import time:       115 |        115 | esphome
import time:       706 |        706 |     enum
import time:        74 |         74 |       _sre
import time:       277 |        277 |         sre_constants
import time:       361 |        637 |       sre_parse
import time:       317 |       1027 |     sre_compile
import time:       178 |        178 |     copyreg
import time:       483 |       2392 |   re
import time:       867 |        867 |     locale
import time:       858 |       1725 |   gettext
import time:       934 |       5050 | argparse
import time:       162 |        162 |         token
import time:       979 |       1140 |       tokenize
import time:       141 |       1281 |     linecache
import time:       271 |       1552 |   traceback
import time:       161 |        161 |   collections.abc
[... 111 lines cut ...]
                        Commands
    config              Validate the configuration and spit it out.
    compile             Read the configuration and compile a program.
    upload              Validate the configuration and upload the latest
                        binary.
    logs                Validate the configuration and show all MQTT logs.
    run                 Validate the configuration, create a binary, upload
                        it, and start MQTT logs.
    clean-mqtt          Helper to clear an MQTT topic from retain messages.
    wizard              A helpful setup wizard that will guide you through
                        setting up esphome.
    mqtt-fingerprint    Get the SSL fingerprint from a MQTT broker.
    version             Print the esphome version and exit.
    clean               Delete all temporary build files.
    dashboard           Create a simple web server for a dashboard.
    vscode              ==SUPPRESS==
    update-all          ==SUPPRESS==

optional arguments:
  -h, --help            show this help message and exit
  -v, --verbose         Enable verbose esphome logs.
  -q, --quiet           Disable all esphome logs.
  -s key value, --substitution key value
                        Add a substitution
```

onkelbeh, 2020-08-15:
Sorry for keeping you waiting, was a busy week. This is how it looks here:

```
root@g18-hasstest:/usr/portage/homeassistant/dev-embedded/esphome # python3 -X importtime -m esphome --help
import time: self [us] | cumulative | imported package
import time:       158 |        158 | zipimport
import time:       744 |        744 | _frozen_importlib_external
import time:       162 |        162 |     _codecs
import time:       751 |        913 |   codecs
import time:       623 |        623 |   encodings.aliases
import time:       923 |       2457 | encodings
import time:       334 |        334 | encodings.utf_8
import time:       194 |        194 | _signal
import time:       545 |        545 | encodings.latin_1
import time:        84 |         84 |     _abc
import time:       391 |        475 |   abc
import time:       444 |        918 | io
import time:       181 |        181 |       _stat
import time:       367 |        547 |     stat
import time:      1584 |       1584 |     _collections_abc
import time:       208 |        208 |       genericpath
import time:       521 |        728 |     posixpath
import time:      1035 |       3893 |   os
import time:       308 |        308 |   _sitebuiltins
import time:       794 |        794 |   sitecustomize
import time:       119 |        119 |   usercustomize
import time:      1404 |       6516 | site
import time:       540 |        540 |       types
import time:       489 |        489 |       warnings
import time:       280 |       1308 |     importlib
import time:       322 |       1630 |   importlib.machinery
import time:       608 |        608 |     importlib.abc
import time:        99 |         99 |           _operator
import time:      1107 |       1206 |         operator
import time:       237 |        237 |         keyword
import time:       287 |        287 |           _heapq
import time:       310 |        596 |         heapq
import time:       193 |        193 |         itertools
import time:       316 |        316 |         reprlib
import time:       219 |        219 |         _collections
import time:      1294 |       4059 |       collections
import time:        78 |         78 |         _functools
import time:      1071 |       1148 |       functools
import time:      1057 |       6263 |     contextlib
import time:       448 |       7318 |   importlib.util
import time:       307 |        307 |       _weakrefset
import time:       647 |        954 |     weakref
import time:       648 |       1602 |   pkgutil
import time:       573 |      11121 | runpy
import time:       252 |        252 | esphome
import time:       948 |        948 |     enum
import time:       180 |        180 |       _sre
import time:       433 |        433 |         sre_constants
import time:       567 |        999 |       sre_parse
import time:       493 |       1671 |     sre_compile
import time:       150 |        150 |     _locale
import time:       241 |        241 |     copyreg
import time:       938 |       3946 |   re
import time:      1141 |       1141 |     locale
import time:      1629 |       2769 |   gettext
import time:      1442 |       8157 | argparse
import time:       159 |        159 |   time
import time:       234 |        234 |         token
import time:      1314 |       1547 |       tokenize
import time:       239 |       1785 |     linecache
import time:       432 |       2217 |   traceback
import time:       282 |        282 |   collections.abc
import time:        56 |         56 |     _string
import time:      1122 |       1177 |   string
import time:       839 |        839 |   threading
[... 131 lines cut ...]
  -s key value, --substitution key value
                        Add a substitution

```
**But,** if I call it directly (without tornado installed):
```
root@g18-hasstest:/usr/portage/homeassistant/dev-embedded/esphome # esphome
Traceback (most recent call last):
  File "/usr/lib/python-exec/python3.7/esphome", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3251, in <module>
    @_call_aside
  File "/usr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3235, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3264, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 583, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 900, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 786, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'tornado==6.0.4' distribution was not found and is required by esphome
   [1]   root@g18-hasstest:/usr/portage/homeassistant/dev-embedded/esphome #
```

onkelbeh, 2020-09-12:
Yeay.
Current dev tree is even with Home Assitants dependencies!
Thanks!

nagyrobi, 2022-07-05:
Seems fixed to me. If not, please let me know and I reopen.