esphome/feature-requests #588: eBus support for Vaillant (and others) heaters

lwqcz, 2020-02-03:
**Describe the problem you have/What new integration you would like**

Please add support for eBus protocol used on various heaters/boilers like Vaillant or ProTherm brand. It needs voltage shifter as it uses the voltage level of 9-20V, but communication is pure UART.

**Please describe your use case for this integration and alternatives you've tried:**

Genuine WIFI enabled controller for ProTherm hearts is rather expensive, costs like $150 and it is closed-ware. I would much rather use something open source and much more smart, like EPSHome is.

**Additional context**

Proper implementation exists as a separate project, but it deserves to be part of the ESPHome project, to give users another usage possibility. https://github.com/slavikb/BaiMon

Weissnix4711, 2022-01-08:
Are you sure the eBus protocol is just UART?

Either way, another option may be to purchase a 'VR33' module, and use any OpenTherm compatible thermostat. Though I'm afraid OT is not implemented properly in ESPHome yet.

nagyrobi, 2022-07-07:
See https://tasmota.github.io/docs/Smart-Meter-Interface/

linuxdep, 2024-05-16:
Hi, I think an EEBUS implementation will be used a lot, because all the heat pumps will speak this in the future in order to be able to be integrated into the grid in Germany. I don't know how it is in other countries, but EEBUS is probably open source and will soon be indispensable for implementation in EMS. Unfortunately you can't really find much yet, it's probably all still in its infancy. It would be great if it were integrated into ESPhome, because that would open up a new world of communication. Here is information from the WIKI https://de.wikipedia.org/wiki/EEBUS

Translated with DeepL.com (free version)

Weissnix4711, 2024-05-16:
> Hi, I think an EEBUS implementation will be used a lot, because all the heat pumps will speak this in the future in order to be able to be integrated into the grid in Germany. I don't know how it is [...]

This is completely different from Vaillant's eBus?

Also, eebus does seem to have plenty of documentation on their site: [here](https://www.eebus.org/specifications/download-specifications/). Haven't looked through all of it though.

stheid, 2025-03-23:
seems like ebus is officially supported by homeassistant nowadays

https://www.home-assistant.io/integrations/ebusd/

However this is via network. dont know if ebus also has a wired interface like modbus.