esphome/feature-requests #589: SPI example (eg. mcp23S08)

makkie2002, 2020-02-04:
**Describe the problem you have/What new integration you would like**
The documentation on the SPI bus implementation is rather limited (https://esphome.io/api/spi_8cpp.html). I want to implement a custom sensor that is based on the SPI interface but I find it hard to use the library as a non-specialist in C++. It would be nice to start from a (nearly) working example.

**Please describe your use case for this integration and alternatives you've tried:**
I am trying to implement the SPI-based IO expander mcp23S08. The same component with an I2C interface is already implemented in esphome (called MCP23008). So implementing this with SPI should not be too difficult as the same registers can be used. This mcp23s08 drives the Prodino boards (4 relays, 4 isolated inputs): https://kmpelectronics.eu/products/prodino-esp32-ethernet-v1/.

**Additional context**

Sabesto, 2020-10-09:
Check out https://github.com/esphome/esphome/pull/1324

nagyrobi, 2022-07-07:
Ask in the Discord channel concrete questions.

closed 2022-07-07 by nagyrobi