esphome/feature-requests #591: Remote Component: Which devices/modules are supported?

kmplngj, 2020-02-04:
**Describe the problem you have/What new integration you would like**
https://esphome.io/components/remote_transmitter.html?highlight=remote

Which RF and IR Components are supported by this? I could not find the information in den documentation.

Thanks a lot.

**Please describe your use case for this integration and alternatives you've tried:**
Searching the documentation.

**Additional context**

glmnet, 2020-02-04:
there's no list of supported hardware, common sense applies here though... there are bad quality RF receivers and good ones, also an IR LED is an IR Transmitter when it's modulating at a specific frequency (e.g. 38khz) of course the modulation is done by the ESP chip, however the LED light wavelength is inherent to the LED and cannot be controlled by software.
The same logic applies to IR receivers, they do detect different carrier frequencies, e.g. 38khz, and also are usually tuned for specifics wavelengths.
The one you choose depends on the equipment you want to interface to.

kmplngj, 2020-02-04:
Thanks for the help!

closed 2020-02-04 by kmplngj