esphome/feature-requests #592: Prevent esp8266-restore-from-flash from wear-out flash

micw, 2020-02-07:
**Describe the problem you have/What new integration you would like**
Currently enabling esp8266-restore-from-flash brings a high risk of wearing out the flash. This is caused by 2 facts:
- state is always written to flash, regardless if it's changed or not (from what I've seen in preferences and light_state code, so correct me if I just did not see the code that avoids this)
- state is written to with every state update which can be _very_ often

**Please describe your use case for this integration and alternatives you've tried:**
If state has not changes (e.g. variable is "updated" to the same value as before), flash should not be written at all (saving lots of write cycles).

In many use-cases, flash-persistence (across power outage) can easily tolerate the loss of several seconds or even minutes or hours. It would make sense to allow to set-up a "flash write interval" with the following logic:
- every write to flash is defered (e.g. stored to volatile rtc memory only)
- every `flash-write-interval` seconds, check if there are changes, write only to flash if there are any

**Additional context**
Tasmota does it in a similar way described here.

OttoWinter, 2020-02-09:
> state is always written to flash, regardless if it's changed or not (f

That's not correct - if the state is the same, the ESPPreferenceObject class will filter out the write. (see esp8266_flash_dirty).

> state is written to with every state update which can be very often

For example when is it very often? I can only think of two cases, which are already handled:

 - Effects, but when effects are on saving to flash is disabled
 - Transitions, but only the "remote" values (which is the value it is fading *to*) is stored, so also not an issue.

micw, 2020-02-09:
> > state is always written to flash, regardless if it's changed or not (f
> 
> That's not correct - if the state is the same, the ESPPreferenceObject class will filter out the write. (see esp8266_flash_dirty).

Ok found it: https://github.com/esphome/esphome/blob/7a6df38515fb4bdd48db1b25c8004d1d5d755872/esphome/core/preferences.cpp#L137
Thank you for clarification, I'll edit the issue accordingly.

> For example when is it very often? I can only think of two cases, which are already handled:
>
>     * Effects, but when effects are on saving to flash is disabled
>
>     * Transitions, but only the [...]

For example I have some monochromatic lights which should store their state across power loss. I dim it via button (while button hold, dim up/down). From my understanding every state change from within the "while" loop triggers the flashing process, this is ~10 writes/s while dimming:

```
    on_press:
      - then:
        # wait a second
        - delay: 1s
        # if button is still holding
        - if: 
            condition:
            - binary_sensor.is_on: btn1
            then:
            # dim while button is holding
            - while:
                condition:
                - binary_sensor.is_on: btn1
                then:
                - light.dim_relative:
                    id: light1
                    relative_brightness: !lambda |-
                      return id(dim_direction)*0.02;
                    transition_length: 0.01s
                - delay: 0.01s
            # switch dim direction
            - lambda: id(dim_direction)=-id(dim_direction);
```

Another case is with power meter which updates e.g. every second. If the cumulative value (kWh) is persisted to keep it after power loss, this can potentially be updated every second (#461).

natp13, 2020-05-12:
I was actually about to file this one. I'm working on a change, but it turns out there are a lot of components that write state that I will never be able to actually test. It feels like instead of turning on "save to flash" for everything, restore_mode: restore_from_flash should be an option for each individual component. Typically there would be one component that needs to store its state (like is the heat/Ac on/off) and everything else doesn't matter (sensors for temperature, humidity, etc)

natp13, 2020-05-12:
It would be nice to unify restore, restore_value, and restore_state and apply to every "stateful" component the following:
    restore_initial_value: (true/false, float value, enum value, template value, etc)
    restore_mode: (ALWAYS_INITIAL_VALUE, DEFAULT_RESTORE, RESTORE_FROM_FLASH)

Names need some more thought, but the idea would be that every component with a state always has a default initial state, and can specify how it wants to try to restore the initial state (or just use the default). And whether or not you set esp8266_restore_from_flash, you can always control each component individually.

OttoWinter, 2021-10-01:
Closed via https://github.com/esphome/esphome/pull/2119

closed 2021-10-01 by OttoWinter