esphome/feature-requests #594: Tare functionality for HX711

nphil, 2020-02-07:
**Describe the problem you have/What new integration you would like**
I'd like there to be some kind of "tare" functionality for the HX711 component. My load cells drift over time, so the "zero" never corresponds to the same readout.
**Please describe your use case for this integration and alternatives you've tried:**
Unsure if automatons within esphome config would achieve the same, please let me know and I'll close the issue.
**Additional context**

OttoWinter, 2020-02-09:
Yes, automations can achieve this. Have a global variable of type `float` defined in your config, and set that to the current reading of the HX711 in a template switch.

In the HX711 sensor, add a lambda filter that subtracts the global variable, and voila.

(Will close when someone posts a YAML config as an example here)

glmnet, 2020-02-10:
now I see why my scale works like it works:
Display is off, you have to step on it but then leave it unloaded, then it tares itself, then you can go on it and check how fat you are.

nphil, 2020-02-10:
> now I see why my scale works like it works:
> Display is off, you have to step on it but then leave it unloaded, then it tares itself, then you can go on it and check how fat you are.

Actually, this is what I'm hoping to accomplish with an automation. If there's a really _significant_ reduction in weight reading, it should tare to Zero. I use these as bed occupancy sensors, so if there's a pretty significant change in weight reading, it means I'm off the bed.

Multifunk-maker, 2020-04-02:
Anyone who has a working example for this?

definitio, 2020-04-02:
@Multifunk-maker 
```yaml
globals:
  - id: constant_weight
    type: float

sensor:
  - platform: hx711
    id: bed_raw
    dout_pin: 14
    clk_pin: 12
    gain: 128
    update_interval: 5s
    filters:
      - calibrate_linear:
          ...
    on_value:
      - if:
          condition:
            - lambda: |-
                static float last = 0.0;
                float diff = x - last;
                last = x;
                return x < 10.0 && diff < 1.0 && diff > - 1.0;
          then:
            - lambda: 'id(constant_weight) = x;'

  - platform: template
    name: $room bed weight
    state_topic: ${room}/bed/weight
    icon: mdi:weight-kilogram
    unit_of_measurement: kg
    lambda: 'return id(bed_raw).state;'
    update_interval: 5s
    filters:
      - lambda: 'return x - id(constant_weight);'
      - lambda: 'return x >= 10.0 ? x : 0;'
      - delta: 0.5
```
It subtracts weight up to 10 kg if difference between current and last readings is less then 1 kg. Sensor reports only values equal or greater than 10 kg.

Multifunk-maker, 2020-04-02:
@definitio 

Thank you very much, this was just what i needed.