esphome/feature-requests #597: Esp8266: Enable hardware uart for custom component

edge90, 2020-02-09:
**Describe the problem you have/What new integration you would like**

So, a couple of other github users and I reverse engineered  the `shelly dimmer` and I'm now researching the idea of getting it to work with esphome. The two mcus communicate via uart which means uart logging must be disabled (or use the secondary tx port which is hard to get to without soldering). I'm thinking custom uart component but is it possible to enable hardware synced uart? I read that the logging implementation occupies the hardware already. Is it enough to set baud_rate to 0?

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

glmnet, 2020-02-10:
> Is it enough to set baud_rate to 0?

It **IS** the official way to disable the logger on the uart.

You could also just don't set up the `logger` component but then of course it won't be available via  wifi either.

closed 2020-02-10 by glmnet

edge90, 2020-02-10:
@glmnet awesome, cheers!

brandond, 2020-02-10:
@edge90 see https://github.com/esphome/esphome/tree/dev/esphome/components/tuya which implements core parts of Tuya's UART protocol for paired MCU dimmers and other such devices.

edge90, 2020-02-11:
@brandond thanks :)