esphome/feature-requests #598: Power supply manual switch

billdotjr, 2020-02-10:
**Describe the problem you have/What new integration you would like**

Power supply component should have means of powering on manually

**Please describe your use case for this integration and alternatives you've tried:**

I have electronics lab, where I have 750W 12V power supply from an old server. I use it for powering LED strips in the lab, but I am using it as 12V sypply for projects as well. Currently I have configured a switch with power supply assigned to it, but this is eating up one pin on the ESP. I would like to have either:

1) A dummy switch, where I can assign power supply, but it would not require me to configure any pin.

or

2) Manually switchable power supply component.

**Additional context**

glmnet, 2020-02-10:
> 1. A dummy switch, where I can assign power supply, but it would not require me to configure any pin.

try this

```` 
# Example configuration entry
output:
  - platform: gpio
    pin: 25 # A pin you are not using, or the pin of the PS 
    id: 'generic_out'
    power_supply: power_supply_id
switch:
  - platform: output
    name: "PS On"
    output: 'generic_out'

````

billdotjr, 2020-02-10:
This is exactly what I am doing right now - using an unused GPIO pin. But I have pin intensive application and do not want to use port expander just to overcome software limitation.

Using the same pin as the power supply might actually work, but I do not see it as very good practice. I would expect the config parser to reject it,

glmnet, 2020-02-10:
It won't reject it.

A less hacky yet more complicated solution will be creating a custom output which does nothing, look at the documentation of it https://esphome.io/components/output/custom.html

Basically you follow this and leave the `setup` and `write_state` methods empty.

````

// Custom binary output, for exposing binary states
class DummyOutput : public Component, public BinaryOutput {
 public:
  void setup() override {
  }

  void write_state(bool state) override {
  }
};
````

closed 2022-07-06 by nagyrobi