esphome/feature-requests #600: Warn name less components when internal is not set to true

glmnet, 2020-02-13:
**Describe the problem you have/What new integration you would like**
People often creates components, they set them an `id` but they set no `name` then the component does not show up on HA nor mqtt and they don't know why.
ESPHome implicitly sets those components as `internal: true`

I suggest:
* make `name` optional only if internal is set to true, or
* write a warning in the compiler if the name is not set and the component is not marked explicitly as internal

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**
I've seen this issue in several places already.

https://github.com/esphome/esphome-docs/pull/486

brandond, 2020-03-12:
I think Otto specifically wanted to make internal implicit if things are unnamed. If anything I think we need a docs update rather than compile time warnings.

glmnet, 2020-04-25:
Another option would be using id suffixed with ‘_’ like we do in code. 
Those won’t get the warning if internal true is not explicit

davet2001, 2021-11-24:
I think a good solution is to make name compulsory if `internal` is set to false or undefined.  Then we could add to the docs:
```
**name** (**Required** unless internal=true, string): The name of the sensor.
```
That way we prevent unnamed sensors.

Does the automatic 'unnamed implies internal' help besides making yaml files shorter?

Currently the documentation is suggesting that name is compulsory, meaning people (inc me) are creating external entities, then adding the internal=true to stop them from filling up HA with unnecessary entities.  The doc should not say it's compulsory when it isn't.

nagyrobi, 2022-07-06:
It's specified in the [docs](https://esphome.io/components/sensor/index.html) at the `internal` configuration parameter:

> - **id** (Optional, string): Manually specify the ID for code generation. At least one of id and name must be specified.
> - **name** (Optional, string): The name for the sensor. At least one of id and name must be specified.
> - **internal** (Optional, boolean): Mark this component as internal. _Internal components will not be exposed to the frontend (like Home Assistant). Only specifying an id without a name will implicitly set this to true._

davet2001, 2022-07-06:
Yes, the [documentation was updated](https://github.com/esphome/esphome-docs/commit/c0c3a4c6c97d4588da6c550e4cfa6c906c8c1198) making this clearer.

nagyrobi, 2022-07-06:
Indeed, but this feature request was not closed at that time.

glmnet, 2022-07-07:
I as the OP would highly appreciate @jesserockz  and @ssieb  input in this issue. I'm ok of it being closed, but they who are constantly answering questions at discord might really be aware weather this is useful or not.

glmnet, 2022-07-07:
Btw I can really implement this myself. This is more a discussion which was kept open more than a request. I didn't implemented it because  I didn't receive positive feedback.

ssieb, 2022-07-07:
It does happen sometimes, but not that often.  I would personally find it really annoying to have this warning.
A general documentation page with tips like this would be useful.  Or making sure every component describes the internal state.  I think there are some that even describe it wrongly.

jesserockz, 2022-07-07:
I agree that this is something to be documented. 
I don't think forcing `internal: true` on everyones configs is the right approach.

nagyrobi, 2022-07-07:
@jesserockz do you think that @glmnet's original PR against the docs https://github.com/esphome/esphome-docs/pull/1645 fulfills your requirement?

wtadler, 2024-10-12:
I was also confused by the wording in the docs here. I incorrectly thought that:
1. If I specified an `id` with no `name`, `internal` would be implicitly set to be `true`.
2. If I were to explicitly set `internal: false`, that would override (1).

(2) is apparently incorrect, but that's not necessarily made clear in the docs (or at least it wasn't clear to me). I think it should probably be made clear in the docs!