esphome/feature-requests #606: Pull-style OTAs

sowbug, 2020-02-21:
**Describe the problem you have/What new integration you would like**

I have some ESPHome devices on guest networks configured with client isolation enabled. This means that it's not possible to OTA them, as nobody else on the network can reach the OTA port. ESPHome uses a "push"-style OTA where the central maintainer of OTA binaries initiates connections with individual devices and then pushes OTAs to them. It would be nice to have a "pull"-style option, where the devices periodically check the central service for new updates, and then grab the binaries from it.

Not only would this reduce the need for the central service to know as much about individual devices (particularly their IP addresses), but it would also allow more sophisticated users to set up static web servers holding the OTA binaries, which would scale very nicely. The "upload" operation for ESPHome would generate the OTA image and then upload it to the web server, and then the individual devices would poll the server, or subscribe to an MQTT topic asking them to check the OTA server right away. Finally, this would look more like a traditional web- or web-service network topology, where firewalls allow outgoing connections (and associated replies), but don't allow incoming connections.

The configuration might look like this:

```
ota:
  ota_server_url: https://example.com/esphomeota/v1
  check_frequency: 100h
```

The HTTPS request might include a few bits of identifying information, either in POST form-encoded format, or as GET query params, or even as macros in the URL. This would enable the server to serve different OTAs to different devices.

I'm pretty sure this isn't possible today. I wish it were.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

jperquin, 2020-12-08:
This would be very high on my wish list as well..

wizzor, 2022-05-11:
There are similar requests with OTA initiation via MQTT, but what many people seem to want is the capability to publish ota images on a server. I'm commenting on this for visibility: I want it too!

nagyrobi, 2022-07-07:
Duplicate of https://github.com/esphome/feature-requests/issues/801

closed 2022-07-07 by nagyrobi