esphome/feature-requests #607: Sensor Filter: Ignore the first values after ESP boot

thEngal, 2020-02-21:
**Describe the problem you have/What new integration you would like**
I need a filter that ignores incoming values for a specified time after ESP bootup. 

**Please describe your use case for this integration and alternatives you've tried:**
I want to add a gas sensor (MQ-135) which needs a specified time to heat up. With this filter, the heat up values wouldn’t passed forward to Home Assistant.

**Additional context**
[image]

marklister, 2020-02-28:
One could probably accomplish this with a simple lambda filter looking at sensor.uptime...

CarlosGS, 2020-08-08:
I am missing this too ;(

When using `sliding_window_moving_average` and `send_first_at` to guarantee a filtered value, the first odd reading (right after rebooting) introduces a distorted sample, which is annoying.
[image]

 A parameter like `n_skip: 1` or `n_ignore: 1` would be awesome IMHO.

closed 2022-07-06 by nagyrobi