esphome/feature-requests #608: Support for ssd1327 128x128 OLED displays

igg, 2020-02-21:
**Describe the problem you have/What new integration you would like**

Support for SSD1327-based OLED displays, typically 128x128. Support should include I2c in addition to SPI.

**Please describe your use case for this integration and alternatives you've tried:**

Use case is for a multi-sensor display like temperature/pressure/humidity/motion, which would be common for a thermostat, for e.g. A local display for this sensor package, including a chart for barometric trends, benefits from a large display.

**Additional context**

This should be quite similar to ssd1325, but needs I2c support, which is a common requirement in cases where sensors are already on the i2c bus. Additional GPIO use like relay drive and motion input can easily preclude the use of SPI in addition to I2c in common GPIO-limited packages like the esp8266.

igg, 2020-02-22:
Instead of defining a new device, this pull request esphome/esphome#985 "patches" the existing ssd1325 device, and adds a `SSD1327 128x128` model.  Still missing: I2c support.

kbx81, 2021-07-17:
See https://github.com/esphome/esphome/pull/1406

closed 2021-07-17 by kbx81