esphome/feature-requests #609: Support AZ-Delivery ESP-32 Dev Kit C V4

kaikuehne, 2020-02-22:
**Describe the problem you have/What new integration you would like**
Please add support for [AZ-Delivery ESP-32 Dev Kit C V4](https://www.az-delivery.com/products/esp-32-dev-kit-c-v4). To get this device up and running, I followed the description on the [esphome documentation](https://esphome.io/devices/esp32.html). For the `board` option, it refers to platformio's documentation to get the actual `board ID` -- which according to platformio's docs -- would be "[az-delivery-devkit-v4](https://docs.platformio.org/en/latest/boards/espressif32/az-delivery-devkit-v4.html#board-espressif32-az-delivery-devkit-v4)". The ID is listed when using `platformio` CLI locally:

```sh
$ platformio boards | grep -B6 az-delivery-devkit-v4                                         ~/Desktop
Platform: espressif32
================================================================================
ID                               MCU    Frequency    Flash    RAM     Name
-------------------------------  -----  -----------  -------  ------  --------------------------------
...
az-delivery-devkit-v4            ESP32  240MHz       16MB     520KB   AZ-Delivery ESP-32 Dev Kit C V4
...
```

**Please describe your use case for this integration and alternatives you've tried:**
I want to use the device with ESPHome, which currently doesn't work.
This is the configuration I'm using:

```yaml
esphome:
  name: esptest
  platform: ESP32
  board: az-delivery-devkit-v4
```

Running `esphome <file.yaml> run` exits with this error though:

```sh
Failed config

esphome: [source fileyaml:2]
  name: esptest
  platform: ESP32

  Could not find board 'az-delivery-devkit-v4'. Valid boards are [a lot, excpect "az-delivery-devkit-v4"].
```

**Additional context**
The board ID is missing in (at least!) esphome/pins.py. If I override `ESP32_BOARD_PINS` in pins.py to include the ID, I'm getting a new error mesage:

```sh
# Running with manually modified esphome/pins.py.
INFO Reading configuration esp32.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Deleting esptest/.pioenvs
INFO Deleting esptest/.piolibdeps
INFO Compiling app...
INFO Running:  platformio run -d esptest
Processing esptest (board: az-delivery-devkit-v4; framework: arduino; platform: espressif32@1.11.0)
---------------------------------------------------------------------------------------------------------
**Error: Unknown board ID 'az-delivery-devkit-v4'**
```

Unfortunately, I haven't found the source of this error message.

Right now, I'm experimenting with using `esp32dev`, `nodemcu-32s` and other similar ones to make this work in the meantime. Please note that **az-delivery-devkit-v4** has more RAM and a bigger Flash than the other boards, so I think it is reasonable to add this as a new device / board ID.

Let me know if I can provide more information and/or help with testing the real device.
Thanks!

thomasrehm, 2020-04-04:
hi @kaikuehne  did you find a board id like `esp32dev` or `nodemcu-32s` to get it running?

kaikuehne, 2020-04-10:
Hello @thomasrehm, I'm using `nodemcu-32s` which works really well, because my configuration is pretty short. Let me know if I can provide info/help to support the `az-delivery-devkit-v4` board to make use of its additional resources.

estebanz01, 2020-05-07:
Hello! I'm finding this problem too. I cannot specify `az-delivery-devkit-v4` as a board. I'll test with `esp32dev` for a BLE device tracker.

estebanz01, 2020-05-07:
It works fine with `esp32dev`, at least for the BLE device tracker proof of concept I built:
```yaml
$ cat ble_tracker.yaml 
esphome:
  name: esp32tracker
  platform: ESP32
  board: esp32dev

wifi:
  ssid: "Linksys ssid home"
  password: "A super secure password."

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32Tracker Fallback Hotspot"
    password: "Blah"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "Fibonacci"

ota:
  password: "Fibonacci"

esp32_ble_tracker:

# This tells the ESP32 what to look
binary_sensor:
  - platform: ble_presence
    mac_address: mac_address: AA:BB:CC:DD:EE:FF
    name: 'BLE Tracker'

sensor:
  - platform: ble_rssi
    mac_address: AA:BB:CC:DD:EE:FF
    name: 'BLE rssi'

deep_sleep:
  run_duration: 1min
  sleep_duration: 3min
```

TRon993, 2020-10-25:
Had the same problem but found a solution for that:

In your device yaml file in esphome make use of
```
...
esphome:
  name: $devicename
  platform: ESP32
  board: esp32dev
  platformio_options: 
    platform: espressif32
    board: az-delivery-devkit-v4
    board_build.mcu: esp32
    board_build.f_cpu: 240000000L
...
```
and it will compile just smoothly using the right board in the background (not the generic esp32dev board as given in example) :)

Taxy7, 2020-11-12:
Hi TRon993,

can you please point me what "device yaml file in esphome"  do you mean? where can I find that file? Do you mean new created file from new device? 

thanks Mark

TRon993, 2020-11-12:
Hi Mark,

Sure, I'm referring to the yaml file that esphome is creating when you run through the setup wizzard.

Snippet from one of my devices using this esp32:
```
substitutions:
  devicename: bad
  upper_devicename: Bad
  bme280_refresh: 60s

esphome:
  name: $devicename
  platform: ESP32
  board: esp32dev
  platformio_options: 
    platform: espressif32
    board: az-delivery-devkit-v4
    board_build.mcu: esp32
    board_build.f_cpu: 240000000L
  
wifi:
  networks:
   - ssid: !secret wifi0_ssid
     password: !secret wifi0_password
     priority: 20
...
```

HTH,
Tron

Taxy7, 2020-11-12:
thank U Tron 👍 ))))

phiwa99, 2020-12-18:
hello
the code

`esphome:
  name: $devicename
  platform: ESP32
  board: esp32dev
  platformio_options: 
    platform: espressif32
    board: az-delivery-devkit-v4
    board_build.mcu: esp32
    board_build.f_cpu: 240000000L`

works, but the espflasher does not use the external ram. 

I need the space for my code. Does anybody have an idea?

the log:
`sing 'COM4' as serial port.
Connecting........_
Detecting chip type... ESP32
Connecting.....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 10:52:1C:62:8F:E8
Uploading stub...
Running stub...`

neilpdj, 2021-08-10:
> Hi Mark,
>
> Sure, I'm referring to the yaml file that esphome is creating when you run through the setup wizzard.
>
> Snippet from one of my devices using this esp32:
>
> ```
> substitutions:
>   devicename: [...]

Hi @TRon993, I've given this a go but am getting an error resolving the IP address of the board. I'm wondering if this is likely to do with the fact I have previously flashed it to micropython (I see it mentions arduino framework) or if I've maybe just completely missed a step somewhere else?!

zbrozek, 2021-08-28:
> esphome:
>   name: $devicename
>   platform: ESP32
>   board: esp32dev
>   platformio_options: 
>     platform: espressif32
>     board: az-delivery-devkit-v4
>     board_build.mcu: esp32
>     board_build.f_cpu: 240000000L

This workaround is functional, but it seems less-than-deluxe not to support it directly in esphome's `board` field, especially given that the [documentation](https://esphome.io/components/esphome.html) suggests using platformio's board fields directly.

Any word on what it would take to support *all* of platformio's targets programmatically?

closed 2022-07-01 by nagyrobi

bf8392, 2022-09-29:
I have issues with the board...I try to switch a relais, but at some operations, it keeps switching randomly...what can be the cause of this?

ssieb, 2022-09-29:
> I have issues with the board...I try to switch a relais, but at some operations, it keeps switching randomly...what can be the cause of this?

This is not the place for support requests.  Either try Discord or one of the forums.

bf8392, 2022-09-30:
Couldn't it be an issue with the esphome software? Because the switch is snapping back when pushed...

mrtwnklr, 2022-11-04:
Maybe a timesaver for other users:

I used the above mentioned configuration for more than 2 years...
and I recently run into the following error when compiling a project on a new system:

```
[...]
data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/xtensa-esp32-elf/no-rtti/bits/gthr-default.h:781:12:
error: 'pthread_mutex_timedlock' was not declared in this scope
     return __gthrw_(pthread_mutex_timedlock) (__mutex, __abs_timeout);
            ^~~~~~~~
[...]
```

#### tl;dr;

Pin the version of platform `espressif32` to max `3.5.0`, as each higher version leads to this compiler error.

For example:

```yaml
esphome:
  name: ${device_name}
  platform: esp32
  # tool esphome does not support board value 'az-delivery-devkit-v4' directly,
  # see also: https://github.com/esphome/feature-requests/issues/609
  board: esp32dev
  platformio_options:
    platform: espressif32@3.5.0
    board: az-delivery-devkit-v4
```

#### ntl;dr;

`~/.platformio` directory does not exist on the new system. So latest version for platform espressif32 was downloaded. In my case: `espressif32@5.2.0`.

Hunting showed the error occured with versions including and above `espressif32@4.0.0`.

Alternative: define `platformio/framework-arduinoespressif32 @ ~3.20005.0` as `platform_package` (latest for now; default seems to be `@ ~3.10006.0`). This fixes the compiler error too. But I am not able to evaluate possible side effects. So I'll go the `@3.5.0`-route.

Just as a reference:

```yaml
esphome:
  name: ${device_name}
  platform: esp32
  # tool esphome does not support board value 'az-delivery-devkit-v4' directly,
  # see also: https://github.com/esphome/feature-requests/issues/609
  board: esp32dev
  platformio_options:
    platform: espressif32@5.2.0
    platform_packages:
      - platformio/framework-arduinoespressif32 @ ~3.20005.0
    board: az-delivery-devkit-v4
```