esphome/feature-requests #610: Add feature to allow the ESPHome HA plugin to discover existing devices

cjramseyer, 2020-02-23:
**Describe the problem you have/What new integration you would like**
Had to reload/rebuild my HA from scratch.  Installing ESPHome from scratch also.  This reaults in the ESPHome components that I have able to be added to HA as they are configured currently, but they are NOT known by ESPHome.  So OTA updates, changes, mean physically connecting them and reflashing from scratch.

**Please describe your use case for this integration and alternatives you've tried:**
Please add a config discovery function with the ESPHome addin, to allow a brand new instance of ESPHome to load the current configuration.

**Additional context**
I have existing sensors that are NodeMCU ESP8266 based, and already configure for OTA, this would allow ESPHome to contact/scan an existing device and determine its current configuration and build the ESPHome .yaml file.

Misiu, 2020-03-02:
@cjramseyer I think it is possible to scan the network for ESPHome devices, but building the `.yaml` file will be very hard if impossible.

randybb, 2020-03-02:
As is mentioned in [FAQ](https://esphome.io/guides/faq.html#can-configuration-files-be-recovered-from-the-device)) _yaml_ is not stored on devices, just compiled code, so this is not possible and never will be. Theoretically you can store a file to ESP and then pull it, but due security concerns and as this is normally not possible on any MCUs I don't think this will be ever implemented. Definitely not possible just by scanning. Just keep your _yaml_ file safe.

closed 2022-07-07 by nagyrobi

nagyrobi, 2022-07-07:
Also see https://github.com/esphome/feature-requests/issues/1507