esphome/feature-requests #611: New Device Sonoff D1 Dimmer

robbz23, 2020-02-24:
**Describe the problem you have/What new integration you would like**

Sonoff has released a new device called the D1 which is a mains dimmer using a ESP8285.  With the help of others, we have been able to determine that the dimming functions are carried out by an MCU and this MCU talks to the ESP over the serial bus.  By sending a command to the MCU over the serial bus I can control the dimmer functions, on, off, and set brightness have been sniffed out.  Also in normal function from Sonoff, is a an optional 433mhz remote that communicates with the MCU directly.  When changes are made with the remote, it is communicated over serial to the ESP chip.  

**Please describe your use case for this integration and alternatives you've tried:**

To be able to use ESP home with this new device and to be able to control brightness of all lights connected to it.

**Additional context**

I have already flashed it with Tasmota and have used the console feature in Tasmota to send and receive commands over UART.  Here is a sample of that 
```
00:17:59 CMD: Baudrate 9600
00:17:59 SER: Set to 8N1 9600 bit/s
00:17:59 RSL: stat/tasmota_D9E56D/RESULT = {"Baudrate":9600}
00:25:32 CMD: SerialSend5 aa550104000a0122ffffffffffffffff29
00:25:32 RSL: stat/tasmota_D9E56D/RESULT = {"SerialSend":"Done"}
00:26:35 CMD: SerialSend5 aa550104000a0122ffffffffffffffff2a
00:26:35 RSL: stat/tasmota_D9E56D/RESULT = {"SerialSend":"Done"}
00:26:35 RSL: tele/tasmota_D9E56D/RESULT = {"SerialReceived":AA550104000005}
00:28:58 CMD:  SerialSend5 aa550104000a0101ffffffffffffffff09
00:28:58 RSL: stat/tasmota_D9E56D/RESULT = {"SerialSend":"Done"}
00:28:58 RSL: tele/tasmota_D9E56D/RESULT = {"SerialReceived":AA550104000005}
00:29:12 RSL: tele/tasmota_D9E56D/RESULT = {"SerialReceived":AA550104000A013CFFFFFFFFFFFFFFFF44}
00:29:43 RSL: tele/tasmota_D9E56D/RESULT = {"SerialReceived":AA550104000A0101FFFFFFFFFFFFFFFF09}
00:29:53 RSL: tele/tasmota_D9E56D/RESULT = {"SerialReceived":AA550104000A0164FFFFFFFFFFFFFFFF6C}
00:30:02 RSL: tele/tasmota_D9E56D/RESULT = {"SerialReceived":AA550104000AFF1EFFFFFFFFFFFFFFFF24}
```
Over at [tasmota's issue](https://github.com/arendst/Tasmota/issues/7598) have we determined the message format to tell the MCU to turn on and off the light.  

The message aa550104000a0122ffffffffffffffff29 breaks down like this

aa550104000a -- header
01 -- on or 00 for off
22 -- on/off toggle flag 
ffffffffffffffff -- footer 
29 -- checksum.  It is a  Checksum 8 Module 256 so you add up each byte and take the module with 256.  However, you must add a byte of 01 to your sum before the module.  This is verified with about 8 messages.

AA550104000A013CFFFFFFFFFFFFFFFF44 -- 60%
AA550104000A0101FFFFFFFFFFFFFFFF09 -- 1%
AA550104000A0164FFFFFFFFFFFFFFFF6C -- 100%

In the logs above you can see where I sent commands to the ESP using tasmota and serial send.  Later you can see where I am adjusting the light with the remote and Tasmota is receiving the commands from the MCU.

epiniguin, 2020-03-08:
There is a working [Sonoff D1 Tasmota driver](https://github.com/arendst/Tasmota/blob/2d4a6a29ebc7153dbe2717e3615574ac1c84ba1d/tasmota/xdrv_37_sonoff_d1.ino)

Drew765, 2020-03-11:
Would be nice to get it into ESPHome but so it can be easier to integrate and more flexible.

h0jeZvgoxFepBQ2C, 2020-04-09:
Anyone working on this? would be really nice..

craigueh, 2020-09-13:
Any progress on this? Tasmota have had full control of this device for a while now with dimming and RF support but esphome is not catching up. I hate to say it but I'm considering going back to Tasmota because esphome can't support recent devices I've bought like this D1 and the Shelly i3.

craigueh, 2020-09-13:
Ok, so I've done a bit of playing for a few hours and found a way to bypass the MCU and decode RF signals directly. You need to connect a wire from the top right corner pin on the 590R RF chip to solder pad 4 (gpio4). 
Then use the following core in esphome to receive the RF codes and map to buttons:

remote_receiver:
  pin: 4
  dump: rc_switch
#filter: 200us
  filter: 4us
  idle: 4ms
  tolerance: 50

binary_sensor:
  - platform: status
    name: "Sonoff D1 Status"

  - platform: remote_receiver
    name: "Sonoff Remote Button 1"
    rc_switch_raw:
      code: '100010101110101111101000'
      protocol: 1
    filters:
      delayed_off: 1s

  - platform: remote_receiver
    name: "Sonoff Remote Button 2"
    rc_switch_raw:
      code: '100010101110101111101100'
      protocol: 1
    filters:
      delayed_off: 1s

  - platform: remote_receiver
    name: "Sonoff Remote Button 3"
    rc_switch_raw:
      code: '100010101110101111100100'
      protocol: 1
    filters:
      delayed_off: 1s

  - platform: remote_receiver
    name: "Sonoff Remote Button 4"
    rc_switch_raw:
      code: '100010101110101111101001'
      protocol: 1
    filters:
      delayed_off: 1s

  - platform: remote_receiver
    name: "Sonoff Remote Button 5"
    rc_switch_raw:
      code: '100010101110101111100010'
      protocol: 1
    filters:
      delayed_off: 1s

  - platform: remote_receiver
    name: "Sonoff Remote Button 6"
    rc_switch_raw:
      code: '100010101110101111100101'
      protocol: 1
    filters:
      delayed_off: 1s

  - platform: remote_receiver
    name: "Sonoff Remote Button 7"
    rc_switch_raw:
      code: '100010101110101111100001'
      protocol: 1
    filters:
      delayed_off: 1s
  - platform: remote_receiver
    name: "Sonoff Remote Button 8"
    rc_switch_raw:
      code: '100010101110101111100011'
      protocol: 1
    filters:
      delayed_off: 1s

robbz23, 2020-09-14:
@craigueh That's amazing because it is the missing link that Tasmota can't do now.  And it opens up the D1 to be so interesting as a dimmer with 8 controllable buttons.  

Here is a [link](https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_37_sonoff_d1.ino) to the implementation in Tasmota that works perfectly.  Basically the esp chip needs to talk to the RF chip via uart .  The messages are formatted like I mentioned in my first comment here.  I would like to help but I really have trouble understanding how to code this in Python.

It may be possible to do this without even creating a new component but by using the UART component.

https://esphome.io/components/uart.html

craigueh, 2020-09-14:
Yeah it's pretty useful to be able to decode the RF stuff straight on the ESP. I tried out Tasmota and the delay of 1-2s after pressing a button was horrible. I found that button presses were sometimes missed altogether which was annoying. At least this way the response is instantaneous and the extra buttons can be used for other functions, or alternatively re-use the Sonoff remote elsewhere and use any RF remote for the dim up-down function.

I'm tinkering with the serial to see if I can write to manually control the dimmer as a star:

`# Enable logging
logger:
  baud_rate: 0

uart:
  baud_rate: 9600 
  tx_pin: GPIO1
  rx_pin: GPIO3

switch:
  - platform: template
    name: 'Sonoff Dimmer'
    id: dimmer
    turn_on_action:
      - uart.write: [0xA0, 0x01, 0x01, 0xA2]
    turn_off_action:
      - uart.write: [0xA0, 0x01, 0x00, 0xA1]
    optimistic: true
    `

I'm not sure how to convert the strings like AA550104000A013CFFFFFFFFFFFFFFFF44 to something like 0xA0, 0x01, 0x01, 0xA2
Any ideas?

I've also just stumbled across this post - https://community.home-assistant.io/t/esphome-custom-component-any-way-to-publish-a-light-state-manually/159757
This guy cleverly wrote a custom component to control another dimmer, perhaps we could modify his code on github to be used here for dimmer control?

craigueh, 2020-09-14:
Ok so this works to write to the uart and turn on/off the dimmer:

`switch:
  - platform: template
    name: 'Sonoff Dimmer Switch'
    id: dimmer
    turn_on_action:
      - uart.write: [0xAA, 0x55, 0x01, 0x04, 0x00, 0x0A, 0x01, 0x64, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x6C]
    turn_off_action:
      - uart.write: [0xAA, 0x55, 0x01, 0x04, 0x00, 0x0A, 0x00, 0x64, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x6B]
    optimistic: true`

I tried some other values to set the dimmer at 40% and 60% and it works. It should now be possible to set the buttons turn_on action to write to uart and set the dimmer at the desired level.

craigueh, 2020-09-14:
Here we go:

binary_sensor:
  - platform: status
    name: "Snoff D1 Status"

  - platform: remote_receiver
    name: "Sonoff Remote Button 1"
    rc_switch_raw:
      code: '100010101110101111101000'
      protocol: 1
    filters:
      delayed_off: 500ms
    on_press:
      then:
        - uart.write: [0xAA, 0x55, 0x01, 0x04, 0x00, 0x0A, 0x01, 0x64, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x6C]

  - platform: remote_receiver
    name: "Sonoff Remote Button 2"
    rc_switch_raw:
      code: '100010101110101111101100'
      protocol: 1
    filters:
      delayed_off: 500ms
    on_press:
      then:
        - uart.write: [0xAA, 0x55, 0x01, 0x04, 0x00, 0x0A, 0x00, 0x64, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x6B]

This will now turn on and off the dimmer with button 1 and 2 on the remote as per the original functionality.
We can do the same for the other buttons at 1%, 40%, 60% and 100% or whatever increments we choose.

If I was clever enough I could probably write a lambda to present a slider or something that Homeassistant can use to increment the dimmer brightness to any custom value but we'll have to see if time permits.
Ideally if someone had enough python skills they could write a custom light component to present the dimmer 1% to 100% as any other light with variable brightness. Maybe down the line we'll get lucky!!

robbz23, 2020-09-14:
My plan was to try to find another component that also uses UART as a basis for the code.  I think the Tuya stuff does but I have trouble understanding the code.  Sending codes out on UART is only half the battle.  You also have to listen to the responses on the ESP chip when the other chip responds that it has a new state.  IE if you pair the remote and use the remote buttons to update the dimming level you want to report this back to the ESP chip so it can update the state in HA.

craigueh, 2020-09-14:
That’s the reason I bypassed the mcu and unpaired the remote. Now all states are controlled by esphome so regardless of whether a button is pressed on the remote or a switch is triggered in HA it’s esphome that controls and updates the state. 

Esphome could have a template switch that updates it’s brightness depending on which button is pressed so HA is always aware.

craigueh, 2020-09-14:
An alternative to the hardware modification is to go back to using the MCU with a paired remote and the custom UART Text sensor - https://esphome.io/components/text_sensor/uart.html

With this method I was able to get the following in the logs when I pressed the on button:

[16:14:37][VV][uart_esp8266:132]:     Read 0b10101010 (0xAA)
[16:14:37][VV][uart_esp8266:132]:     Read 0b01010101 (0x55)
[16:14:38][VV][uart_esp8266:132]:     Read 0b00000001 (0x01)
[16:14:38][VV][uart_esp8266:132]:     Read 0b00000100 (0x04)
[16:14:38][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[16:14:38][VV][uart_esp8266:132]:     Read 0b00001010 (0x0A)
[16:14:38][VV][uart_esp8266:132]:     Read 0b00000001 (0x01)
[16:14:38][VV][uart_esp8266:132]:     Read 0b01000110 (0x46)
[16:14:38][VV][uart_esp8266:132]:     Read 0b11111111 (0xFF)
[16:14:38][VV][uart_esp8266:132]:     Read 0b11111111 (0xFF)
[16:14:38][VV][uart_esp8266:132]:     Read 0b11111111 (0xFF)
[16:14:38][VV][uart_esp8266:132]:     Read 0b11111111 (0xFF)
[16:14:38][VV][uart_esp8266:132]:     Read 0b11111111 (0xFF)
[16:14:38][VV][uart_esp8266:132]:     Read 0b11111111 (0xFF)
[16:14:38][VV][uart_esp8266:132]:     Read 0b11111111 (0xFF)
[16:14:38][VV][uart_esp8266:132]:     Read 0b11111111 (0xFF)
[16:14:38][VV][uart_esp8266:132]:     Read 0b01001110 (0x4E)

You can use the custom UART Test sensor to update a template sensor or lambda based on the values sent back from the MCU over serial to the ESP after a button press. The disadvantage with this is that we're back to using the MCU to read the remote buttons and this introduces the 1-2s lag, restricts the control to fixed buttons/dim% and locks out the other buttons or third party RF remotes.

gytisgreitai, 2020-09-18:
And what if I don't need the RF? Is there a working guide/config so I could just control the on/off/brightness via MQTT ? Thanks

JeffResc, 2020-10-09:
> @craigueh That's amazing because it is the missing link that Tasmota can't do now. And it opens up the D1 to be so interesting as a dimmer with 8 controllable buttons.
>
> Here is a [...]

@robbz23 Just wondering, what are you trying to implement here in Python? I played around with merging code from Tasmota’s implementation and used it in a custom ESPHome light component that supports dimming. I have to get a new bulb for my socket to verify everything is working as intended, but after that, I should be able to have working code for a light component that can control on/off/brightness.

> And what if I don't need the RF? Is there a working guide/config so I could just control the on/off/brightness via MQTT ? Thanks

@craigueh Like I said above, I have a mostly working implementation, give me a day to verify and I’d be happy to share the code here for those controls :)

JeffResc, 2020-10-09:
> Ok, so I've done a bit of playing for a few hours and found a way to bypass the MCU and decode RF signals directly. You need to connect a wire from the top right corner pin on the 590R RF chip to solder pad 4 (gpio4).
> 
> [image]
> 
> Then use the following code in esphome to receive the RF codes and map to buttons:
> 
> ```
> remote_receiver:
>   pin: 4
>   dump: rc_switch
> #filter: 200us
>   filter: 4us
>   idle: 4ms
>   tolerance: 50
> 
> binary_sensor:
>   - platform: status
>     name: "Sonoff D1 Status"
> 
>   - platform: remote_receiver
>     name: "Sonoff Remote Button 1"
>     rc_switch_raw:
>       code: '100010101110101111101000'
>       protocol: 1
>     filters:
>       delayed_off: 1s
> 
>   - platform: remote_receiver
>     name: "Sonoff Remote Button 2"
>     rc_switch_raw:
>       code: '100010101110101111101100'
>       protocol: 1
>     filters:
>       delayed_off: 1s
> 
>   - platform: remote_receiver
>     name: "Sonoff Remote Button 3"
>     rc_switch_raw:
>       code: '100010101110101111100100'
>       protocol: 1
>     filters:
>       delayed_off: 1s
> 
>   - platform: remote_receiver
>     name: "Sonoff Remote Button 4"
>     rc_switch_raw:
>       code: '100010101110101111101001'
>       protocol: 1
>     filters:
>       delayed_off: 1s
> 
>   - platform: remote_receiver
>     name: "Sonoff Remote Button 5"
>     rc_switch_raw:
>       code: '100010101110101111100010'
>       protocol: 1
>     filters:
>       delayed_off: 1s
> 
>   - platform: remote_receiver
>     name: "Sonoff Remote Button 6"
>     rc_switch_raw:
>       code: '100010101110101111100101'
>       protocol: 1
>     filters:
>       delayed_off: 1s
> 
>   - platform: remote_receiver
>     name: "Sonoff Remote Button 7"
>     rc_switch_raw:
>       code: '100010101110101111100001'
>       protocol: 1
>     filters:
>       delayed_off: 1s
>   - platform: remote_receiver
>     name: "Sonoff Remote Button 8"
>     rc_switch_raw:
>       code: '100010101110101111100011'
>       protocol: 1
>     filters:
>       delayed_off: 1s
> ```
> 
> Now if only someone could tell me what I need to do to control the other MCU for dimming we would have full esphome control of this device and be able to double it up as a Sonoff RF Bridge and use all 8 buttons to control other devices in Homeassistant.

@craigueh Could you share the original functionality of each button (1-8) so that I can merge them with the other code that I mentioned before to try to implement the default controls?

JeffResc, 2020-10-09:
@craigueh Better yet, since you have access to the RF remote, where you able to decode the messages sent from the RF chip to the ESP?

In the Tasmota code linked below, they have a comment stating `Received ack from Rf chip (aa 55 01 04 00 00 05)` which I interpret as the ESP receiving this code via serial from the RF chip requesting a toggle of the dimmer to occur. Could you verify this or find the values for the other buttons?

[Referenced Tasmota code](https://github.com/arendst/Tasmota/blob/2d4a6a29ebc7153dbe2717e3615574ac1c84ba1d/tasmota/xdrv_37_sonoff_d1.ino#L44-L78)

robbz23, 2020-10-10:
@JeffResc Yes the design of the D1 is that there are 3 chips.  The esp handles wifi, one microcontroller handles RF and the other one handles dimming.  When you press a button on the remote the RF chip tells the dimming chip the value.  The dimming chip will then via UART update the status to ESP so that the app will show the correct state.  You can also control the dimming chip by sending the appropriate commands via uart to the dimming chip.  Check out the [feature request issue on Tasmota.  ](https://github.com/arendst/Tasmota/issues/7598#issuecomment-587207331).  In there you can see how to formulate the uart command to turn on/off and adjust brightness.

markbooth73, 2020-10-10:
I have one of these dimmers and have flashed it with the latest release of Tasmota. 8.5.0

My question relates to adjusting the dimmer output, I have looked for console commands but see nothing for the configuring the D1.

What I observe is that even as low as 1 % the output is way too bright, so is there a way to re-range, calibrate ?

I do not seem to have the well documented issue of ghost switching

JeffResc, 2020-10-10:
@robbz23 Thanks for the reply, I've actually already got full control of the brightness and on/off working in ESPHome. I wanted to implement the RF remote though so we could have a fully working device, but I don't have an RF remote so I was looking if someone that did could provide me the RF codes so I could combine that with my working code in order to have a fully working ESPHome device.

markbooth73, 2020-10-10:
> @robbz23 Thanks for the reply, I've actually already got full control of the brightness and on/off working in ESPHome. I wanted to implement the RF remote though so we could have a fully working [...]

JeffResc, would you mind sharing your ESPHome config as is for now?

craigueh, 2020-10-10:
Here's my config:

uart:
  baud_rate: 9600 # speed to STC15L101EW
  tx_pin: GPIO1
  rx_pin: GPIO3
#  id: uart_bus

remote_receiver:
  pin: 4
  dump: rc_switch
#filter: 200us
  filter: 4us
  idle: 4ms
  tolerance: 50

#text_sensor:
#- platform: custom
#  lambda: |-
#    auto my_custom_sensor = new UartReadLineSensor(id(uart_bus));
#    App.register_component(my_custom_sensor);
#    return {my_custom_sensor};
#  text_sensors:
#    id: "uart_readline"
#    name: UART Data
    
#switch:
#  - platform: template
#    name: 'Sonoff Dimmer Switch'
#    id: dimmer
#    turn_on_action:
#      - uart.write: [0xAA, 0x55, 0x01, 0x04, 0x00, 0x0A, 0x01, 0x64, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x6C]
#    turn_off_action:
#      - uart.write: [0xAA, 0x55, 0x01, 0x04, 0x00, 0x0A, 0x00, 0x64, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x6B]
#    optimistic: true
    
  
binary_sensor:
  - platform: status
    name: "Snoff D1 Status"

  - platform: remote_receiver
    name: "Sonoff Remote Button 1"
    rc_switch_raw:
      code: '100010101110101111101000'
      protocol: 1
    filters:
      delayed_off: 500ms
#    on_press:
#      then:
#        - uart.write: [0xAA, 0x55, 0x01, 0x04, 0x00, 0x0A, 0x01, 0x64, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x6C]

  - platform: remote_receiver
    name: "Sonoff Remote Button 2"
    rc_switch_raw:
      code: '100010101110101111101100'
      protocol: 1
    filters:
      delayed_off: 500ms
#    on_press:
#      then:
#        - uart.write: [0xAA, 0x55, 0x01, 0x04, 0x00, 0x0A, 0x00, 0x64, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x6B]

  - platform: remote_receiver
    name: "Sonoff Remote Button 3"
    rc_switch_raw:
      code: '100010101110101111100100'
      protocol: 1
    filters:
      delayed_off: 500ms

  - platform: remote_receiver
    name: "Sonoff Remote Button 4"
    rc_switch_raw:
      code: '100010101110101111101001'
      protocol: 1
    filters:
      delayed_off: 500ms

  - platform: remote_receiver
    name: "Sonoff Remote Button 5"
    rc_switch_raw:
      code: '100010101110101111100010'
      protocol: 1
    filters:
      delayed_off: 500ms

  - platform: remote_receiver
    name: "Sonoff Remote Button 6"
    rc_switch_raw:
      code: '100010101110101111100101'
      protocol: 1
    filters:
      delayed_off: 500ms

  - platform: remote_receiver
    name: "Sonoff Remote Button 7"
    rc_switch_raw:
      code: '100010101110101111100001'
      protocol: 1
    filters:
      delayed_off: 500ms
      
  - platform: remote_receiver
    name: "Sonoff Remote Button 8"
    rc_switch_raw:
      code: '100010101110101111100011'
      protocol: 1
    filters:
      delayed_off: 500ms

JeffResc, 2020-10-10:
@craigueh Do you have the ESP directly connected to the RF receiver? I thought that there was another chip that the ESP had to communicate with via serial in order to receive RF commands

craigueh, 2020-10-10:
Sorry I don't know how to insert the code without the formatting going crazy..

Yes, I have the esp chip connected directly to the rf receiver, I found the delay going through the other chip unbearable.

JeffResc, 2020-10-10:
@craigueh You have them named as "Sonoff Remote Button 3," "Sonoff Remote Button 4," etc... What are the actual functions of these buttons? For example, which is on, off, brightness up, brightness down, etc?

craigueh, 2020-10-10:
I’m not sure off hand what the actual original functions were but if you have a pic of the remote button 1 is top left and button 8 is bottom right so you should be able to work the rest out from that.

JeffResc, 2020-10-10:
@craigueh Awesome, thanks. I'm making a full write-up on how to make this all work and it should be done shortly :)

JeffResc, 2020-10-10:
@craigueh I have some code that I would like you to test. Would be be interesting and if so, do you have somewhere where I can instant message you like Discord?

craigueh, 2020-10-11:
Yeah cool. I’d be happy to test. I’m on discord. 

craggy#0561

JeffResc, 2020-10-11:
@craigueh I sent you a friend request. I think you have to accept before I can send a DM.

JeffResc, 2020-10-11:
I've posted a small write-up [on my website](https://JeffResc.dev/blog/2020-10-10). Please open an issue directly in the project's GitHub page, should you encounter one.

Drew765, 2020-10-11:
Thanks for the time you've spent on this Jeff. I'll be trialling this in the week or so.

markbooth73, 2020-10-12:
Jeff, many thanks for your efforts, I have compiled the binary and OTA uploaded to the D1.
For now I see NO issues and will provide feedback thus needed, again thankyou.
Regards,

robbz23, 2020-10-12:
@JeffResc  Thanks so much.  I didn't know about the custom light component.  This is much easier than adding a new device, which I looked at and was a bit overwhelmed by.  

I tried messing with it and it works well from HA to D1, but when I use the remote strange things happen.  Where should I post more info?  here or your github?

JeffResc, 2020-10-12:
So as for ghost switching/dimming, it seems like that's coming from the RF chip. But good news is, it looks like the RF chip reports the ghost switch/dim over serial, so what I was planning on doing for users that don't need RF is when the ESP receives an RF signal, it will automatically just override it which I think could fix the issue.

robbz23, 2020-10-12:
@JeffR I just added an issue on your github after testing the RF file today. Also I don't think you can have it ignore rf chip just using software.  AFAIK the RF chip talks to the dimming chip which in turn sends messages to the ESP.  So when you push a button it would dim then inform the esp of the updated level.  The best you could do is program the esp to set the brightness back to what it thinks it should be if it is changed by something else.

JeffResc, 2020-10-13:
@robbz23 That's what I meant when I said override. I was planning on when the ESP receives a command from the RF chip, it would just override the change made by the RF chip and send the previous value again. This could fix the issues for those that don't have an RF remote.

craigueh, 2020-10-13:
@ robbz23 So the codes are unique to each remote? That's interesting. I presumed they would be the same across remotes. That's actually good because using one D1 in my kitchen I can use multiple Sonoff remotes in my kitchen/living room/dining room and control things like ceiling speaker volume and media playback in each room with separate remotes.

robbz23, 2020-10-13:
@craigueh  Im not sure they all were different, but I know some were.

robbz23, 2020-10-14:
I was just going to ask you that.  How do you unpair?

JeffResc, 2020-10-14:
I don't have a remote so please excuse me if this is wrong, but I think you may be looking for [this resource](https://sonoff.tech/product-tutorials/how-to-pair-and-clear-433-mhz-remote-controller-in-sonoff-devices), @robbz23.

craigueh, 2020-10-14:
Just hold the bottom left button on the remote for about 5-10s after you power up the d1 until you hear the beeps.

8osman, 2021-01-28:
I'm looking for some help. 
I want to use the D1 as a replacement for a traditional dimmer switch (uk) and I'm trying to work out the gpio pins which would be best, ideally I'd like to use GPIO4 as per @craigueh suggestion of faster custom rf support.

I would like to connect a rotary encoder to some other gpio pins. I need to GPIO's I would really appreciate some suggestions as my electronics knowledge is limited.

GPIO 16 is currently connected to the OTA switch, I was thinking of just soldering off this for the on off.

I was trying to work out what GPIO14 was doing, it looks to be connected to a resistor / capacitor combination then to ground?

GPIO1 + 3 are on the rx / tx uart?

Thank you

craigueh, 2021-01-28:
Gpio1/3 are needed for uart communication to the dimming chip/zero cross detection so that rules them out. 

You could use gpio0 for the rotary encoder SW push button and gpio16 for DT. I can’t remember off hand what gpio14 is used for but if possible use it for CLK. Is there an led on the board that can be hijacked for a gpio?
Mine is up in a ceiling so have no access to take a peek inside.

8osman, 2021-01-28:
Thank you. I'll explore a bit more and see what i can do.

There is an led on the board so might use that.

marcospsn, 2021-03-20:
> @ robbz23 Obrigado pela resposta, na verdade já tenho controle total do brilho e liga / desliga trabalhando no ESPHome. Eu queria implementar o remoto RF para que pudéssemos ter um dispositivo totalmente funcional, mas eu não tenho um remoto RF, então eu estava procurando se alguém que tivesse poderia me fornecer os códigos RF para que eu pudesse combiná-los com meu código de trabalho em a fim de ter um dispositivo ESPHome totalmente funcional.

vc poderia compartilhar o seu compilado do esphome pra esse D1?

venkattetali, 2021-04-24:
> And what if I don't need the RF? Is there a working guide/config so I could just control the on/off/brightness via MQTT ? Thanks

Did it work for you I mean without remote

Eenoo, 2021-06-24:
I have this working, so many thanks.
Is there any progress on merging this into the release code? Or has this been done and just needs some documentation drafting?

Drew765, 2021-06-25:
@Eenoo have you flashed ESPhome or using Tasmota?

Eenoo, 2021-06-25:
@Drew765 I used the code posted by Jeff and his guide, so it shows up under ESPhome - https://jeffresc.dev/blog/2020-10-10/

Would be nice to have this properly included though.

Drew765, 2021-06-25:
Thanks @Eenoo  😊
Hope it gets merged soon to.

luca-iodice, 2021-10-01:
Unfortunately the dimmer programmed with Jeff's code does not turn off the lights after the release of esphome 2021.9 and in particular this modification https://github.com/esphome/esphome/pull/2320

anatoly-savchenkov, 2021-11-20:
I was inspired by @JeffResc code and created a standard Light component for ESPHome. Code was reworked to:
- Implement a Light component
- Better support request / response UART based protocol with the companion MCU (which actually implements dimming and talks to RM433). This turned to be the key for all the changes mentioned below
- Support configurable use of RM433 remote
- Properly support restoration of a previous state on power up
- Fight with infamous ghost RF commands in case RM433 is not used

Code works well in my setup and has been in use for more than a month already. I would highly appreciate extra testing before I submit a PR. You can try it by adding the following into your YAML file.

1. Add an external component
````
external_components:
  - source:
      type: git
      url: https://github.com/anatoly-savchenkov/esphome
      ref: sonoff-d1-dimmer-support
    components: [ sonoff_d1 ]
````
2. Make sure UART is properly configured:
````
uart:
  - id: uart_bus
    tx_pin: TX
    rx_pin: RX
    baud_rate: 9600
````
3. And finally, add a light component like this:
````
light:
  - platform: sonoff_d1
    uart_id: uart_bus
    use_rm433_remote: False
    name: Sonoff D1 Dimmer
    id: dim_light
    restore_mode: RESTORE_DEFAULT_OFF
    gamma_correct: 0.0
    default_transition_length: 1s
````

My setup has no RM433 remotes and very rare ghost commands. So, testing on your side would be helpful if:
- 433MHz remotes are used. My code should process commands coming from a companion MCU, change light intensity and propagate new values to HA. I tested that only with ghost commands. Don't forget to change YAML parameter to `use_rm433_remote: True`.
- You don't use any remote but suffer from frequent ghost RF commands. I expect not random switches or intensity changes
- Your D1 device has been modified as shown on one of the photos above to hook up 433MHz receiver directly to ESP. I don't have such modifications, equipment and skills to try, but looking at the code from @JeffResc I don't see anything which should prevent this new component from working.

If log level is set to `VERBOSE`, UART protocol is dumped. I would love to see dumps when RM433 is active.

dbrand666, 2021-12-09:
@anatoly-savchenkov I've been using TasmOTA for some time with no serious issues with other (non-D1) devices but I've been seeing frequent ghost switching on this device, usually every few hours with TasmOTA, since I added it a few weeks ago.

So I set up ESPHome and installed this branch on my D1. The one setup issue I had was that I saw conflicting uart configurations in this thread vs your branch. I went with

    uart:
      rx_pin: RX
      tx_pin: TX
      baud_rate: 9600
which seems to work.

The one anomaly I've seen so far is that I have stable dimming from level 1 to 70. If I go to level 71, the lights start flashing. Not just flickering but real rapid on/off. If I go to 72 and above, the light turns off. Going back to 71 goes back to flashing and 70 and below is fine - i.e. this is totally reproducible.
This was similar to the behavior I saw with TasmOTA. I went in and set the dimming range using the command line interface (I forget the variable) so that full brightness in HomeAssistant was stable.
Does ESPHome have a similar setting?

I will be following up on the ghost switching issue as I keep an eye on it for the next few days.

anatoly-savchenkov, 2021-12-10:
@dbrand666 As you probably know, actual dimming in D1 is implemented with an external MCU which is not under control of ESPHome or Tasmota. What we can do from ESP is to request a certain brightness level over UART link. So, I don't know how to really fix flashing on your side. This is probably also bulb specific. What I can do is to limit maximal (and minimal) brightness levels. I've just added min_value / max_value parameters (similar to Tuya dimmer). Both are in the range of 0..100 and can be set independently or together. Below is the extract from the docs:

- **min_value** (*Optional*, int): The lowest dimmer value allowed. Acceptable value for your
  setup will depend on actual light bulbs installed and number of them. Start with the default
  value and check what will be the minimal brightness bulbs can render. Pay attention that for
  some dimmable LED lamps minimal turn-on brightness will be higher that the minimal achievable
  brightness if you just decrease it when lamp is already turned on. Defaults to 0.
- **max_value** (*Optional*, int): The highest dimmer value allowed. Use this to hard-limit light
  intensity for your setup. For some bulbs this parameter might be also useful to prevent
  flickering at high brightness values. Defaults to 100.

My test config looks like below
````
light:
  - platform: sonoff_d1
    use_rm433_remote: False
    max_value: 80
    min_value: 20
    name: My dimmer
    id: dim_light
    restore_mode: RESTORE_DEFAULT_OFF
    gamma_correct: 0.0
    default_transition_length: 1s
````

Please give it a try.

dbrand666, 2021-12-11:
Truly a thing of beauty. Thank you!

dbrand666, 2021-12-11:
I want to mention one thing I hit today when I tried to update. Out of nowhere I started getting the following error during validation:

    Failed config
    
    external_components: [source .esphome/master-bedroom-chandelier.yaml:2]
      - [source .esphome/master-bedroom-chandelier.yaml:2]
        
        usage: git stash list [<options>]
           or: git stash show [<stash>]

The only change I'd made since yesterday was to add the `max_value` attribute. Reverting the change didn't fix the error.
I was finally able to fix it by adding a slash at the end of the github url ¯\_(ツ)_/¯.
I realize this is an esphome issue and unrelated to this PR but I figured I should mention it in case anyone else hits it.

anatoly-savchenkov, 2021-12-12:
Thanks @dbrand666, I've updated my comment above with a slash at the end of the source URL. Both variants work on my side, I have no clue why it started to complain in your case.

Diverjim84, 2021-12-18:
@dbrand666 to be clear, you migrated to esp home and eliminated ghost flashing?  I’ve got tas on mine and have horrible ghost ons. 

 Is there a way to migrate via OTA? I’ve not used esp home, but if it makes these dimmers work, I’ll migrate the whole house.

dbrand666, 2021-12-18:
Yes. I ended up migrating all my devices from tasmota to esphome. Just make sure to use @anatoly-savchenkov 's branch for now until his PR gets merged.
You can migrate via OTA. Generate the images with esphome and flash with the tasmota UI.
I had to gzip the images to get them to flash - the uncompressed images were failing to flash, probably not enough space on the device. Other than that it went pretty smoothly.

anatoly-savchenkov, 2021-12-18:
@Diverjim84, just to set your expectations properly, ESPHome light component mentioned in one of my previous comments eliminates ghost switching only for setups where RM433 (or any compatible) remote is **not** used.

I've recently learned that ghost switching can happen even if a remote is paired. For such setups please try recommendations [here](https://github.com/arendst/Tasmota/issues/7598#issuecomment-992921603). I cannot verify those as I don't have a remote. If someone confirms, I'll update the documentation.

[26 more replies]

closed 2022-04-25 by robbz23