esphome/feature-requests #615: [RFC] Webserver - add more controls

Elkropac, 2020-02-25:
**Describe the problem you have/What new integration you would like**

I would like to replace "Toggle" buttons for fan and monochromatic light components with buttons allowing more control.

**Please describe your use case for this integration and alternatives you've tried:**

There is no way to set fan speed and light brightness now.

**Additional context**

I would like to discuss, how should this controls look like.
i was thinking 
 - for fan: replace "Toggle" with "Off|Low|Middle|High" (all buttons other than "Off" also send "On + level")
 - for light: replace "Toggle" with "arrow_down|arrow_up|Toggle" (arrow_down and arrow_up changes level by 5% and also sends on signal)

glmnet, 2020-02-28:
So this is about replacing mostly what happens here: https://github.com/esphome/esphome/blob/67cbaabd99ef05b3b9f9510220766bdc3aa6625e/esphome/components/web_server/web_server.cpp#L32

I see the script will attach events to this button when it is loaded https://github.com/esphome/esphome-docs/blob/d7b9d7995c9148fe5a1dd5cdb41677907ad7155b/_static/webserver-v1.js#L34

What you want to accomplish might be doable by just modifying this script, the script would have to replace the html tough, I'm ok with that, I would not even bother the web_server.cpp creating the toggle button element at all, this way we move workload from the esp to the browser.

glmnet, 2020-02-28:
Also I suppose all you want to command is already available on the rest api, if not, more api will be needed

kdayns, 2020-11-29:
I just wanted to use esphome with it's web interface on my ESP32. At first I was thinking this is a bug and something does not work well because I cannot do such a basic thing in the web as a dimming my led light.
Now I need to install full blown HomeAssistant just to be able to dim one damn led strip.
You could at least specify in your documentation that the web servers does not support anything more than a toggle buttons.

nagyrobi, 2022-07-06:
Implemented in webserver v2: https://esphome.io/components/web_server.html?server#version-2

closed 2022-07-06 by nagyrobi