esphome/feature-requests #616: Acces point SSID longer than 32 characters

stefanroelofs, 2020-02-25:
**Describe the problem you have/What new integration you would like**

I have a fallback accesspoint like this:

```
wifi:
  ssid        : !secret wifi_attic_ssid
  password    : !secret wifi_attic_password
  ap:
    ssid      : "${devicelocation} Fallback Hotspot"
    password  : !secret wifi_fallback_password
  manual_ip:
    static_ip : ${ip_address}
    gateway   : !secret gateway_ip
    subnet    : 255.255.255.0

```

Sometimes the AP SSID is longer than 32 characters. This gives a compilation error. 
Might be a good idea to automatically truncate the string at 32 characters.

**Please describe your use case for this integration and alternatives you've tried:**

Something like this:

`std::string ssid = ssid.substr (0,31);`
**Additional context**

randybb, 2020-02-26:
Based on the standard max length of SSID is 32 chars and for longer SSID it is better just raise an error. Hiding wrong settings/configurations is not a good idea (somebody other then could say, that esphome has to compile the code all the times and just ignore section with any errors). This will force user to correct it in his infrastructure.

stefanroelofs, 2020-02-26:
Fair enough.

klatvo, 2020-03-30:
I had the same issue and I use an own substitution now for the ap. So I can limit the size easily by my own. Looks like this:
`  devicename: eg_diningarea_ceiling
  upper_devicename: EG_Diningarea Ceiling
  wifi_fallback: egDining
  ....
  ap:
    ssid: $wifi_fallback
    password: !secret fallbackPW`

JohanBraeken, 2020-06-08:
> Based on the standard max length of SSID is 32 chars and for longer SSID it is better just raise an error. Hiding wrong settings/configurations is not a good idea (somebody other then could say, [...]

FYI: Technically, the standard says an SSID contains a maximum of 32 octets, without any specified encoding. This is not the same as 32 characters.
This implies a valid SSID can be arbitrary binary data, or contain whitespace characters including NUL.
Today, most devices do support UTF-8 encoded SSIDs.

OttoWinter, 2020-07-23:
Yes the standard doesn't say which encoding to use, but does specify a max length of 32 octets (=bytes).

The current validator though doesn't produce any false-negatives though, as it checks the length of the string (so number of code points). That will only produce false-positives. It also doesn't accept anything other than unicode (purely out of laziness, the YAML loader would have to be updated to handle raw byte data)

But I'm too lazy to fix that because you'd have to look pretty long to find something like that actually used in the wild.

closed 2020-07-23 by OttoWinter