esphome/feature-requests #617: Random-value Sensor for testing

DazWilkin, 2020-02-26:
This is a wonderful project, thank you for creating it. Somehow I'd not heard of it until yesterday.

**Describe the problem you have/What new integration you would like**

As I begin to explore ESPHome, I'm interested in having a 'test' sensor that I can configure on devices that will generate data that I can then use to test e.g. deliver to an MQTT broker.

It would be useful to have a device that doesn't generate a constant value.

Ideally, it would be possible to configure this sensor to provide values within e.g. a range.

**Please describe your use case for this integration and alternatives you've tried:**

I'm exploring ESPHome. I have ESP (8266) devices but I've yet to plug sensors into them. I would like to deliver sensor data to e.g. an MQTT broker. In order to minimize complexity, it would be helpful to quickly configure an ESP with a random-data generator. This then permits me to focus on configuring e.g. MQTT broker. Once I know that path is working, I can then configure real sensors.

**Additional context**

randybb, 2020-02-26:
Basically to implement something like "demo: true" for each component that will simulate presence of this sensor. You can achieve the same with "Template *" and output defined via lambda

DazWilkin, 2020-02-26:
Thank you. I (suspect) I only need one. I'll try it!

glmnet, 2020-02-26:
It is a great exercise to learn ESPHome by implementing your own sensor, of course it will need some understanding of C++

_not tested code:_

````yaml
sensor:
  - platform: template
    name: "Template Sensor"
    id: sensor1

interval:
  - interval: 1s
    then:
      lambda: |-
         static int value1 = 0;
         id(sensor1).publish_state(value1++);

```` 

Note: you can also use the lambda from the template sensor with update interval to a desired value, etc. Done this way to show you more options.

Everything written here is pretty well documented, even the use of the `static` inside the lambda

For the random is another whole stuff which I believe is not the most important part of this request.

closed 2020-02-26 by glmnet

DazWilkin, 2020-02-26:
You're awesome, thank you!

I created:

```YAML
sensor:
  - platform: template
    name: "Random Numbers"
    id: random_sensor
    unit_of_measurement: "%"
    lambda: |-
      return rand() % 100;
    update_interval: 60s
```