esphome/feature-requests #619: Serial Wifi bridge to delegate control to other hosts

dewi-ny-je, 2020-02-26:
**Describe the problem you have/What new integration you would like**

I would like ESPhome to also act as wifi-serial bridge.

**Please describe your use case for this integration and alternatives you've tried:**

For example, I have a air conditioning unit which talks via modbus. Programming a module in ESPhome requires knowledge and may need to be done per each modbus device, while a serial-wifi bridge would allow me to delegate all the control to Home Assistant, which does not require real programming skills. This would help to control less common devices which talk via modbus.

**Additional context**

oomelchukBB, 2020-03-03:
i had a similar problem.
You might take a look at register_service() function https://esphome.io/api/classesphome_1_1api_1_1_custom_a_p_i_device.html#a31b1777f3e960dd9fdd07ca7ea27fed3

I used it to set_text() for my custom display https://github.com/sashao/TM1638lite/blob/master/examples/esphome/TM1638Sensor.h#L73

Custom C++ text sensor can return data for you from modbus.

glmnet, 2020-04-17:
Duplicate of #660

closed 2020-04-17 by glmnet