esphome/feature-requests #621: DS2482 or DS2484 1-Wire master support

pilotak, 2020-02-28:
**Describe the problem you have/What new integration you would like**

It would be nice to have a support for DS2484 or DS2482 so that 1-wire bus is more reliable over a long distance.

**Please describe your use case for this integration and alternatives you've tried:**

DS18B20 over a long distance

**Additional context**

https://github.com/cybergibbons/DS2482_OneWire

Misiu, 2020-03-02:
+1 on this. I had crazy problems when trying to connect 5 sensors on wires longer than 10m.
I was able to connect three sensors without any problems but after connecting fourth sensors one or two sensors didn't return any results.
Finally, I've connected a single sensor to wemos d1 mini and put that around the house. This is a workaround. Having this would allow me to remove unneeded wemos boards.

pilotak, 2020-03-02:
@Misiu exactly

Erd86, 2020-11-17:
would be great to get DS2482 support in future. There are still sporadic issues in my setup with 7 DS18B20 sensors.
If there is something to test, please let me know. I have DS2482-100 available.

Erd86, 2020-11-28:
today I tried to get my ds2482-100 working as a custom sensor component.
I used this library: https://github.com/cybergibbons/DS2482_OneWire

Here my h-file.
[my_custom_sensor.h.txt](https://github.com/esphome/feature-requests/files/5611256/my_custom_sensor.h.txt)

and this yaml settings:

```
sensor:
- platform: custom
  lambda: |-
    auto my_sensor = new MyCustomSensor();
    App.register_component(my_sensor);
    return {my_sensor};

  sensors:
    name: "My Custom Sensor"

i2c:
   - id: bus_a
#     sda: 13
#     scl: 16
     scan: True
```

Scanning for sensors works great with this code and so I can see all my sensor ids. :-)
The problem is now to read the sensor values.. I always get value 255.. :-(

Can anyone please support? I have no programming experience for such a task..

itsieber, 2022-01-18:
Do you have found any solution?

saschaludwig, 2023-01-01:
With more and more entities reporting OneWire problems all over the place, how good are the chances of getting support for 1-Wire master chips?
https://github.com/esphome/issues/issues/3909

green2blue, 2023-01-25:
I agree, i have also problems with random scratchpad reading errors, because my wires are longer than 5m. A solid Bus Master would be great. Is there a plan for the future to integrate this?

ericmo1, 2023-03-10:
+1 for integrating support for i2c 1-wire bus masters like DS2482, DS2484. Looking forward to that :-) 
Have a nice day!

makuser, 2023-06-13:
Is someone by any chance already working on this?

mknjc, 2023-07-21:
I've started writing a component (https://github.com/mknjc/esphome/tree/ds248x). Its mostly copied from the dallas component. Its already working with DS18B20 sensors on an esp32 with a ds2484.

TODOs:
* Adding configurable support for APU and SPU.
* Sleep pin optional
* Adding bus and master sleep modes
* Cleanup

I would love to receive some help

pilotak, 2023-07-22:
@mknjc should you need inspiration, here my implementation for mbed https://github.com/pilotak/DS248X

troky, 2023-08-03:
> I've started writing a component 

I can help... can you provide simple config (yaml) example?

pilotak, 2023-08-04:
i can test on DS2482-100

itsieber, 2023-08-04:
I can also test on multiple DS2482-100

How can install this branch on my homeassistant?
Can you give me pls a demo yaml. THX

mknjc, 2023-08-04:
Hi, this would be cool.
I've updated the repository with optional support for using the sleep pin (needed on my board) and support for bus sleep, active pull up, and strong pull up settings.

An example yaml would be:

```
external_components:
  - source: github://mknjc/esphome@ds248x
    components:
      - ds248x

i2c:
  sda: 13 # set your i2c pins
  scl: 16 # set your i2c pins
  id: bus_a
  frequency: 400kHz # 400 is the maximum the ds2484 supports

ds248x:
  i2c_id: bus_a
  address: 0x18 # should be the default
  sleep_pin: 32 # remove if not needed, if set the ds248x sleeps when no transaction is in progress
  active_pullup: true # adds a active pullup which should improve signal integrity
  strong_pullup: true # enables a strong pullup used to provide the needed current for temperature reading without vcc connected
  bus_sleep: true # power down the bus when no transaction is in progres

sensor:
  - platform: ds248x
    address: 0x0 # set your sensor address
    name: "Temp"
    resolution: 12
```

like the onewire component the esp prints the found sensors on startup.

itsieber, 2023-08-05:
Perfect THX

1 have installed on two esp's.
ESP 1 with 2 Sensors work fine.

ESP 2 with 1 Sensor permanent 85°C

`i2c:
  sda: 22
  scl: 21
  scan: true
  id: bus_a
  #frequency: 400kHz # 400 is the maximum the ds2484 supports

ds248x:
  i2c_id: bus_a
  address: 0x18 # should be the default
  sleep_pin: 32 # remove if not needed, if set the ds248x sleeps when no transaction is in progress
  active_pullup: true # adds a active pullup which should improve signal integrity
  strong_pullup: true # enables a strong pullup used to provide the needed current for temperature reading without vcc connected
  bus_sleep: true # power down the bus when no transaction is in progres

sensor:
 - platform: ds248x
   address: 0x20000005fe78db28
   name: "PoolT"
   id: PoolT
   resolution: 12
   `

Result:

`[11:55:19][C][i2c.arduino:053]: I2C Bus:
[11:55:19][C][i2c.arduino:054]:   SDA Pin: GPIO22
[11:55:19][C][i2c.arduino:055]:   SCL Pin: GPIO21
[11:55:19][C][i2c.arduino:056]:   Frequency: 50000 Hz
[11:55:19][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[11:55:19][I][i2c.arduino:069]: Results from i2c bus scan:
[11:55:19][I][i2c.arduino:075]: Found i2c device at address 0x18
[11:55:19][C][ds248x:103]: DS248x:
[11:55:19][C][ds248x:105]:   Sleep Pin: GPIO32
[11:55:19][C][ds248x:107]:   Address: 0x18
[11:55:19][D][ds248x:114]:   Found sensors:
[11:55:19][D][ds248x:116]:     0x20000005fe78db28
[11:55:19][C][ds248x:119]:   Update Interval: 60.0s
[11:55:19][C][ds248x:122]:   Device 'PoolT'
[11:55:19][C][ds248x:122]:     Device Class: 'temperature'
[11:55:19][C][ds248x:122]:     State Class: 'measurement'
[11:55:19][C][ds248x:122]:     Unit of Measurement: '°C'
[11:55:19][C][ds248x:122]:     Accuracy Decimals: 1
[11:55:19][C][ds248x:130]:     Address: 0x20000005fe78db28
[11:55:19][C][ds248x:131]:     Resolution: 12
[11:55:19][C][captive_portal:088]: Captive Portal:
[11:55:19][C][mdns:108]: mDNS:
[11:55:19][C][mdns:109]:   Hostname: 1w-pool
[11:55:19][C][ota:093]: Over-The-Air Updates:
[11:55:19][C][ota:094]:   Address: 1w-pool.local:3232
[11:55:19][C][ota:097]:   Using Password.
[11:55:19][C][api:138]: API Server:
[11:55:19][C][api:139]:   Address: 1w-pool.local:6053
[11:55:19][C][api:143]:   Using noise encryption: NO
[11:56:03][D][ds248x:184]: 'PoolT': Got Temperature=85.0°C
[11:56:03][D][sensor:094]: 'PoolT': Sending state 85.00000 °C with 1 decimals of accuracy`

pilotak, 2023-08-05:
@itsieber this is due to weak pull up. Things tested

@mknjc test results
- :white_check_mark: no APU & SPU + external 1K resistor (sometimes `[E][ds248x:284]: Bus is shorted` `[E][ds248x:147]: Reset failed` but it could be due a breadboard)
-  :white_check_mark: only APU - wrong readings as expected
- :x: APU & SPU -  wrong readings, i checked on scope and PCTLZ pin never turns on (LOW level)

itsieber, 2023-08-05:
What do you mean "weal pull up"? Where can i search it to fix it?

pilotak, 2023-08-05:
@itsieber DS18B20 needs more power in order to do temp conversion, standard pullup 4K7 is not enough - this is the reason MAXIM intergrated strong pull up feature. Or you can skip APU & SPU and put 1K resistor between IO and VCC

pilotak, 2023-08-05:
I should mention that above tests were done in parasitic mode

in "classic mode"
-  :white_check_mark: only APU
-  no APU +no pullup should not work i believe but it works - maybe APU is not actually turn off?

randybb, 2023-08-05:
Will test that... Anyway, can it be modified to be more universal, as the original dallas component supports only temp sensors? With this we could support addressable switches, counters DS2423, key fobs (iButton),...

makuser, 2023-08-05:
> Will test that... Anyway, can it be modified to be more universal, as the original dallas component supports only temp sensors? With this we could support addressable switches, counters DS2423, key fobs (iButton),...

That's what I also wanted to comment on.
I'd really like to have it unified into a single component (the existing dallas component), and depending on the configuration (pin or i2c_addr) the appropriate translation layer would be used.
Will share more about this when I have more time.

A unified component makes much more sense though, because it would not require a reimplementation of all the bus members.

itsieber, 2023-08-05:
1k Resistor not fix the problem.
1x sensor with 3 wires

mknjc, 2023-08-05:
> x APU & SPU - wrong readings, i checked on scope and PCTLZ pin never turns on (LOW level)

I've checked the DS2482-101 datasheet (I've tested it only with the DS2484) and it doesn't have a bus sleep mode, maybe it's angry when you set the sleep bit. Have you tried to disable bus sleep? `bus_sleep: false`

>  * no APU +no pullup should not work i believe but it works - maybe APU is not actually turn off?

As far as I understand the DS2484 and the DS2482-101 contains a internal pullup (see datasheet page 2 "1-Wire Weak Pullup Resistor")

>ESP 2 with 1 Sensor permanent 85°C

interesting, did you tried to supply the ds18b20 with 5v? You could also try to reduce the adc resolution which might help:

`resolution: 9`

itsieber, 2023-08-05:
its allready on 5v
resolution doesnt help
wire lenght 2m

mknjc, 2023-08-05:
Did you try to disable the bus sleep?

itsieber, 2023-08-05:
Yea, same issue

pilotak, 2023-08-06:
> As far as I understand the DS2484 and the DS2482-101 contains a internal pullup (see datasheet page 2 "1-Wire Weak Pullup Resistor")

silly me you are right, i should have checked the datasheet first

> Have you tried to disable bus sleep? bus_sleep: false

yes, still the same, i did some debugging and even if you set SPU config and read the register back it's not set

pilotak, 2023-08-06:
i managed to fix the SPU please see https://github.com/mknjc/esphome/pull/2

randybb, 2023-08-11:
I have tested it with 4 sensors in TO92 package and with 7 sensors connected via 2m long cables to my heating controller. Everything works fine as before.

mknjc, 2023-09-08:
I've used the component for multiple weeks on an ESP32 and an ESP8266 without problems.

@itsieber did you try a version since 12th August? Does your sensor work with the dallas component?

I currently don't find the time to integrate the dallas and the ds248x components better, even though I would try to get ds248x upstream anyway if no other problems arise.

itsieber, 2023-09-08:
i have update today. still 85°C

randybb, 2023-11-30:
Could you please push this ds248x component to esphome?

timomer, 2023-12-23:
Hello,

I'm keen to move away from a pi with PI Zero DS2482 HAT to Arduino with esphome for monitoring my 1-wire sensors. I have basic coding skills and will see if I can assist with testing this code.

I have two quick questions before purchasing my components....
1. Has support for DS2482-800 been considered? This provides 8 1-wire masters and would be helpful to segment a 1-wire network if needed
2. Can anyone share their schematics for connecting the DS2482 to their Arduino?

Many thanks

mknjc, 2023-12-24:
Hello,

thank you. I appreciate every help.
Especially writing the esphome documentation would be needed to get the upstreaming process going.

My setup contains only 20 sensors so no segmentation is needed, but I guess it would be easy to add support for the DS2482-800.
I don't have any DS2438 to test it, but if somebody is willing to test it, it shouldn't be hard.

timomer, 2023-12-26:
Hi,

My issue is that I have a mix-up of Topologies (linear and star), while its around 12 sensors and appears reliable-ish, splitting this into two or three networks would be ideal. The DS2482-800 costs a fair bit more than the DS2482-100, as I would need max 3 channels it may be more affordable to support multiple DS2482-100 - do you believe your integration would allow this if 3 separate "external_components" are declared?

I will be ordering my parts soon and should get them in January and will do my best to help, ESPHome and 1-Wire are the way forward for me.

Can anyone recommend a DS2482-100 supplier on aliexpress?

> I don't have any DS2438 to test it, but if somebody is willing to test it, it shouldn't be hard.
Im happy too

green2blue, 2024-01-04:
I want to thank you. I have tried to apply the example that I found above in this Thread and it worked. It would be great if someone can push this to the main project. I have ordered the DS2482-800 on Ali and will receive it at the end of January. If you provide the code I will be happy to test.

`[20:53:01][C][wifi:414]:   Hostname: 'node-test'
[20:53:01][C][wifi:416]:   Signal strength: -52 dB ▂▄▆█
[20:53:01][C][wifi:420]:   Channel: 11
[20:53:01][C][wifi:421]:   Subnet: 255.255.255.0
[20:53:01][C][wifi:422]:   Gateway: 192.168.0.1
[20:53:01][C][wifi:423]:   DNS1: 192.168.0.1
[20:53:01][C][wifi:424]:   DNS2: 192.168.0.1
[20:53:01][C][logger:443]: Logger:
[20:53:01][C][logger:444]:   Level: DEBUG
[20:53:01][C][logger:445]:   Log Baud Rate: 115200
[20:53:01][C][logger:447]:   Hardware UART: UART0
[20:53:01][C][i2c.arduino:053]: I2C Bus:
[20:53:01][C][i2c.arduino:054]:   SDA Pin: GPIO21
[20:53:01][C][i2c.arduino:055]:   SCL Pin: GPIO22
[20:53:01][C][i2c.arduino:056]:   Frequency: 400000 Hz
[20:53:01][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[20:53:01][I][i2c.arduino:069]: Results from i2c bus scan:
[20:53:01][I][i2c.arduino:075]: Found i2c device at address 0x18
[20:53:01][C][ds248x:103]: DS248x:
[20:53:01][C][ds248x:105]:   Sleep Pin: GPIO32
[20:53:01][C][ds248x:107]:   Address: 0x18
[20:53:01][D][ds248x:114]:   Found sensors:
[20:53:01][D][ds248x:116]:     0xbc0416639d72ff28
[20:53:01][C][ds248x:119]:   Update Interval: 60.0s
[20:53:01][C][ds248x:122]:   Device 'Temp'
[20:53:01][C][ds248x:122]:     Device Class: 'temperature'
[20:53:01][C][ds248x:122]:     State Class: 'measurement'
[20:53:01][C][ds248x:122]:     Unit of Measurement: '°C'
[20:53:01][C][ds248x:122]:     Accuracy Decimals: 1
[20:53:01][C][ds248x:130]:     Address: 0xbc0416639d72ff28
[20:53:01][C][ds248x:131]:     Resolution: 12
[20:53:01][C][captive_portal:088]: Captive Portal:
[20:53:01][C][mdns:115]: mDNS:
[20:53:01][C][mdns:116]:   Hostname: node-test
[20:53:01][C][ota:097]: Over-The-Air Updates:
[20:53:01][C][ota:098]:   Address: node-test.local:3232
[20:53:01][C][ota:101]:   Using Password.
[20:53:01][C][api:139]: API Server:
[20:53:01][C][api:140]:   Address: node-test.local:6053
[20:53:01][C][api:142]:   Using noise encryption: YES
[20:53:22][D][ds248x:199]: 'Temp': Got Temperature=23.6°C
[20:53:22][D][sensor:094]: 'Temp': Sending state 23.56250 °C with 1 decimals of accuracy
[20:54:22][D][ds248x:199]: 'Temp': Got Temperature=23.6°C
[20:54:22][D][sensor:094]: 'Temp': Sending state 23.56250 °C with 1 decimals of accuracy
[20:55:22][D][ds248x:199]: 'Temp': Got Temperature=23.9°C
[20:55:22][D][sensor:094]: 'Temp': Sending state 23.93750 °C with 1 decimals of accuracy
[20:56:22][D][ds248x:199]: 'Temp': Got Temperature=23.5°C
[20:56:22][D][sensor:094]: 'Temp': Sending state 23.50000 °C with 1 decimals of accuracy
[20:57:22][D][ds248x:199]: 'Temp': Got Temperature=23.4°C
[20:57:22][D][sensor:094]: 'Temp': Sending state 23.43750 °C with 1 decimals of accuracy
[20:57:44][I][ota:117]: Boot seems successful, resetting boot loop counter.
[20:57:44][D][esp32.preferences:114]: Saving 1 preferences to flash...
[20:57:44][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[20:58:22][D][ds248x:199]: 'Temp': Got Temperature=23.5°C
[20:58:22][D][sensor:094]: 'Temp': Sending state 23.50000 °C with 1 decimals of accuracy
[20:59:22][D][ds248x:199]: 'Temp': Got Temperature=23.4°C
[20:59:22][D][sensor:094]: 'Temp': Sending state 23.43750 °C with 1 decimals of accuracy
[21:00:22][D][ds248x:199]: 'Temp': Got Temperature=23.4°C
[21:00:22][D][sensor:094]: 'Temp': Sending state 23.43750 °C with 1 decimals of accuracy`

[image]

green2blue, 2024-01-05:
Today I tried all the DS18B20 sensors I have on one big network. These are some Original and Counterfit ones. This Chip is amazing. I have got about 25 Sensors with different Sensors Wire length from 1m to 7m working without any problem on 3.3V. 

Here is my config:
`esphome:
  name: node-test
  friendly_name: node-test

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

web_server:
  port: 80

external_components:
  - source: github://mknjc/esphome@ds248x
    components:
      - ds248x

i2c:
  sda: GPIO21 # set your i2c pins
  scl: GPIO22 # set your i2c pins
  id: bus_a
  frequency: 400kHz # 400 is the maximum the ds2484 supports

ds248x:
  i2c_id: bus_a
  address: 0x18 # should be the default
  sleep_pin: 32 # remove if not needed, if set the ds248x sleeps when no transaction is in progress
  active_pullup: true # adds a active pullup which should improve signal integrity
  strong_pullup: true # enables a strong pullup used to provide the needed current for temperature reading without vcc connected
  bus_sleep: true # power down the bus when no transaction is in progres
  update_interval: 10s

sensor:
  - platform: ds248x
    address: 0xdd031865b23fff28 # set your sensor address
    name: "Temp-Node-Test"
    resolution: 12
  - platform: ds248x
    address: 0xc90218606406ff28 # set your sensor address
    name: "Temp-1"
    resolution: 12
  - platform: ds248x
    address: 0xb303186623d7ff28 # set your sensor address
    name: "Temp-2"
    resolution: 12
  - platform: ds248x
    address: 0x860218606d67ff28 # set your sensor address
    name: "Temp-3"
    resolution: 12
  - platform: ds248x
    address: 0xf20218604307ff28 # set your sensor address
    name: "Temp-4"
    resolution: 12
  - platform: ds248x
    address: 0x7b031865ba07ff28 # set your sensor address
    name: "Temp-5"
    resolution: 12
  - platform: ds248x
    address: 0x830118678de3ff28 # set your sensor address
    name: "Temp-6"
    resolution: 12
  - platform: ds248x
    address: 0x8f01186791ddff28 # set your sensor address
    name: "Temp-7"
[... 552 lines cut ...]
[11:40:16][D][sensor:094]: 'Temp-13': Sending state 22.06250 °C with 1 decimals of accuracy
[11:40:16][D][ds248x:199]: 'Temp-14': Got Temperature=19.5°C
[11:40:16][D][sensor:094]: 'Temp-14': Sending state 19.50000 °C with 1 decimals of accuracy
[11:40:16][D][ds248x:199]: 'Temp-15': Got Temperature=22.0°C
[11:40:16][D][sensor:094]: 'Temp-15': Sending state 22.00000 °C with 1 decimals of accuracy
[11:40:16][D][ds248x:199]: 'Temp-16': Got Temperature=22.0°C
[11:40:16][D][sensor:094]: 'Temp-16': Sending state 22.00000 °C with 1 decimals of accuracy
[11:40:16][D][ds248x:199]: 'Temp-17': Got Temperature=22.5°C
[11:40:16][D][sensor:094]: 'Temp-17': Sending state 22.50000 °C with 1 decimals of accuracy
[11:40:16][D][ds248x:199]: 'Temp-18': Got Temperature=21.8°C
[11:40:16][D][sensor:094]: 'Temp-18': Sending state 21.75000 °C with 1 decimals of accuracy
[11:40:16][D][ds248x:199]: 'Temp-19': Got Temperature=21.6°C
[11:40:16][D][sensor:094]: 'Temp-19': Sending state 21.62500 °C with 1 decimals of accuracy
[11:40:16][D][ds248x:199]: 'Temp-20': Got Temperature=22.0°C
[11:40:16][D][sensor:094]: 'Temp-20': Sending state 22.00000 °C with 1 decimals of accuracy
[11:40:16][D][ds248x:199]: 'Temp-21': Got Temperature=22.1°C
[11:40:16][D][sensor:094]: 'Temp-21': Sending state 22.06250 °C with 1 decimals of accuracy
[11:40:16][D][ds248x:199]: 'Temp-22': Got Temperature=23.0°C
[11:40:16][D][sensor:094]: 'Temp-22': Sending state 23.00000 °C with 1 decimals of accuracy
[11:40:16][D][ds248x:199]: 'Temp-23': Got Temperature=22.0°C
[11:40:16][D][sensor:094]: 'Temp-23': Sending state 22.00000 °C with 1 decimals of accuracy
[11:40:16][D][ds248x:199]: 'Temp-24': Got Temperature=23.0°C
[11:40:16][D][sensor:094]: 'Temp-24': Sending state 23.00000 °C with 1 decimals of accuracy
[11:40:16][D][ds248x:199]: 'Temp-25': Got Temperature=22.2°C
[11:40:16][D][sensor:094]: 'Temp-25': Sending state 22.25000 °C with 1 decimals of accuracy

green2blue, 2024-01-12:
> Hello,
>
> thank you. I appreciate every help. Especially writing the esphome documentation would be needed to get the upstreaming process going.
>
> My setup contains only 20 sensors so no segmentation [...]

I now have the chip here and can test it. Did you already apply the changes to the code?
Is it possible to have multiple of these chips connected? (on the esphome side, i2c I know that is possible I only have to change the address)

timomer, 2024-03-19:
Hello, 

I finally got the parts delivered and finished stumbling along to build an ESPHome device based on WT32-WTH01 and DS2482-100, also a humidity sensor with a DS2438 and a HIH-4000-001 Honeywell - Humidity Sensor.

Plugging in the humidity sensor into my existing pi with a DS2482-100 HAT shows the sensor in Home Assistant where I can see Temp and Humidity.
Plugging into the ESPHome with your external component `github://mknjc/esphome@ds248x` the humidity sensor is not detected, but Temperature sensors (DS18B20) are.

This is as far as I have gotten, looking over the code I was expecting it to report at least an unknown sensor was detected:
https://github.com/mknjc/esphome/blob/67c37246151e152a34282f1a93125f0476401cf9/esphome/components/ds248x/ds248x.cpp#L81
I have increased logging to VERBOSE but no extra info is provided. 

Unfortunately this is where I hit the limit of my knowledge, happy to test or given some direction on where in the code adding support for a DS2438 slave is needed.

randybb, 2024-03-19:
ESPHome supports only DS18B20 temp sensors. Not sure about the detection mechanism, probably it shows only supported sensors, not everything detected on the bus.

alainstark, 2024-03-19:
Hello,

I am very interested in this new module in ESPHome.

My wish is to integrate a reliable network of numerous DS18B20 temperature sensors into Home Assistant.

I went with the [Olimex-PoE-ISO](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware) microcontroller. I added a voltage adapter to the I2C bus to go from 3.2V to 5V. Behind, I use a DS2482-800 and I compiled my first firmware last night.

And it works :-) With only one sensor for now.

On the other hand, the DS2482-800 has 8 1-wire buses and for the moment, only the first bus (which is configured by default) works.

The component data sheet talks about the C3h "Channel select" command but it is not used in the module's .cpp code.

I also found the code from [rickkas7](https://github.com/rickkas7/DS2482-RK) which dates from 7 years ago but which I found interesting and above all well documented.

I will already self-train on the creation of a new module for ESPHome and then, I am ready to help you improve, or even integrate this module into ESPHome.

[image]
[image]

alainstark, 2024-04-12:
Hello,

I modified the ds248x component code to add ds2482-800 management and it works :-)

I carried out a test with 14 ds18b20 sensors distributed over five different channels and over long distances using category 7 rj45 cable and everything works correctly.

Here is the yaml configuration file:

```
esphome:
  name: "olimex-01"
  friendly_name: "olimex-01"

esp32:
  board: esp32-poe-iso

# Enable logging
logger:
  # level: INFO
  level: DEBUG
  # level: VERBOSE
  # level: VERY_VERBOSE

<<: !include inc/olimex_network.yaml
<<: !include inc/auth.yaml

external_components:
  # source: github://mknjc/esphome@ds248x
  source:
    type: local
    path: local_components
  components: [ ds248x ]

i2c:
  sda: GPIO13 # set your i2c pins
  scl: GPIO16 # set your i2c pins
  id: bus_a
  frequency: 400kHz # 400 is the maximum the ds2484 supports

ds248x:
  i2c_id: bus_a
  type: ds2482-800
  address: 0x1f # should be the default
#  sleep_pin: 32 # remove if not needed, if set the ds248x sleeps when no transaction is in progress
  active_pullup: true # adds a active pullup which should improve signal integrity
  strong_pullup: false
  bus_sleep: false
#  strong_pullup: true # enables a strong pullup used to provide the needed current for temperature reading without vcc connected
#  bus_sleep: true # power down the bus when no transaction is in progres
  update_interval: 10s

sensor:

# Channel 0

  - platform: ds248x
    address: 0x80000002a5fea828
    channel: 0
    name: "Temp-01"
    resolution: 12

  - platform: ds248x
    address: 0x3a000002ceac6928
    channel: 0
    name: "Temp-02"
    resolution: 12

  - platform: ds248x
    address: 0x3a000002ceac6928
    channel: 0
[... 263 lines cut ...]
[18:06:40][D][sensor:094]: 'Temp-04': Sending state 17.43750 °C with 1 decimals of accuracy
[18:06:40][D][ds248x:342]: 'Temp-05': Got Temperature=17.5°C
[18:06:40][D][sensor:094]: 'Temp-05': Sending state 17.50000 °C with 1 decimals of accuracy
[18:06:40][D][ds248x:342]: 'Temp-06': Got Temperature=17.6°C
[18:06:40][D][sensor:094]: 'Temp-06': Sending state 17.56250 °C with 1 decimals of accuracy
[18:06:40][D][ds248x:342]: 'Temp-07': Got Temperature=17.5°C
[18:06:40][D][sensor:094]: 'Temp-07': Sending state 17.50000 °C with 1 decimals of accuracy
[18:06:40][D][ds248x:342]: 'Temp-08': Got Temperature=17.4°C
[18:06:40][D][sensor:094]: 'Temp-08': Sending state 17.43750 °C with 1 decimals of accuracy
[18:06:40][D][ds248x:342]: 'Temp-09': Got Temperature=17.6°C
[18:06:40][D][sensor:094]: 'Temp-09': Sending state 17.62500 °C with 1 decimals of accuracy
[18:06:40][D][ds248x:342]: 'Temp-10': Got Temperature=17.8°C
[18:06:40][D][sensor:094]: 'Temp-10': Sending state 17.75000 °C with 1 decimals of accuracy
[18:06:40][D][ds248x:342]: 'Temp-11': Got Temperature=17.4°C
[18:06:40][D][sensor:094]: 'Temp-11': Sending state 17.37500 °C with 1 decimals of accuracy
[18:06:40][D][ds248x:342]: 'Temp-12': Got Temperature=17.8°C
[18:06:41][D][sensor:094]: 'Temp-12': Sending state 17.81250 °C with 1 decimals of accuracy
[18:06:41][D][ds248x:342]: 'Temp-13': Got Temperature=17.8°C
[18:06:41][D][sensor:094]: 'Temp-13': Sending state 17.75000 °C with 1 decimals of accuracy
[18:06:41][D][ds248x:342]: 'Temp-14': Got Temperature=17.8°C
[18:06:41][D][sensor:094]: 'Temp-14': Sending state 17.75000 °C with 1 decimals of accuracy
```
Can you help me to add this new component to esphome ?
[image]
[image]

timomer, 2024-08-18:
Hello all,

Amazing to see the progress with this and now see we have ds2482-800 support!

For my project I need DS2438 sensor support, also I agree with @randybb and @makuser that this should be "unified into a single component" and support [additional sensors that HA can already display](https://www.home-assistant.io/integrations/onewire/) - this would be the best for ongoing HA community needs.

While I am keen to progress this, I lack the time and experience to do so, my own project I'm struggling to find the time\skill for. Therefore I ask is there a possibility to sponsor a developer for this work? Is there a formal HA way of doing this? I would be keen to support an existing developer to do this and contribute the code back to the community. 

From what I can see the outline of work to do is...

1. Add support for detecting other [Sensors that HA can display](https://www.home-assistant.io/integrations/onewire/)
2. Merge this code with [ESPHome 1-wire bus component](https://esphome.io/components/one_wire.html)

szogun1987, 2024-08-22:
I'm here to vote this thread up. Recent release added platform parameter to the 1-wire integration. Is it preparation for support of DS2482?

timomer, 2024-09-22:
Only just updated ESPHome and seen the recent 1-wire chnages that @szogun1987 has hinted: https://github.com/esphome/esphome/pull/6860

@ssieb the developer says...
> The dallas component has been split up into the one_wire bus and the dallas_temp sensor. This will allow for other 1-wire interfaces to be added and other 1-wire devices that aren't temperature sensors.

This sounds very promising to officially support the DS2482-100 and DS2482-800 as interfaces and the DS2438 as a new sensor.

@ssieb are you able to share any additional insight into next steps with 1-wire and ESPHome? I am keen to support this work and happy to sponsor a developer \ the project. As you have done recent work on the 1-wire component can you advise if this is possible? Many thanks

szogun1987, 2024-10-08:
I just found ongoing FR with some code 
https://github.com/esphome/feature-requests/issues/2693

@timomer just out of curiosity: why do you need DS2438?
Do you power remote 1-wire sensors with it's own batteries?

reijop, 2024-10-09:
I would bet they are using the extra a/d pin hooked to an analog sensor... Or doing some dark differential wizardry with the vbatt+/- pins.

> I just found ongoing FR with some code #2693
> 
> @timomer just out of curiosity: why do you need DS2438? Do you power remote 1-wire sensors with it's own batteries?

pilotak, 2024-10-09:
It's much reliable at long range, that is my main reason as well as 8 channel included - one central point, separated bus from each other.

saschaludwig, 2024-10-09:
> @timomer just out of curiosity: why do you need DS2438? Do you power remote 1-wire sensors with it's own batteries?

It supports parasite powered sensors

timomer, 2024-10-09:
Hey @szogun1987 , 

I've had too many DHT22 sensors die on me, im using them for monitoring humidity and temperature and want something more robust as I control my heating and ventilation levels at home based on these readings. Having something on 1 wire would be ideal and simplify my home setup.

1 wire does not have a native humidity sensor, but DS2438 supports temperature and humidity by adding a HIH sensor as @reijop has suggested.

Details of an example sensor [here](https://skyboo.net/2017/03/ds2438-based-1-wire-humidity-sensor/)

This is very common and H[ome Assistant supports](https://www.home-assistant.io/integrations/onewire/) reading the DS2438 with different sensors attached, I want that in ESPHome as well as that means I can feed this info back to HA from multiple ESPHome devices

falke1987, 2025-01-22:
Hello Guys, i am also very Happy if esphome will support ds2483🤩
Does anybody know when WE will get a First offical Version?

Thanks a lot

fran6120, 2025-02-01:
I tried this component with a ds2484 to read a ds18b20, but no luck. I can confirm that it works on Arduino Uno using Adafruit's ds248x library.

The funny thing is that the log shows the address of the ds18b20, even though no temperature readings are obtained, I always get the same error reporting that the bus is shorted.

```
[12:17:31][C][i2c.arduino:071]: I2C Bus:
[12:17:31][C][i2c.arduino:072]:   SDA Pin: GPIO4
[12:17:31][C][i2c.arduino:073]:   SCL Pin: GPIO5
[12:17:31][C][i2c.arduino:074]:   Frequency: 400000 Hz
[12:17:31][C][i2c.arduino:079]:   Timeout: 5000000 us
[12:17:31][C][i2c.arduino:086]:   Recovery: bus successfully recovered
[12:17:31][I][i2c.arduino:096]: Results from i2c bus scan:
[12:17:31][I][i2c.arduino:102]: Found i2c device at address 0x18
[12:17:31][C][ds248x:169]: DS248x:
[12:17:31][C][ds248x:173]:   Address: 0x18
[12:17:31][C][ds248x:177]:   Type: DD2482-100
[12:17:31][D][ds248x:190]:   Found sensors:
[12:17:31][D][ds248x:193]:     0xbd3cf2d444ce3628 (channel=0)
[12:17:31][C][ds248x:196]:   Update Interval: 5.0s
[12:17:31][C][ds248x:201]:   Device 'Temperatura célula carga'
[12:17:31][C][ds248x:201]:     Device Class: 'temperature'
[12:17:31][C][ds248x:201]:     State Class: 'measurement'
[12:17:31][C][ds248x:201]:     Unit of Measurement: '°C'
[12:17:31][C][ds248x:201]:     Accuracy Decimals: 4
[12:17:31][C][ds248x:201]:     Icon: 'mdi:thermometer-probe'
[12:17:31][C][ds248x:209]:     Address: 0xbd3cf2d444ce3628
[12:17:31][C][ds248x:210]:     Channel: 0
[12:17:31][C][ds248x:211]:     Resolution: 12

[12:17:35][W][component:170]: Component ds248x cleared Warning flag
[12:17:35][E][ds248x:435]: Bus is shorted
[12:17:35][W][component:157]: Component ds248x set Warning flag: unspecified
[12:17:35][E][ds248x:064]: Reset failed
[12:17:35][W][ds248x:320]: 'Temperatura célula carga' - Resetting bus for read failed!
[12:17:35][D][sensor:093]: 'Temperatura célula carga': Sending state nan °C with 4 decimals of accuracy
```

timomer, 2025-02-06:
Hello all,

Unfortunately, I have nothing new to add but wanted to say that I have had to put a pause on this as I need to finish some other projects right now, but very, very keen to pick this up at some point.

What is needed is a dev to pick this up, the below still stands:

> Only just updated ESPHome and seen the recent 1-wire chnages that [@szogun1987](https://github.com/szogun1987) has hinted: [esphome/esphome#6860](https://github.com/esphome/esphome/pull/6860)
> 
> [@ssieb](https://github.com/ssieb) the developer says...
> 
> > The dallas component has been split up into the one_wire bus and the dallas_temp sensor. This will allow for other 1-wire interfaces to be added and other 1-wire devices that aren't temperature sensors.
> 
> This sounds very promising to officially support the DS2482-100 and DS2482-800 as interfaces and the DS2438 as a new sensor.
> 
> [@ssieb](https://github.com/ssieb) are you able to share any additional insight into next steps with 1-wire and ESPHome? I am keen to support this work and happy to sponsor a developer \ the project. As you have done recent work on the 1-wire component can you advise if this is possible? Many thanks

falke1987, 2025-02-20:
i am still interessted did you have some news?
I searched some another options but nothing found there i can connect easy the sensors to HA :-(

dbl-0, 2025-02-23:
> i am still interessted did you have some news? I searched some another options but nothing found there i can connect easy the sensors to HA :-(

Also keen but do not have the skills to jump into this.

timomer, 2025-02-26:
Unfortunately, I have no update, but if there is a dev out there who is interested in being sponsored for this work I would be happy to help.
As of now, I am going to proceed with my project by...

1. Moving all temp sensors to ESPHome
2. For my humidity sensors (DS2438),[ use owserver to share them to HA directly ](https://www.home-assistant.io/integrations/onewire/#interface-adapter-setup)

falke1987, 2025-03-13:
also from my side i will support.

falke1987, 2025-04-17:
Good morning, any news regarding this topic?

tomtom13, 2025-07-17:
Sorry guys, but this issue is not resolved as it covers only ONE variant ds2484, while ds2482 is not fully supported. ds2484-800 only has ONE line out of 8 properly working on it.

[4 more replies]