esphome/feature-requests #622: TinyGPS Sensor data for all tracked data

kipwittchen, 2020-02-28:
**Describe the problem you have/What new integration you would like**
Esphome to pull in sensor data from all tinygps tracked data. Ie, location (lat/long), speed, course, altitude, # satellites, and hdop

**Please describe your use case for this integration and alternatives you've tried:**
I'm currently using an arduino script I wrote with TinyGPS++ for measuring fuel usage on my boat. The fuel flow meter uses pulses to measure flow rate and then with the GPS speed data I'm able to calculate fuel usage per and hour. Unfortunately I left the adafruit GPS board on the boat when I winterized it so I won't be able to test this integration for another couple months.

**Additional context**

kipwittchen, 2020-05-28:
Has anyone else tried taking a stab at this with any luck? I was able to get sensor data pulling into HA but realized I had set up the sensor component incorrectly. Or at least I think I did because the d1 mini I'm running it on was very unstable, frequently crashing with a soft WDT reset. I've tried separating out the sensor into a subfolder under gps but esphome isn't recognizing sensor.gps. 

I'm very new to github, haven't figured out yet how to push my code up for others to play with.

Puntoboy, 2020-06-24:
> Has anyone else tried taking a stab at this with any luck? I was able to get sensor data pulling into HA but realized I had set up the sensor component incorrectly. Or at least I think I did because [...]

How did you get sensor data into HA?

kipwittchen, 2020-06-24:
> How did you get sensor data into HA?

I had incorrectly modified the component to pull the sensor information in. It worked but also crashed every 30-120 seconds and I was unable to correctly write the code to work.

I ended up just writing a standard arduino ide script to publish the values to mqtt. I still would prefer using esphome but it serves my purposes well enough in the time being. If that's something you're interested in, I can share that code.

Puntoboy, 2020-06-24:
> I had incorrectly modified the component to pull the sensor information in. It worked but also crashed every 30-120 seconds and I was unable to correctly write the code to work.
>
> I ended up just [...]

Thanks. I've never used IDE, I'm just starting out with ESPHome. Was disappointed to read that only time can be used so was hoping for a way to fix it.

kipwittchen, 2020-06-27:
> Thanks. I've never used IDE, I'm just starting out with ESPHome. Was disappointed to read that only time can be used so was hoping for a way to fix it.

@Puntoboy sorry for the delay, here's the arduino script I'm using: https://github.com/kipwittchen/sandbox/blob/master/boat/mqtt_gps_fuel_sensor.ino

It requires a MQTT server such as mosquito to be running. It works reliably but definitely a bit of a hack at the moment, hopefully one of us can get this integration fully working with esphome soon!

Puntoboy, 2020-06-27:
Thanks. Struggling with IDE at the moment. Keep getting a stupid error that stops me uploading any code.  Never used it before so unsure if it's something I'm doing wrong.

kipwittchen, 2020-06-27:
Check out this video from bruh: https://www.youtube.com/watch?v=jpjfVc-9IrQ
It's a tad outdated but still a good reference for how to get the esp8266 based boards working with arduino IDE as well as get the libraries loaded in.

Puntoboy, 2020-06-27:
> 
> 
> Check out this video from bruh: https://www.youtube.com/watch?v=jpjfVc-9IrQ
> It's a tad outdated but still a good reference for how to get the esp8266 based boards working with arduino IDE as well as get the libraries loaded in.

Thanks. I've actually watched that before. I seem to be stuck at the first hurdle where IDE is complaining about the esptool directory not being in the same directly as upload.py, when it is.

Puntoboy, 2020-07-02:
Any idea when we will see full GPS in ESPHOME?

fi-sch, 2020-07-08:
I also would LOVE to see all GPS data in ESPHome! I'm wondering about doing DIY GPS trackers for my valuables – e.g. dog, car, bike, bicycle, photographic bag, etc.

kipwittchen, 2021-02-05:
Has anyone had time to look into this further? I tried again a couple months back but have unfortunately not had the time to figure it out.

closed 2021-04-11 by OttoWinter