esphome/feature-requests #624: Avoid burned pixels in OLED displays

DIYGuy01, 2020-03-02:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

dougle03, 2020-05-10:
Ideally you want to shift the image around rather than simply blanking..

DIYGuy01, 2020-05-11:
That won't help as some pixels will overlap.

------------------------
Andrius

nagyrobi, 2022-07-07:
Screen antiburn needed just like OpenHASP has: https://github.com/HASwitchPlate/openHASP/issues/134

RootUser3885, 2023-05-16:
I looked in to inverting the display during the day (white background and black letters). Then un-inverting during the night. Then all pixels get used rather evenly. Though, I haven't found a good way to change the "inverted" parameter on the fly.

jostsalathe, 2023-08-03:
I just posted a feature request (#2341) to implement a `set_invert(bool)` method for the `esphome::ssd1306_base::SSD1306` class.

You could then use `((esphome::ssd1306_base::SSD1306 &) it).init_invert(daytime);` or similar in your rendering lambda.

jostsalathe, 2023-08-22:
Update: [PR #5292](https://github.com/esphome/esphome/pull/5292) has kindly been created by @max246 which exposes the method `set_invert(bool)` as described above =]