esphome/feature-requests #625: Add support for Adafruit HT16K33 based LED displays

patfelst, 2020-03-03:
**Describe the problem you have/What new integration you would like**
Could you please add support for Adafruit [LED backpack displays](https://learn.adafruit.com/adafruit-led-backpack/featured_products) all based on the HT16K33 I2C chip. Their products include an LED bargraph, LED 8x8 dot matrix (single and multi-color), 7-segment LED, and 4-digit alpha-numeric LED display.

**Please describe your use case for this integration and alternatives you've tried:**
ESPHome currently does not have support for an alpha-numeric LED display which I am most interested in, yes ESPHome has the max7219 platform, and the upcoming TM1637 but they only support 7-segment displays.

**Additional context**
There is an existing [Arduino library](https://github.com/adafruit/Adafruit_LED_Backpack) which supports all the LED backpack types of displays. It is I2C based so should be a relatively simple adaption of the lcd_pcf8574 display platform.

patfelst, 2020-03-03:
Samuel Sieb has kindly created a custom component which currently supports just the 4-digit alpha-numeric display. I will not be able to test until the weekend.
https://github.com/ssieb/custom_components/tree/master/lcd_ht16k33

docteurzoidberg, 2020-11-22:
@ssieb made a custom component for theses, here: https://github.com/ssieb/custom_components/tree/master/ht16k33_alpha
Would be great to include !

42Network, 2021-02-04:
How do we vote? I vote for this one for sure. ssieb has quite a few components which could be pulled in to ESPhome.

ssieb, 2021-02-04:
Pulling isn't something that happens.  I need to write some docs and make PRs.

42Network, 2021-02-04:
Yes, I know you'd have to submit a PR. Meanwhile I'm glad for your work and I can include locally. However, I just now realized your code will not work for my HT16K33-based numeric LED display. It actually does work, but randomly displays garbage. I am using it for a simple temperature display, e.g. 104.1. 

I got it working though using this example: https://github.com/JAAlperin/esphome-adafruit-7-segment-LED-backpack-example. This isn't as nice as yours, though, because it requires much more code inside the display lambda.

I am not much of a programmer but I may try to either make your code work with the simple 4 digit numeric display, or may that other code work directly with ESPhome, with the appropriate Python wrappers.

ssieb, 2021-02-04:
Right, my component is for the alphanumeric displays that have 14 segments, not the 7 segment displays.
Is it the one with the colon in the middle? I just checked the wiring for that one and it seems a little wacky, but a simple component should be easy.  I don't see much point in including the scrolling support for a display like that.

42Network, 2021-02-05:
I agree - scrolling for number-only 4-digit 7-segment isn’t needed. But brightness control and access to the dots and colon and possibly individual segment control (ie there are a few letters and symbols that can be shown with just 7 segments) would be useful.

cwross, 2021-08-04:
Very interested in this as well, particularly for the 4-digit 7-segment display. I can test if needed. Thanks!

feaver12, 2022-02-09:
i am also interested in support for this same display!!!

smartroad, 2022-07-03:
Just to throw my towel into the ring, this is also used for Adafruits "Trellis" button/led board (https://www.adafruit.com/product/1616)

Could make a nice HMI self lighting buttons? (showing when things are on/off, maybe flashing for attention?). I don't know enough to add this myself :(

cTurtle98, 2023-04-15:
any progress on this one? i would love to be able to use this

kj4dxk, 2023-05-01:
I would like to use it to make a thermostat display.

SteveRMann, 2024-02-21:
Also interested. I have devices that use the 14-segment display from Adafruit, but I have to run them on Arduino and use MQTT to update them.

eifachtimon, 2025-05-08:
I have the same Adafruit 7 Segment Display! Is there any progress so far on this one?

SteveRMann, 2025-05-08:
I did it in ESPHome.  Here is my YAML
Hope it helps.. (I would like to see what you do with it).

`# cgm-monitor.yaml
# This configuration displays the time and glucose value on a
# 4-digit, 14-segment LED on an Adafruit ht16k33 backpack.

esphome:
  name: cgm-monitor
  friendly_name: CGM Monitor

  on_boot:
    priority: -100.0
    then:
      - script.execute: ht16k33_display_startup

esp8266:
  board: d1_mini

# Enable logging
logger:

# Enable I2C on Wemos D1 Mini
i2c:
  id: i2c_bus
  sda: D2
  scl: D1
  scan: true

# Enable Home Assistant API
api:

ota:
  - platform: esphome

#----------------------------- WiFi --------------------------------------
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

text_sensor:
  # Get the WiFi details
  - platform: wifi_info
    ip_address:
      name: IP
    ssid:
      name: SSID
    mac_address:
      name: Mac Address  

  # Get the Home Assistant Glucose value
  - platform: homeassistant
    id: glucose_sensor
    entity_id: sensor.dexcom_stevermann_glucose_value
    on_value:
      then:
        - script.execute: ht16k33_display_glucose      

  # Get the Home Assistant trend value
  - platform: homeassistant
    id: trend_sensor
    entity_id: sensor.dexcom_trend_val
    internal: true
  
    
#-------------------------------------------------------------------------

web_server:
  port: 80

[... 306 lines cut ...]
    then:
      - lambda: |-
         uint8_t brightness[] = {0xE8};
         id(i2c_bus).write(0x71, brightness, sizeof(brightness));
         delay(10);
         id(i2c_bus).write(0x70, brightness, sizeof(brightness));
         delay(10);

  # Script to flash the data indicator LED
  - id: flash_led
    mode: queued
    then:
      - switch.turn_on: data_indicator
      - delay: 1ms
      - switch.turn_off: data_indicator

# Blue LED, data indicator
switch:
  - platform: gpio
    id: data_indicator
    pin:
      number: D6
      mode: output

`