esphome/feature-requests #627: Custom networking components

sashao, 2020-03-03:
**Describe the problem you have/What new integration you would like**

ESPHome has hardcoded dependency to AsyncTCP. 
It would be great if at least api server can use other communication options.
E.g. Having ESPHomeTCP interface - fallback connectivity scenarios can be possible. Or new ways of communication using serial, i2c, lo-ra... 

**Please describe your use case for this integration and alternatives you've tried:**

For my home automation I would rather try to use wired-serial communication with devices as much as possible. Would someone else be interested in alternative networking implementations?

**Additional context**

I am ready to start working on this kind of component
1. extract AsyncTCP-like interface for esphome
2. Add default implementation using AsyncTCP
3. Add possibility to create custom c++ networking components.

Things like usb attached devices, one wire mesh, long range radio, vpn-proxyng, mocked api clients for unit testing might become possible.

clomads, 2023-04-02:
I love the idea of USB connected ESPHome nodes on a Pi powered HA deploy. In fact, this is basically how Klipper for 3D Printing works.

Could make off grid or otherwise remote systems more reliable without needing a router/switch/ap. You mentioned Lora, but I'm also excited for Thread/Zigbee on the newer ESP32s. Fallbacks alone would be killer. 

I'm down for testing whatever comes of this.

nagyrobi, 2025-02-07:
Making progress in https://github.com/esphome/esphome/pull/8187, and first implementation is for UART: https://github.com/esphome/esphome/pull/8214. This is a generalised approach, which will make possible to develop simpler components for exchanging data directly between nodes by any means, like TCP, ESP-Now, LoRa, etc.