esphome/feature-requests #628: Provide detailed example of custom component using existing Arduino libraries

patfelst, 2020-03-05:
**Describe the problem you have/What new integration you would like**
I have some reasonable plain C and OOP python programming skills, but I'm not an expert C++ coder and have been struggling to create custom components based on the existing documentation and DIY examples.

The custom component examples in ESPHome documentation focus on sensors and not outputs like displays, and there are not really any full worked examples to show how to integrate existing Arduino libraries.

I found this when trying to create a custom component for I2C LED display with an existing Adafruit library (refer to this [feature ](https://github.com/esphome/feature-requests/issues/625)request).

I would appreciate a documentation update to provide a more detailed example / tutorial on how to integrate existing Arduino libraries as custom components, and how to call them from yaml.

**Please describe your use case for this integration and alternatives you've tried:**
I've studied the ESPHome custom component documentation, DIY examples, and other custom components but fail to fully grasp how to integrate an existing Arduino library.

**Additional context**
None

nickrout, 2020-03-15:
You may find some good examples here https://esphome.io/guides/diy.html#custom-components-code

patfelst, 2020-03-16:
Thanks Nick, I had already seen those, I was really after more of a tutorial with a bit more explanation on the framework and objects / methods etc.