esphome/feature-requests #630: Can it support smartconfig, including (airkiss, smartconfig) and other protocols

yulei745, 2020-03-11:
**Describe the problem you have/What new integration you would like**

China's WeChat supports one-click network configuration, using the smartconfig airkiss protocol. Now it seems that esphomse does not support one-click network configuration, and can only be configured in the configuration file.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nickrout, 2020-03-15:
What advantages do you see in this? Most of us don't mind typing in the details of our ssid.

yulei745, 2020-03-25:
If you have used WeChat, you know why you should use this function. Http://wx.ai-thinker.com/api/old/wifi/config, this connection can be directly connected to esp8266 in WeChat. The connection The protocol used is smartconfig.

NdK73, 2020-04-09:
It's really useful if you have to give away the devices, so that your wifi credentials are not hardcoded in 'em.

brandond, 2020-04-09:
Otto has been pretty adamant in the past that configuration be done at build time. Maybe at some point in the future he'll change his mind, but at the moment it seems pretty unlikely.

nagyrobi, 2022-07-01:
See https://esphome.io/components/esp32_improv.html

closed 2022-07-01 by nagyrobi