esphome/feature-requests #632: Add support for i²c UV sensor Si1132

erichiller, 2020-03-13:
**Describe the problem you have/What new integration you would like**

Add support for the i²c sensor Si1132.

**Please describe your use case for this integration and alternatives you've tried:**

UV & ambient light sensor from hardkernel. That includes an Si1132 and a BMP280.

**Additional context**

[Product page for Hardkernel Weatherboard 2](https://www.hardkernel.com/shop/weather-board-2/)

[datasheet for Si1132](https://media.digikey.com/pdf/Data%20Sheets/Silicon%20Laboratories%20PDFs/Si1132.pdf)

CarlosGS, 2021-07-25:
Hi Eric! Did you manage to read luminosity with ESPHome for this specific sensor?

jjotto, 2022-11-02:
Does anyone have this working?

MaykThewessen, 2023-05-19:
This has the temp/humid sensor part of it working: the [BME280](https://www.bosch-sensortec.com/de/homepage/products_3/environmental_sensors_1/bme280/bme280_1) sensor
https://esphome.io/components/sensor/bme280.html