esphome/feature-requests #634: Bluetooth Battery Monitor

adrianmihalko, 2020-03-13:
**Describe the problem you have/What new integration you would like**

It would be great to add support for Bluetooth Battery Monitor

https://www.aliexpress.com/item/32794632957.html?
http://www.quicklynks.com/pro_show.php?id=159

**Please describe your use case for this integration and alternatives you've tried:**

It's a car/12V battery monitor. It would be great to install an Esphome gateway in the garage and monitor car battery voltage in Home Assistant.

About protocol/app: it does not need to pair with device, iOs app automatically finds the device and receive data. I would like to help If anyone shows me how to debug/record BLE traffic.

**Additional context**

closed 2020-03-13 by close-issue-app[bot]

adrianmihalko, 2020-03-13:
Really? What the heck is wrong with bot, there is no template when I open new issue:

[image]

randybb, 2020-03-13:
[image]