esphome/feature-requests #635: Bluetooth Battery Monitor

adrianmihalko, 2020-03-13:
**Describe the problem you have/What new integration you would like**

It would be great to add support for Bluetooth Battery Monitor

https://www.aliexpress.com/item/32794632957.html?
http://www.quicklynks.com/pro_show.php?id=159

**Please describe your use case for this integration and alternatives you've tried:**

It's a car/12V battery monitor. It would be great to install an Esphome gateway in the garage and monitor car battery voltage in Home Assistant.

About protocol/app: it does not need to pair with device, iOs app automatically finds the device and receive data. I would like to help If anyone shows me how to debug/record BLE traffic.

**Additional context**

[image]

nickrout, 2020-03-15:
Maybe this? https://support.honeywellaidc.com/s/article/How-to-capture-Bluetooth-traffic-from-and-to-an-Android-Device

adrianmihalko, 2020-03-15:
Here is the log:

[image]

[btsnoop_hci_202003151951.log](https://github.com/esphome/feature-requests/files/4334916/btsnoop_hci_202003151951.log)

I think the last packets are including the interesting data (in app 11.06V, 11.05V, 11.04V).

[image]
[image]

adrianmihalko, 2020-03-15:
Voltage set to 18.00V (sometimes it reads 18.01V):

```
    ][LE]> connect F8:30:02:2D:4D:9C
Attempting to connect to F8:30:02:2D:4D:9C
Connection successful
Notification handle = 0x002e value: b4 db ec df bc dc 19 b5 d6 0f fa e1 cf 9b 79 44
Notification handle = 0x002e value: c0 fc fd 6f c3 ad 32 9a 56 43 39 57 91 16 ad 90
Notification handle = 0x002e value: 09 90 cf 1d 17 59 17 6e 8a 40 ca 8f ea 6b e8 69
Notification handle = 0x002e value: 8a b3 b1 26 3d 82 8c 56 6e 9e 55 0b fd c7 e4 a2
Notification handle = 0x002e value: 56 f4 19 a3 d2 ff 96 de a0 93 0b d7 78 d9 8a d8
Notification handle = 0x002e value: 77 88 82 c5 85 4d 75 0b bb ed 68 22 45 b3 99 3f
Notification handle = 0x002e value: 9c c9 37 d7 0c f9 34 db ca cd 80 db 23 ed d7 eb
Notification handle = 0x002e value: 33 22 c8 59 be 5a 2e cb 0f 88 6b 6d d5 09 fe a6
Notification handle = 0x002e value: 90 8e 97 ce be 46 f0 be 0a 60 d3 44 a6 3a 06 7e
Notification handle = 0x002e value: d7 6f d8 81 71 a0 2f 1e 86 ad a4 f1 a6 07 57 37
Notification handle = 0x002e value: b9 61 70 6e 4a a0 c2 ba 56 ad 64 ef e1 8b 32 bc
Notification handle = 0x002e value: f0 09 9a cd ba 03 98 5c c4 14 bd 60 50 07 67 a2
Notification handle = 0x002e value: 12 5c a7 99 b9 0d 8d 55 1c a7 42 d6 48 21 92 50
Notification handle = 0x002e value: 2e 16 c9 b1 60 ca c7 a2 0f 20 b0 85 77 e2 31 b0
```

brucysNZ, 2021-03-25:
Did you get this to work at all? I have one of these devices and are also looking for way to use ESPHOME with Bluetooth code to read the data from the battery monitor. I am using Esphome to monitor temperature and humidity sensors that are Bluetooth and it works perfect. Just want to be able to read the Battery Monitor?

Ras195, 2022-01-14:
I'm interested too. Any news?

adrianmihalko, 2022-01-14:
Nothing, not even in any project.

beyeriii, 2022-06-26:
Related project
https://github.com/KrystianD/bm2-battery-monitor

johnczaia, 2022-07-18:
Was searching for this for-ever and just found this, looks very promising, esp that the incoming data is decrypted(!) first. I suspected it to be encrypted or at least a bit altered as it looks way too random. Not tested yet but about to:

https://github.com/KrystianD/bm2-battery-monitor
If this works, I'll chant Krystian's name very day at noon in reverie :)

johnczaia, 2022-07-18:
Very interesting read, especially the how-to re-engineering part. He managed to decompile and deobfuscate the Android app to get the encryption keys. I bet this works. My Linux box has sadly Python 3 3.10 and apparently they introduced "something" in 3.10 that breaks a needed part in a needed Python library (no clue of the details, I am more of a C#/Powershell guy, sorry).
Will fire up a Raspberry Pi tomorrow with an older Linux and try again. He supplied two .py files, one for MQTT (perfect, exactly what I need) and one for ESPhome.

johnczaia, 2022-07-18:
It works! I quickly set up a Raspberry PI 3+ with Raspberry OS and Python < 3.10 and it works beautifully!
MQTT: bm2 = 13.50 (I am currently charging my ICE car battery thus higher voltage).
Now to slow down the once per second MQTT sending (30 secs  or once per minute should suffice) and bliss :)
Big thanks to Krystian!

beyeriii, 2022-07-18:
Thanks for researching Krystian project and contributing results.

adrianmihalko, 2022-07-18:
Awesome news guys. 👍💯

TheOriginal92, 2023-05-27:
Any update on this?

adrianmihalko, 2023-05-27:
At this moment you can use OpenMQTTGateway which supports this device out of the box.

adrianmihalko, 2023-06-27:
https://doubleagent.net/2023/05/21/a-car-battery-monitor-tracking-your-location.html