esphome/feature-requests #639: Add support for Aranet4 Home CO2, relative humidity and temperature sensor over BLE

kasparsd, 2020-03-14:
**Describe the problem you have/What new integration you would like**

I would like to add [Aranet4 Home](https://aranet4.com) support. It reports the current sensor readings and historic values over BLE.

**Please describe your use case for this integration and alternatives you've tried:**

There are several integrations with other platforms that can be used for reference:

- https://github.com/SAF-Tehnika-Developer/com.aranet4
- https://github.com/Anrijs/Aranet4-Python

**Additional context**

nagyrobi, 2022-07-07:
See https://esphome.io/components/sensor/xiaomi_ble.html

bjeanes, 2022-10-25:
The Xiaomi BLE doesn't support Aranet nor any device which publishes CO2.

I also would like to see support for the Aranet4.

There seems to be some prior art on this for the ESP32: https://github.com/Anrijs/Aranet4-ESP32

stefanthoss, 2022-12-10:
ESPHome already supports CO2/temperature/humidity, you just have to configure sensors using the correct BLE characteristic UUIDs. I wrote an ESPHome package to do this for the Aranet4: https://github.com/stefanthoss/esphome-aranet4