esphome/feature-requests #641: Add ports to docker-file instead of network-mode:host

GoingOffRoading, 2020-03-16:
**Describe the problem you have/What new integration you would like**

Setting up ESPHome in Docker with the UI requires setting the container to network_mode:host.  For security and consistency concerns, it would be more ideal to expose specific ports instead of giving the container more control.  

**Please describe your use case for this integration and alternatives you've tried:**

n/a - I have no idea what ports the ESPHome container uses as it's not documented nor in the DockerFile

**Additional context**

brandond, 2020-03-16:
Direct network attachment is required for mDNS to work, because it runs over multicast on the local network segment. mDNS is used to look up node names in the .local zone for API connections and flash uploads. You can either use host mode or bridged mode, but simply exposing ports will break several core functions of the esphome docker container.

myradon, 2020-04-10:
`docker run -itd -p 6052:6052 -p 6123:6123 -v /home/raymond/docker/esphome:/config -e TZ=Europe/Amsterdam -e ESPHOME_DASHBOARD_USE_PING= true --name=esphome --hostname=esphome --ip=192.168.132.132 --net=macvlan132 --dns=192.168.132.1 --dns-search=domain.com --restart=always  --user=1000:1000  config/ dashboard `

This config runs perfect on IOT-vlan which is pretty restricted by firewall (pfSense). Also I installed mDNS/Bonjour/Avahi on firewall to forward mDNS packets between vlans and a rule to allow traffic mDNS IP-address.

nagyrobi, 2022-07-07:
It's like this by design.

closed 2022-07-07 by nagyrobi