esphome/feature-requests #642: R307 Fingerprint sensor support on ESP Home

ceaswaran, 2020-03-17:
**Describe the problem you have/What new integration you would like**
Can I use this sensor R307 for esphome ?
https://www.electronicscomp.com/sensors-module/sensors/finger-print-sensor/finger-print-sensor-module-r307-india

**Please describe your use case for this integration and alternatives you've tried:**
I tried in arduino code but I don't know how to make it in ESPHOME.
the audrino code uses UART software serial  communication
following are the audrino code that i have tested & found Ok. I dont know how to use in esphome
Fingerprint Registry code : https://www.codepile.net/pile/RKDg2yeX
Fingerprint Check code : https://www.codepile.net/pile/G2El56Dy 
I want to use this sensor for security purposes for various home assistant integration to trigger various states/devices in a secure way.

**Additional context**

puneit-hassio, 2020-03-17:
Oh, I was looking for this same sensor integration. Thanks for creating the request. @OttoWinter any chance we can have this integration in esphome. It will make esphome based biometric locks.

zurpia, 2020-05-07:
This integration would be awesome! Other identification systems are great but this one could increase the security of our esphome devices.  Thanks for creating this request and thanks to all developers!

OnFreund, 2020-09-23:
I have a working custom component that works with fingerprint readers, but it needs a lot of adjustments to be a real core component, and I don't have any experience with contributing to ESP Home. If anyone with more experience is willing to help, I'm sure we can make this happen.

loongyh, 2020-10-07:
> I have a working custom component that works with fingerprint readers, but it needs a lot of adjustments to be a real core component, and I don't have any experience with contributing to ESP Home. If anyone with more experience is willing to help, I'm sure we can make this happen.

Sure thing! Are you using the [adafruit FP lib](https://github.com/OnFreund/Adafruit-Fingerprint-Sensor-Library)? I see in your fork you have added support for instantiating a sensor with a serial stream, how does your custom component consume it?

OnFreund, 2020-10-07:
> Sure thing! Are you using the adafruit FP lib? I see in your fork you have added support for instantiating a sensor with a serial stream, how does your custom component consume it?

My initial plan was to use a `UARTComponent` for that, which is why I added the stream support, but I ended up just using `SoftwareSerial` for my purposes.

I did do some work to generalize it and uploaded it here:
https://github.com/OnFreund/esphome-fingerprint

I haven't tested it (didn't even compile it yet), and it's missing some things (e.g. led control), but the code that it's based on is working. Happy to get your thoughts, PRs, etc...

zurpia, 2020-10-07:
I have no experience in coding but I can test code because I have just ordered one R307 Fingerprint device. Let me know if you need anyone testing it. Thanks!

OnFreund, 2020-10-08:
Thanks @zurpia. More testing can definitely help.
@loongyh let me know what you think of the code, and feel free to open PRs against it, or change it completely.

loongyh, 2020-10-08:
@OnFreund @zurpia Thanks, need some time to look through. Will update within the next week.

loongyh, 2020-10-13:
@OnFreund Have [forked](https://github.com/loongyh/esphome-fingerprint/tree/dev) your component and currently working on integrating the FP lib. Pushing my progress regularly to the `dev` branch. Will submit a PR when the first working version is ready.

Just a few thoughts during working on the component:
1. Allow the user to set the address instead of the hardcoded `0xFFFFFFFF`, such that it will be possible to hook up multiple FP readers thru a multidrop bus such as RS485. Probably will need to then instantiate each one under seperate `FingerprintDevice` instances.
2. Is `sensing_pin_` used for triggering an interrupt when a finger is detected? Not sure if [interrupts are supported](https://github.com/esphome/esphome-core/issues/120).
3. Shouldn't we expose internal status stuff such as `fingerprint_count`, `capacity`, `last_finger_id`, etc. as state attributes instead of individual sensor entities?

OnFreund, 2020-10-13:
Thanks @loongyh!

1. Yes, that makes total sense.
2. The sensing pin detects whether a finger is on the reader. Right now, I'm using it to return quickly instead of calling the reader when no finger is present. On an ESP32, it can also be potentially used to wake up from deep sleep. if interrupts are ever supported, that's also a use case.
3. I thought that Esphome doesn't support attributes and you have to expose them as separate entities...

I also noticed that you've re-implemented the Adafruit library on your branch. What's the motivation? In any case, I would still keep it as a separate class, even if it's implemented within the component.

loongyh, 2020-10-13:
> 2. The sensing pin detects whether a finger is on the reader. Right now, I'm using it to return quickly instead of calling the reader when no finger is present. On an ESP32, it can also be potentially used to wake up from deep sleep. if interrupts are ever supported, that's also a use case.

Noted, will think about how best to implement this pin.

> 3. I thought that Esphome doesn't support attributes and you have to expose them as separate entities...

Weird, I'm using [esphome-mitsubishiheatpump](https://github.com/geoffdavis/esphome-mitsubishiheatpump) that does have HA attributes, will have to investigate how it's implemented: [image]

> I also noticed that you've re-implemented the Adafruit library on your branch. What's the motivation?

Well, the way the Adafruit FP lib is structured, it natively initializes Arduino Serial, so I was going about thinking how best to refactor it to use `UARTComponent` instead: Why fork the lib to refactor the functions, when we can optimize it by integrating just the relevant functions into `FingerprintComponent`?
Also to conform to this [codebase standard](https://esphome.io/guides/contributing.html#codebase-standards):
`The number of external libraries should be kept to a minimum. If the component you’re developing has a simple communication interface, please consider implementing the library natively in ESPHome.`
I have been referencing other existing ESPHome components, and seems like cherry picking functions from libs is common practice here:
- [max31856](https://github.com/esphome/esphome/blob/0c87a9ad2c7b73127eab80f582468456ceedec4b/esphome/components/max31856/max31856.cpp#L11)
- [max31865](https://github.com/esphome/esphome/blob/0c87a9ad2c7b73127eab80f582468456ceedec4b/esphome/components/max31865/max31865.cpp#L180)
- [tcs34725](https://github.com/esphome/esphome/blob/0c87a9ad2c7b73127eab80f582468456ceedec4b/esphome/components/tcs34725/tcs34725.cpp#L89)

Lastly, because it's fun and educational ;)

> In any case, I would still keep it as a separate class, even if it's implemented within the component.

Haven't got about thinking how to structure all the functions into classes. Do you propose just keeping the FP communication functions in one seperate class, or shall we instead split them among many functional classes such as  enrollment, template/model manipulation, led control, config settings (i.e. set address, password, baud rate etc.)?

OnFreund, 2020-10-13:
> Weird, I'm using esphome-mitsubishiheatpump that does have HA attributes

That's because the climate component has built-in attributes, but AFAIK esphome doesn't support generic attributes. In order to use attributes here, we'll need to add a fingerprint reader component to HA (and implement the relevant platform in the HA Esphome integration), and a fingerprint reader component type to esphome. That's a pretty big undertaking, and I'm not even sure it'll be accepted.

> Well, the way the Adafruit FP lib is structured, it natively initializes Arduino Serial, so I was going about thinking how best to refactor it to use UARTComponent instead

I think that adding another constructor that accepts a stream does most of the work.

> Also to conform to this codebase standard

Cool, didn't know that, although it does mean we'll have to keep up with any relevant changes to the AdaFruit library. I feel like this is somewhat borderline as having "a simple communication interface" and can go both ways.

> Lastly, because it's fun and educational ;)

That's definitely a goal I can get behind :)

> Do you propose just keeping the FP communication functions in one separate class, or shall we instead split them among many functional classes such as enrollment, template/model manipulation, led control, config settings (i.e. set address, password, baud rate etc.)?

I would start with keeping it as close as possible to the Ada fruit library and keeping the low level communication encapsulated, while high level workflows (initializing, scanning & matching, enrolling, etc...) in the component.
Also, since we know the library is working (also within Esphome as a generic component), I think it's best to focus on getting the Esphome component work right. After that's done, we can think about additional ways to refactor.

loongyh, 2020-10-13:
> ...we'll need to add a fingerprint reader component to HA (and implement the relevant platform in the HA Esphome integration), and a fingerprint reader component type to esphome.

Ok 'pretty big undertaking' seems like a gross understatement - sensors we go, for now.

> I think that adding another constructor that accepts a stream does most of the work.

Yep that's what you did for your fork of the Adafruit FP lib, however if I understand correctly in `UARTComponent` a Stream is already instantiated and by inheriting it in `FingerprintReaderComponent` we can directly access it. This is just my conjecture for now, until I get about finally finishing it and ready for compile ;)

> ...keep up with any relevant changes to the AdaFruit library.

Hmm are you expecting the Adafruit lib to change significantly?

> ...low level communication encapsulated, while high level workflows (initializing, scanning & matching, enrolling, etc...) in the component.

You mean to structure like this?:
LowLevelComponent::enum class PacketType
LowLevelComponent::enum class Command
LowLevelComponent::enum class Response
LowLevelComponent::enum class LED
LowLevelComponent::struct FingerprintPacket
LowLevelComponent::verify_password()
LowLevelComponent::get_parameters()
LowLevelComponent::get_image()
LowLevelComponent::image_2_tz()
LowLevelComponent::create_model()
LowLevelComponent::store_model()
LowLevelComponent::load_model()
LowLevelComponent::get_model()
LowLevelComponent::delete_model()
LowLevelComponent::empty_database()
LowLevelComponent::finger_fast_search()
LowLevelComponent::finger_search()
LowLevelComponent::get_template_count()
LowLevelComponent::led_control()
LowLevelComponent::writeStructuredPacket()
LowLevelComponent::getStructuredPacket()

FingerprintReaderComponent::set_fingerprint_count_sensor()
FingerprintReaderComponent::set_status_sensor
FingerprintReaderComponent::set_capacity_sensor()
FingerprintReaderComponent::set_security_level_sensor()
FingerprintReaderComponent::set_last_finger_id_sensor()
FingerprintReaderComponent::set_last_confidence_sensor()
FingerprintReaderComponent::set_enrolling_binary_sensor()
FingerprintReaderComponent::set_password_sensor()
FingerprintReaderComponent::set_fingerprint_count_sensor()
FingerprintReaderComponent::enroll_fingerprint()
FingerprintReaderComponent::finish_enrollment()
FingerprintReaderComponent::scan_and_match()
FingerprintReaderComponent::scan_image()
FingerprintReaderComponent::scan_and_match()

Too many functions, considering consolidating them according to type, e.g. grouping the *_model() functions into one handling function with a switch statement, [something like this](https://github.com/esphome/esphome/blob/0c87a9ad2c7b73127eab80f582468456ceedec4b/esphome/components/tuya/tuya.cpp#L112-L191). Hence the refactoring of `#define`s to `enum class`es.

> Also, since we know the library is working (also within Esphome as a generic component), I think it's best to focus on getting the Esphome component work right.

It's from my understanding that since what you have already works right, my scope of work here is just refactoring :sweat_smile:. Afaik for the high level functions you have implemented enrollment and automation triggers on finger scan? Are there any other essential features that need to be implemented for this to be a component?

loongyh, 2020-10-13:
> ...along with releasing a new reader model...

You mean the [R503](https://www.adafruit.com/product/4651)? Yep I'm using that model :smirk:. Seems like aside from the LED control, everything else remains the same.

> First of all, everything you do to help here is really appreciated.

Likewise! tbf this is also my first time working on a brand new ESPHome component, I've been learning a lot on the go and happy to help since I need to get my R503s working with ESPHome and you already managed to get it working...

> ...the configuration (pins, password, etc...) is hard coded

For the pins, it's just sensing_pin, tx, rx right? I see in your repo you have sensing_pin defined in as a conf in py already, the tx and rx will be handled by `UARTComponent`.

> I would say the first priority would be to get it working

So what you have actually working is one big component class that has everything hardcoded, meaning no py at all? In your esphome-fingerprint repo, what I've been working off now, is your first attempt at making it into an ESPHome component which is probably not working yet. Asking because I haven't actually tried to compile your original repo.

Basically what I did was copying over most of the functions from the lib and referencing other ESPHome components and learning the idiomatic way. I intended to cut away a few unneeded functions, simplify the functions to eventually end up with a lean and standalone idiomatic component. I've been too engrossed with the cpp and haven't actually looked through the py. I thought what you mean by 'needs a lot of adjustments to be a real core component' and idiomaticizing it to be refactoring to follow the ESPHome coding standards. Now I understand you mean a more user-centric functional context.

Ok I'll put aside what I'm doing now and let's get started on the py side. I've [added a project tracker](https://github.com/loongyh/esphome-fingerprint/projects/1), also added you as collaborator. I think we should plan out the scope and order of work first so please do open issues and use the project boards.

OnFreund, 2020-10-13:
I also use the R503. The API is mostly the same, but besides LED control it also has 6 buffers rather than just 2 (although I only mentioned to enroll with up to 5), and it's capacitive rather than optical.

> For the pins, it's just sensing_pin, tx, rx right? I see in your repo you have sensing_pin defined in as a conf in py already, the tx and rx will be handled by UARTComponent.

Correct.

> So what you have actually working is one big component class that has everything hardcoded, meaning no py at all? In your esphome-fingerprint repo, what I've been working off now, is your first [...]

Exactly. My working code (already in use), is one class, no py, and no customization available - it's always the same pins, the details of the HA events and services are hardcoded, LED control is also hard coded, etc... The way I envision an idiomatic core component (which I've never done), is using triggers and actions, and the actual application level behavior in API (so for example, instead of hard coding a service from HA to enroll, and an event when it's done, the YAML code should define any HA service, and define a trigger for the enrollment to send an event. This is what I attempted to do with my repo, but I've never actually tested it (nor even compiled it).

I'lll take a look at the tracker.

loongyh, 2020-10-14:
What should we call this component?
- "Fingerprint Reader" - too generic and precludes the implementation of other types of readers.
- "R307/R503 Fingerprint Reader" - too specific for future models that may use this same protocol.
- "Adafruit Fingerprint Reader" - not sure if the name is trademarked and we're allowed to use it.

OnFreund, 2020-10-14:
Since I haven't really seen good fingerprint modules outside of the Rxxx family, I think I'm ok with a generic component and a yaml configuration option.

loongyh, 2020-10-14:
Hmm I'm thinking we can also implement abstractions/traits in this generic Fingerprint Reader platform just like the climate platform.

- status (string "Idle", "Scanning", "Enrolling", "Unavailable", etc.)
- currently enrolled templates (int)
- last scanned finger (perhaps an int representing which slot the scanned finger corresponded to)
...that's all I can think of for now

loongyh, 2020-10-14:
On second thought, since there's only Rxxx, I don't think creating a new base platform will make sense for now.

zurpia, 2020-10-14:
I have just received a R307 device. If you need any testing tell me!

loongyh, 2020-10-14:
@zurpia definitely! Currently we are still planning out the core functions and yaml config options.

If there's any other features/use cases you'd like to see, do let us know!

loongyh, 2020-10-27:
@zurpia we have finished the component, do give it a test drive in my [pr here](https://github.com/OnFreund/esphome-fingerprint/pull/1)

closed 2021-04-28 by jesserockz