esphome/feature-requests #643: Set clock on Xiaomi LYWSD02

kain, 2020-03-17:
**Describe the problem you have/What new integration you would like**

Synchronize clock on Xiaomi BLE Thermometer LYWSD02 Model via SNTP

**Please describe your use case for this integration and alternatives you've tried:**

Before you can use this device you must setup it using smartphone and official app to setup clock and timezone. But if this FR will be implemented this step can be skipped. After adding device to ESPhome it can be synced via SNTP and get timezone automatically (from HomeAssistant or configured in YAML via options).

**Additional context**

Setting timezone can be skipped or forced to another value by supplying option in YAML config. My LYWSD02 clock drift to +1 minute in 1 month, so periodical SNTP syncing would be good too. Syncing frequency can be set in YAML option too. I found working code to set clock and timezone for this device at Mikhail Baranov's github: https://github.com/h4/lywsd02

thehaxxa, 2020-06-19:
As a temporary work around for those who do not want to install MiHome you can use GadgetBridge on Android to set the time. Hopefully this gets added as setting the time each month is a bit of a pain.

regnets, 2020-10-11:
I have posted a bounty on this.
https://www.bountysource.com/issues/90271090-set-clock-on-xiaomi-lywsd02

saso5, 2020-12-24:
Another temporary workaround: I've created a one-time script to update the clock that you can run from the browser https://saso5.github.io/LYWSD02-clock-sync/

mivilleb, 2021-01-14:
I have tried https://saso5.github.io/LYWSD02-clock-sync/  but it does not work for New Zealand in the summer time as we are GMT + 13 hours and the script only works up to + 12 hours, any way to adjust it?

saso5, 2021-01-15:
It would probably be best to have this conversation on my repo, so as not to spam other people here, but as I'm already replaying, I might as well answer your question :)

I don't think it's my script that has the problem, but the clock doesn't recognize +13 GMT. Maybe try setting it to -11?

mivilleb, 2021-01-15:
Thank so much -11 worked, brilliant!

ChefkeGremmen, 2021-03-12:
I would be happy to test any beta version in my setup. Unfortunately I do not have the spare time to implement this feature myself.

suhail8487, 2021-05-22:
Can someone help me in setting the timezone for India which is +530. Any help is much appreciated. Also hope that if I reconnect to the Mi Home App it doesn't reset it to +5.

Joseris, 2021-05-22:
I use Gadgetbridge app, just connect and wait for time to sync.

suhail8487, 2021-05-22:
I did that but it only work with timezones which doesn't have 30mins offset. The only app that seems to do the job is MijiaTemp. But I want to use something which is permanent and can be included in Mi Home app

ssieb, 2021-05-24:
These last few comments are not related to esphome.  You're in the wrong place for that.

marciogranzotto, 2021-11-08:
Guys, any idea if its possible to create a `Custom Switch` with this funcionality? I'm trying to do that, but I think that if I extend `BLEClient` or `BLEClientSwitch` I can't get access to the write GATT stuff...

ChefkeGremmen, 2022-01-21:
A little addition to the original request. For me it does not necessarily have to sync with SNTP, with HomeAssistant is also fine.

TheHomieFox, 2022-04-21:
I'm also looking for a way to automate the time adjustment... Is it possible to run @saso5's web script from a Home Assistant automation, in an unattended manner? Any thoughts would be appreciated!

finipini, 2022-09-07:
Hi, it would be awsome this could be implemented now with either the new ble proxy capability with home assistant.
Anyone knows what thee codes to use to do this?

lolorc, 2022-10-06:
especially now the ble proxy can be active.
I'm using https://github.com/sullin/hyg to set clock atm.

kain, 2022-10-06:
> especially now the ble proxy can be active. I'm using https://github.com/sullin/hyg to set clock atm.

Thank you for this finding! Now I'm able to sync clocks time too and do it unattended. Also thanks goes to the author of this mini tool.

Quick how-to if anybody needs it:
- clone repository: `git clone https://github.com/sullin/hyg`
- `cd hyg`
- `sudo apt install build-essential`
- `sudo apt install --no-install-recommends libbluetooth-dev`
- `make`
- find your device MAC address: `sudo hcitool lescan`
- `./hyg -c -t 02_0042 E7:2E:XX:XX:XX:XX`
- add to crontab (optionally): `crontab -e`
- adjust and paste to the end (run at night on 26th day of month):
`0  5  26  *   *     /home/kain/hyg/hyg -c -t 02_0042 E7:2E:XX:XX:XX:XX &>/dev/null`

finipini, 2022-10-11:
> Thank you for this finding! Now I'm able to sync clocks time too and do it unattended. Also thanks goes to the author of this mini tool.
>
> Quick how-to if anybody needs it:
>
> * clone repository: `git [...]

That,s interesting. Can that be done with ble_client ?

hwinkel, 2022-11-03:
wold be interested as well to have it as a service in Homeassistant via BLE Proxy, or directly in ESP Home what ever fits better

finipini, 2022-11-22:
> wold be interested as well to have it as a service in Homeassistant via BLE Proxy, or directly in ESP Home what ever fits better

Yes it would. Maybe we should make this PR there too ?!

fire1ce, 2022-12-23:
following.

ephirial, 2022-12-26:
Hey, I am successfully updating my clock via SNTP directly with an ESP32 using ESPHome.

```
time:
  - platform: sntp
    id: sntp_time
    servers: "192.168.2.254"
    timezone: "Europe/Berlin"

esp32_ble_tracker:
  
ble_client:
  - mac_address: E7:2E:00:73:4C:C5
    id:  lywsd02_clock
    on_connect:
      then:
        - lambda: |-
            ESP_LOGD("ble_client_lambda", "Connected to lywsd02_clock!");
        - delay: 20s
        - ble_client.ble_write:
            id: lywsd02_clock
            service_uuid: ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6 
            characteristic_uuid: EBE0CCB7-7A0A-4B0C-8A1A-6FF2997DA3A6
            value: !lambda |-
                auto time = id(sntp_time).utcnow();
                uint8_t time_byte1 =  time.timestamp & 0x000000ff;
                uint8_t time_byte2 = (time.timestamp & 0x0000ff00) >> 8;
                uint8_t time_byte3 = (time.timestamp & 0x00ff0000) >> 16;
                uint8_t time_byte4 = (time.timestamp & 0xff000000) >> 24;
                uint8_t timezone_byte = 0x01;
                return {time_byte1, time_byte2, time_byte3, time_byte4, timezone_byte};
        - delay: 20s
        - switch.turn_off: lywsd02_ble_connection
    on_disconnect:
      then:
        - lambda: |-
            ESP_LOGD("ble_client_lambda", "Disconnected from lywsd02_clock!");

switch:
  - platform: ble_client
    ble_client_id: lywsd02_clock
    id: "lywsd02_ble_connection"
    restore_mode: ALWAYS_OFF # Disable the client by default, as connection is only triggered via automation

interval:
  - interval: 4h
    then:
      - switch.turn_on: lywsd02_ble_connection
```

It only connects to the device within the specified interval, sets the clock, then disconnects to save battery.
Receiving data works via HomeAssistant with BLE Proxy, so I am only setting the clock manually.
The timezone has to be set in the variable _timezone_byte_.

My code is not perfect, I think I could also enable/disable the _esp32_ble_tracker_ via the automation. But I am not sure if that would mess with the _bluetooth_proxy_.

You could also turn on the _lywsd02_ble_connection_ switch via HomeAssistant and make an automation there. But for me it´s enough that the ESP just silently updates the clock every x hours.
My interval of 4 hours also might drain the battery too much, once a week is probably fine. But I was still testing this recently.

finipini, 2022-12-28:
> Hey, I am successfully updating my clock via SNTP directly with an ESP32 using ESPHome.
>
> ```
> time:
>   - platform: sntp
>     id: sntp_time
>     servers: "192.168.2.254"
>     timezone: "Europe/Berlin" [...]

That works great, Nice job.

Althow i have one problem, can,t get right my time zone  ( Europe/Lisbon) . I tryed diferent things but always get One more hour then it should

koenvervloesem, 2022-12-28:
> Althow i have one problem, can,t get right my time zone ( Europe/Lisbon) . I tryed diferent things but always get One more hour then it should

It's because of this:

```
uint8_t timezone_byte = 0x01;
```

This is the timezone offset, so in this case UTC+1. If you're in UTC timezone, as is the case now for Europe/Lisbon, you need to set this byte to 0.

finipini, 2022-12-28:
Ok, Will try thanks.

Joseris, 2022-12-28:
Thanks @ephirial for your code.

I have LYWSD02 for couple of years and only needed to sync clock after time zone has changed to and from DST.
So I modified that code to sync time zone from HomeAssistant.
Just had to subtract 30 from received offset to set correct time zone, not sure if it is format or conversion issue:
```
uint8_t timezone_byte = (time.timezone_offset() & 0x000000ff) - 30;
```
Also added trigger to sync clock just every Sunday night (this time is in UTC) and not repeatedly every 4 hours:
```
time:
  - platform: homeassistant
    id: homeassistant_time
    on_time:
      - seconds: 0
        minutes: 0
        hours: 3
        days_of_week: SUN
        then:
          - switch.turn_on: lywsd02_ble_connection
```

ephirial, 2022-12-28:
Thanks @Joseris for your addition! I did not know that you could get the timezone via _time.timezone_offset()_, of course that is more elegant than setting it manually.
When I look at the variable _time.timezone_offset()_, I get the offset in seconds. So I think this would just have to be divided by 3600?

`uint8_t timezone_byte = time.timezone_offset() / 3600;`

This works for me, maybe you can try it with your timezone?
If you live in a timezone with 30 minute offset, the offset has to be added to the time variable beforehand (you could just add 1800 seconds to _time.timestamp_). The clock only supports full hour offsets.
I think I will update my previous comment with this change and the weekly trigger, my 4 hour trigger is too much.

Joseris, 2022-12-28:
You are right, time.timezone_offset() is in seconds. I'm in UTC+2 and dividing by 3600 works for me.
Thanks.

hwinkel, 2022-12-28:
Can someone provide the fully working ESPhome yaml, with the recent changes, eventually as a gist or repo ?

marciogranzotto, 2023-03-12:
I've tried with `uint8_t timezone_byte = time.timezone_offset() / 3600;` but it doesn't work for UTC-3. Probably it needs a different format for negative timezones?

marciogranzotto, 2023-03-12:
got it working! Just need to add 256 to the timezone value if it's negative:

```yaml
time:
  - platform: sntp
    id: sntp_time
    timezone: "America/Sao_Paulo"

esp32_ble_tracker:
  
ble_client:
  - mac_address: $ble_sensor_mac
    id:  lywsd02_clock
    on_connect:
      then:
        - lambda: |-
            ESP_LOGD("ble_client_lambda", "Connected to lywsd02_clock!");
            id(sntp_time).update();
        - delay: 20s
        - ble_client.ble_write:
            id: lywsd02_clock
            service_uuid: ${time_service_uuid}
            characteristic_uuid: ${time_characteristic_uuid}
            value: !lambda |-
                auto time = id(sntp_time).now();
                auto timezone = time.timezone_offset() / 3600;
                if (timezone < 0) {
                  timezone += 256;
                }

                uint8_t time_byte1 =  time.timestamp & 0x000000ff;
                uint8_t time_byte2 = (time.timestamp & 0x0000ff00) >> 8;
                uint8_t time_byte3 = (time.timestamp & 0x00ff0000) >> 16;
                uint8_t time_byte4 = (time.timestamp & 0xff000000) >> 24;
                uint8_t timezone_byte = timezone;
                return {time_byte1, time_byte2, time_byte3, time_byte4, timezone_byte};
        - delay: 20s
        - switch.turn_off: lywsd02_ble_connection
    on_disconnect:
      then:
        - lambda: |-
            ESP_LOGD("ble_client_lambda", "Disconnected from lywsd02_clock!");
            
```

finipini, 2023-03-26:
Any one with more then one of this LYWSD02 ?  

i have it working with just one ble_client, but ifi had more ( even making an switch for each, and turning on one at a time) this doesn,t work.

goldbingo, 2023-10-11:
> hello, i have 3 ble proxy esp32 in ha.
> 
> is it possible to use the same esp32 as ble proxy and for this time sync task?

I am using my ble proxy for this time sync. It works well.

propi62, 2023-10-11:
timezone is ok for me like this, but what is the server ip? (is it save to remove this line?)

what to fill in in this two lines?
            service_uuid: ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6 
            characteristic_uuid: EBE0CCB7-7A0A-4B0C-8A1A-6FF2997DA3A6

my device extracted with mi token extractor:
   NAME:     Mi Temperature and Humidity Monitor Digital Clock
   ID:       blt.3.18jriud48ls00
   BLE KEY:  3f5d2248e40711aea253a25tFFFFFFFF (faked)
   MAC:      17:75:BC:E1:74:99
   TOKEN:    83b65b7f56237365f5a25257 (faked)
   MODEL:    miaomiaoce.sensor_ht.t1

hwinkel, 2023-11-04:
Thanks for the effort, it works from a single ESP32 with a single Clock in reach.

I have 4 Clocks around the house and some ESP32 acting as proxy already, would it be sensible to shift the function to HA and an automation and connect vie Bluetooth Proxy like described here:

https://community.home-assistant.io/t/set-clock-over-bluetooth-proxy-on-lywsd02-by-xiaomi-bluetooth-temp-humidity-clock-device/560924

hwinkel, 2023-11-05:
Or defining the Sync Clock call as Service with variables for the Clock MAC?

https://esphome.io/components/api.html#api-services

finipini, 2023-11-05:
> Or defining the Sync Clock call as Service with variables for the Clock MAC?
> 
> https://esphome.io/components/api.html#api-services

Yes, that is what i tryed. But it seems that the mac can,t be and variable. But maybe we can do it in other way.

I believe that ble proxy would do it easally, for exemple ble proxy already pulls automaticaly the value of the battery of mi flora devices , and for that it needs to BE an active connection.

Btw i already mentioned that setting the clock of these thermometer is easaly made with tasmota, even with more then one click/thermometer, i,m not comparing the two just saying that we could see how they do it. But i really don,t know where that info is.

hwinkel, 2023-11-05:
Thanks for the Tasmota Hint, will google and check.

philiprusinov, 2024-04-10:
Any idea, why I can't write to the clock?
I've just modified the timezone and the clock MAC address, but still unable to upload the code to the board, due to the error message below:

`Unable to find action with the name 'ble_client.ble_write'.
          ble_client.ble_write:  [source /config/esphome/esphome-web-73f758.yaml:47]`

finipini, 2024-04-10:
> Any idea, why I can't write to the clock? I've just modified the timezone and the clock MAC address, but still unable to upload the code to the board, due to the error message below:
>
> `Unable to [...]

You got to show all your config to someone helps you.

Btw i already quite on this . An user created an nice custom component that sets the hours on that clocks, and it uses the ble of home assistant ( proxys included) 

https://github.com/ashald/home-assistant-lywsd02/tree/main

philiprusinov, 2024-04-10:
The code is exact the same like posted here, only MAC address and wifi settings and timezone were changed:
```

esphome:
  name: bletracker
  platform: ESP32
  board: esp32dev

# Enable logging
logger:

# Enable Home Assistant API
api:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  
time:
  - platform: sntp
    id: sntp_time
    timezone: "Europe/Sofia"

esp32_ble_tracker:
  
ble_client:
  - mac_address: 17:75:BC:E1:FC:E1
    id:  lywsd02_clock
    on_connect:
      then:
        - lambda: |-
            ESP_LOGD("ble_client_lambda", "Connected to lywsd02_clock!");
            id(sntp_time).update();
        - delay: 20s
        - ble_client.ble_write:
            id: lywsd02_clock
            service_uuid: ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6 
            characteristic_uuid: EBE0CCB7-7A0A-4B0C-8A1A-6FF2997DA3A6
            value: !lambda |-
                auto time = id(sntp_time).now();
                auto timezone = time.timezone_offset() / 3600;
                if (timezone < 0) {
                  timezone += 256;
                }

                uint8_t time_byte1 =  time.timestamp & 0x000000ff;
                uint8_t time_byte2 = (time.timestamp & 0x0000ff00) >> 8;
                uint8_t time_byte3 = (time.timestamp & 0x00ff0000) >> 16;
                uint8_t time_byte4 = (time.timestamp & 0xff000000) >> 24;
                uint8_t timezone_byte = timezone;
                return {time_byte1, time_byte2, time_byte3, time_byte4, timezone_byte};
        - delay: 20s
        - switch.turn_off: lywsd02_ble_connection
    on_disconnect:
      then:
        - lambda: |-
            ESP_LOGD("ble_client_lambda", "Disconnected from lywsd02_clock!");
            
```

pedasmith, 2026-10-11:
> 19\. > got it working! Just need to add 256 to the timezone value if it's negative:

Hot diggity! Thank you for this! I've been trying to figure out why my own Windows bluetooth app would set the time 15 minutes off on the MJWSD05MMC device that I picked up earlier this year. This exactly and perfectly solved the problem!