esphome/feature-requests #644: Esphome REST calls Shelly

MrLaaky, 2020-03-18:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

Hello,
I have shelly1 and I would like to make a manual switch connected to esphome component(in my case D1 mini). I know to how make binary sensor to detect states of the manual switch. What I have trouble with action the the switch. 
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0   <----manual switch is connected to GPIO0 and Ground
      mode: INPUT_PULLUP
      inverted: True
    name: "Shelly button"
    on_press:
      - switch.toggle: shelly

The next logical step is to create toggle switch called shelly.
The website 
https://esphome.io/web-api/index.html
says among other things:
By creating a simple GET request for a URL...
I find it really hard how to do that in esphome yaml file.( The rest component in my case must be POST).

The following is the command used in home assistant and it works 
platform: rest
name: "Living Room Light"
scan_interval: 5
resource: http://192.168.1.252/relay/0  <---shelly ip
body_on: 'turn=on'
body_off: 'turn=off'
is_on_template: '{{ value_json.ison == true}}'
headers:
  content-type: application/x-www-form-urlencoded

Putting it all together:
How do I make a manual switch that is able to change state of a shelly switch via rest component.
Kind regards

brandond, 2020-03-18:
You should be using the Native or MQTT integration to control your esphome device from Home Assistant.

closed 2020-03-18 by brandond

MrLaaky, 2020-03-18:
Additional information:
I would like to use eshome(D1 mini) as standalone product without home assistant.
I  know it would be very easy to configure it with HA, but I don't have the luxury of having HA in that specific instance. 
My intention way using shelly and esphome as a standalone product in combination with each other.
I included HA in the above post because I wanted to get similar rest function in esphome(which is according to documentation easy to configure, but I cannot do it myself).
Do you think it is possible or is it overly complicated to send POST command in esphome?

randybb, 2020-03-18:
ESPHome is aimed to be used with Home Assistant. If you don't want to use the benefits that it gives you (to be honest I don't understand why people are still using mqtt and then do all the mapping in HA), then why do you want to use it?