esphome/feature-requests #645: Prefix for all entities visible in HA.

freekode, 2020-03-19:
**Describe the problem you have/What new integration you would like**
I want to have automatically added prefix for all entities which are visible in HA.

In esphome I have `sensor.temp`, `sensor.pressure`, in HA i want to have `sensor.myweatherstation_temp`, `sensor.myweatherstation_pressure` 

**Please describe your use case for this integration and alternatives you've tried:**
In esphome configuration I have sensors with names: temp, humidity, pressure.
These sensors transfers to HA with the same names: sensor.temp, sensor.humidity, etc. Which can produce name conflicts in HA.

**Additional context**

brandond, 2020-03-19:
Why not just change the names in your esphome configuration? If that feels too verbose you can always use substitutions:

```yaml
substitutions:
  name: "myweatherstation"

sensor:
  - platform: wifi_signal
    name: "${name} wifi"
```

freekode, 2020-03-20:
> Why not just change the names in your esphome configuration? 
manually change name for every sensor? lame

randybb, 2020-03-20:
I would say it is clever and modular. All my esp's are named by their location (like "node_basement") and sensor itself is named by its purpose ("Heating Temp") as each device controls more stuff.

freekode, 2020-03-26:
hm...good point, now it's make sense

closed 2020-03-26 by freekode